mikepurvis/ros-install-osx #3: rosdep wants to install from Taps/indgo after this script

kjeremy, 2015-07-23:
I'm trying to install the turtlebot_simulator using `rosdep install` but it wants to satisfy requirements by tapping ros/indigo.  Is that correct?  It seems like it should prefer a source install though I'm a ROS newbie and might not understand the situation.

mikepurvis, 2015-07-23:
Interesting, yeah, I don't think rosdep will do what you want in this scenario, since we don't have homebrew formula for released ROS packages. You'll probably have the most luck with an invocation such as the following to generate your source workspace:

    rosinstall_generator turtlebot_simulator --deps --exclude RPP > turtlebot_simulator.rosinstall
    wstool init -j8 src turtlebot_simulator.rosinstall
    catkin_make

Alternatively, if there are certain packages outside of desktop_full that you consistently need, you can add them to the `ROS_EXTRA_PACKAGES` env var, and they'll get built and installed into `/opt/ros/indigo` for you.

Glad to hear that the base install went well for you, though!

mikepurvis, 2015-07-28:
Closing the ticket now. If you have an issue with the above suggestions, please create a new ticket or start a discussion on ROS Answers.

closed 2015-07-28 by mikepurvis