mikepurvis/ros-install-osx #4: ld: library not found for -ltbb

zoercai, 2015-07-26:
Using this to create a clean install of ROS on OS X Yosemite, however I always come across this error:

    Starting ==> compressed_depth_image_transport                                                                                                                                                               
	Starting ==> compressed_image_transport                                                                                                                                                                     
	Starting ==> gazebo_ros                                                                                                                                                                                     
	Starting ==> interactive_markers                                                                                                                                                                            
	                                                                                                                                                                                                            
	[compressed_depth_image_transport] ==> '/Users/Zoe/indigo_desktop_full_ws/build/compressed_depth_image_transport/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' in '/Users/Zoe/indigo_desktop_full_ws/build/compressed_depth_image_transport'
	[ 20%] Built target compressed_depth_image_transport_gencfg
	[ 40%] Linking CXX shared library /Users/Zoe/indigo_desktop_full_ws/devel/lib/libcompressed_depth_image_transport.dylib
	ld: library not found for -ltbb
	clang: error: linker command failed with exit code 1 (use -v to see invocation)
	make[2]: *** [/Users/Zoe/indigo_desktop_full_ws/devel/lib/libcompressed_depth_image_transport.dylib] Error 1
	make[1]: *** [CMakeFiles/compressed_depth_image_transport.dir/all] Error 2
	make: *** [all] Error 2
	[compressed_depth_image_transport] <== '/Users/Zoe/indigo_desktop_full_ws/build/compressed_depth_image_transport/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' failed with return code '2'
	                                                                                                                                                                                                             
	Failed   <== compressed_depth_image_transport [ 1.5 seconds ]                                                                                                                                               
	                                                                                                                                                                                                            
	[compressed_image_transport] ==> '/Users/Zoe/indigo_desktop_full_ws/build/compressed_image_transport/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' in '/Users/Zoe/indigo_desktop_full_ws/build/compressed_image_transport'
	[ 20%] Built target compressed_image_transport_gencfg
	[ 40%] Linking CXX shared library /Users/Zoe/indigo_desktop_full_ws/devel/lib/libcompressed_image_transport.dylib
	ld: library not found for -ltbb
	clang: error: linker command failed with exit code 1 (use -v to see invocation)
	make[2]: *** [/Users/Zoe/indigo_desktop_full_ws/devel/lib/libcompressed_image_transport.dylib] Error 1
	make[1]: *** [CMakeFiles/compressed_image_transport.dir/all] Error 2
	make: *** [all] Error 2
	[compressed_image_transport] <== '/Users/Zoe/indigo_desktop_full_ws/build/compressed_image_transport/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' failed with return code '2'
	                                                                                                                                                                                                             
	Failed   <== compressed_image_transport       [ 1.5 seconds ]                                                                                                                                               
	Finished <== gazebo_ros                       [ 2.5 seconds ]                                                                                                                                               
	Finished <== interactive_markers              [ 3.1 seconds ]                                                                                                                                               
	[build] There were '2' errors:                                                                                                                                                                              

	Failed to build package 'compressed_depth_image_transport' because the following command:

	# Command to reproduce:
	cd /Users/Zoe/indigo_desktop_full_ws/build/compressed_depth_image_transport && /Users/Zoe/indigo_desktop_full_ws/build/compressed_depth_image_transport/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j; cd -

	# Path to log:
	cat /Users/Zoe/indigo_desktop_full_ws/build/build_logs/compressed_depth_image_transport.log

	Exited with return code: 2 

	Failed to build package 'compressed_image_transport' because the following command:

	# Command to reproduce:
	cd /Users/Zoe/indigo_desktop_full_ws/build/compressed_image_transport && /Users/Zoe/indigo_desktop_full_ws/build/compressed_image_transport/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j; cd -

	# Path to log:
	cat /Users/Zoe/indigo_desktop_full_ws/build/build_logs/compressed_image_transport.log

	Exited with return code: 2 
	                                                                                                                                                                                                            
	[build] Runtime: 8.5 seconds 

How can I resolve this?

mikepurvis, 2015-07-27:
Looks like an OpenCV problem. Please post the output of:

    brew info opencv

On my machine, it's:

```
$ brew info opencv
homebrew/science/opencv: stable 2.4.11 (bottled), HEAD
Open source computer vision library
http://opencv.org/
/usr/local/Cellar/opencv/2.4.10.1 (219 files, 39M)
  Built from source with: --with-qt --with-tbb
/usr/local/Cellar/opencv/2.4.11_1 (228 files, 36M) *
  Built from source with: --with-qt --with-tbb
From: https://github.com/homebrew/homebrew-science/blob/master/opencv.rb
==> Dependencies
Build: cmake ✔, pkg-config ✔
Required: jpeg ✔, libpng ✔, libtiff ✔
Recommended: eigen ✔, openexr ✔, homebrew/python/numpy ✔
Optional: gstreamer ✘, jasper ✔, libdc1394 ✘, openni ✘, qt ✔, tbb ✘, ffmpeg ✘
```

If you previously installed opencv, you may need to manually run:

```
brew reinstall opencv --with-qt --with-eigen --with-tbb
```

That should fix this for you. I'd suggest deleting the workspace's build and devel folders and re-running the `catkin build` from scratch— you may have an ABI incompatibility between stuff that built earlier on and the updated version of OpenCV.

mikepurvis, 2015-07-28:
Going to close this for now. Please re-open the ticket if the above suggestions do not resolve the issue.

zoercai, 2015-07-28:
I tried reinstalling opencv and deleting the workspace and rebuilding, but it still came up with the same error. Here's the output of `brew info opencv` on my machine:

	homebrew/science/opencv: stable 2.4.11 (bottled), HEAD
	Open source computer vision library
	http://opencv.org/
	/usr/local/Cellar/opencv/2.4.11_2 (228 files, 35M) *
	  Built from source with: --with-qt --with-tbb
	From: https://github.com/homebrew/homebrew-science/blob/master/opencv.rb
	==> Dependencies
	Build: cmake ✔, pkg-config ✔
	Required: jpeg ✔, libpng ✔, libtiff ✔
	Recommended: eigen ✔, openexr ✔, homebrew/python/numpy ✔
	Optional: gstreamer ✘, jasper ✘, libdc1394 ✘, openni ✘, qt ✔, tbb ✔, ffmpeg ✘

mikepurvis, 2015-07-28:
Hmm, interesting. In that case, I wonder if you have OpenCV installed some other way, for example from MacPorts, or via the OS X Framework scheme? Please post the output of:

    ls /Library/Frameworks
    ls /opt/local/lib

zoercai, 2015-07-28:
ls /Library/Frameworks
	AEProfiling.framework		NyxAudioAnalysis.framework
	AERegistration.framework	PluginManager.framework
	Adobe AIR.framework		WacomMultiTouch.framework
	AudioMixEngine.framework	iTunesLibrary.framework

	ls /opt/local/lib
	ls: /opt/local/lib: No such file or directory

mikepurvis, 2015-07-29:
Hmm, well. I'm out of immediate ideas. If you're willing to get more in-depth, a possible next step is, from the indigo_desktop_full_ws folder, to run something like:

```
rm -rf build/compressed_depth_image_transport
catkin build \
    -DCMAKE_BUILD_TYPE=Release \
    -DPYTHON_LIBRARY=$(python -c "import sys; print sys.prefix")/lib/libpython2.7.dylib \
    -DPYTHON_INCLUDE_DIR=$(python -c "import sys; print sys.prefix")/include/python2.7 \
    --make-args VERBOSE=1 \
    --start-with compressed_depth_image_transport
```

Then report the contents of `build/build_logs/compressed_depth_image_transport.log`, which will tell us _exactly_ what it is attempting to link against.

zoercai, 2015-07-29:
I've pasted the log [here](http://pastebin.com/gztJPGD7).

And this is the output in the terminal:

    indigo_desktop_full_ws Zoe$ catkin build \
	>     -DCMAKE_BUILD_TYPE=Release \
	>     -DPYTHON_LIBRARY=$(python -c "import sys; print sys.prefix")/lib/libpython2.7.dylib \
	>     -DPYTHON_INCLUDE_DIR=$(python -c "import sys; print sys.prefix")/include/python2.7 \
	>     --make-args VERBOSE=1 \
	>     --start-with compressed_depth_image_transport
	--------------------------------------------------------------------------------
	Profile:                     default
	Extending:                   None
	Workspace:                   /Users/Zoe/indigo_desktop_full_ws
	Source Space:       [exists] /Users/Zoe/indigo_desktop_full_ws/src
	Build Space:        [exists] /Users/Zoe/indigo_desktop_full_ws/build
	Devel Space:        [exists] /Users/Zoe/indigo_desktop_full_ws/devel
	Install Space:      [exists] /opt/ros/indigo
	DESTDIR:                     None
	--------------------------------------------------------------------------------
	Isolate Develspaces:         False
	Install Packages:            True
	Isolate Installs:            False
	--------------------------------------------------------------------------------
	Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DPYTHON_LIBRARY=/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib -DPYTHON_INCLUDE_DIR=/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/include/python2.7
	Additional Make Args:        VERBOSE=1 --start-with compressed_depth_image_transport
	Additional catkin Make Args: None
	Internal Make Job Server:    True
	--------------------------------------------------------------------------------
	Whitelisted Packages:        None
	Blacklisted Packages:        None
	--------------------------------------------------------------------------------
	Workspace configuration appears valid.
	-------------------------------------------------------------------------------- 
	Found '224' packages in 0.7 seconds. 
	Starting ==> catkin                                                             
	Starting ==> octomap                                                            
	Starting ==> orocos_kdl                                                         
	Starting ==> stage                                                              
	                                                                                
	[octomap] ==> '/Users/Zoe/indigo_desktop_full_ws/build/octomap/build_env.sh /usr/bin/make VERBOSE=1 --start-with compressed_depth_image_transport --jobserver-fds=3,4 -j' in '/Users/Zoe/indigo_desktop_full_ws/build/octomap'
	/Applications/Xcode.app/Contents/Developer/usr/bin/make: unrecognized option `--start-with'
	Usage: make [options] [target] ...
	Options:
	  -b, -m                      Ignored for compatibility.
	  -B, --always-make           Unconditionally make all targets.
	  -C DIRECTORY, --directory=DIRECTORY
	                              Change to DIRECTORY before doing anything.
	  -d                          Print lots of debugging information.
	  --debug[=FLAGS]             Print various types of debugging information.
	  -e, --environment-overrides
	                              Environment variables override makefiles.
	  -f FILE, --file=FILE, --makefile=FILE
	                              Read FILE as a makefile.
	  -h, --help                  Print this message and exit.
	  -i, --ignore-errors         Ignore errors from commands.
	  -I DIRECTORY, --include-dir=DIRECTORY
	                              Search DIRECTORY for included makefiles.
	  -j [N], --jobs[=N]          Allow N jobs at once; infinite jobs with no arg.
	  -k, --keep-going            Keep going when some targets can't be made.
	  -l [N], --load-average[=N], --max-load[=N]
	                              Don't start multiple jobs unless load is below N.
	  -L, --check-symlink-times   Use the latest mtime between symlinks and target.
	  -n, --just-print, --dry-run, --recon
	                              Don't actually run any commands; just print them.
	  -o FILE, --old-file=FILE, --assume-old=FILE
	                              Consider FILE to be very old and don't remake it.
	  -p, --print-data-base       Print make's internal database.
	  -q, --question              Run no commands; exit status says if up to date.
	  -r, --no-builtin-rules      Disable the built-in implicit rules.
[... 154 lines cut ...]
	Report bugs to <bug-make@gnu.org>
	[catkin] <== '/Users/Zoe/indigo_desktop_full_ws/build/catkin/build_env.sh /usr/bin/make VERBOSE=1 --start-with compressed_depth_image_transport --jobserver-fds=3,4 -j' failed with return code '2'
	                                                                                 
	Failed   <== catkin                           [ 1.8 seconds ]                   
	[build] Runtime: 6.2 seconds 
	Traceback (most recent call last):
	  File "/usr/local/bin/catkin", line 11, in <module>
	    sys.exit(main())
	  File "/usr/local/lib/python2.7/site-packages/catkin_tools/commands/catkin.py", line 219, in main
	    sys.exit(args.main(args) or 0)
	  File "/usr/local/lib/python2.7/site-packages/catkin_tools/verbs/catkin_build/cli.py", line 296, in main
	    summarize_build=opts.summarize  # Can be True, False, or None
	  File "/usr/local/lib/python2.7/site-packages/catkin_tools/verbs/catkin_build/build.py", line 756, in build_isolated_workspace
[... cut ...] rror_state)
	  File "/usr/local/lib/python2.7/site-packages/catkin_tools/verbs/catkin_build/build.py", line 695, in set_error_state
	    executors[x].should_shutdown = True
	KeyError: 1
	wifi-172-23-110-22:indigo_desktop_full_ws Zoe$ make --version
	GNU Make 3.81
	Copyright (C) 2006  Free Software Foundation, Inc.
	This is free software; see the source for copying conditions.
	There is NO warranty; not even for MERCHANTABILITY or FITNESS FOR A
	PARTICULAR PURPOSE.

Can you spot anything unusual?

mikepurvis, 2015-07-30:
Blah, it didn't do the right thing. Please try:

```
rm -rf build/compressed_depth_image_transport
catkin build \
    -DCMAKE_BUILD_TYPE=Release \
    -DPYTHON_LIBRARY=$(python -c "import sys; print sys.prefix")/lib/libpython2.7.dylib \
    -DPYTHON_INCLUDE_DIR=$(python -c "import sys; print sys.prefix")/include/python2.7 \
    --start-with compressed_depth_image_transport \
    --make-args VERBOSE=1
```

That should get us verbose make output, which will show that paths that it is trying to link against. Sorry for the runaround.

zoercai, 2015-07-30:
All good! [Here's the new log](http://pastebin.com/GdxHc42u)

And the output in terminal:

	indigo_desktop_full_ws Zoe$ catkin build \
	>     -DCMAKE_BUILD_TYPE=Release \
	>     -DPYTHON_LIBRARY=$(python -c "import sys; print sys.prefix")/lib/libpython2.7.dylib \
	>     -DPYTHON_INCLUDE_DIR=$(python -c "import sys; print sys.prefix")/include/python2.7 \
	>     --start-with compressed_depth_image_transport \
	>     --make-args VERBOSE=1
	--------------------------------------------------------------------------------
	Profile:                     default
	Extending:                   None
	Workspace:                   /Users/Zoe/indigo_desktop_full_ws
	Source Space:       [exists] /Users/Zoe/indigo_desktop_full_ws/src
	Build Space:        [exists] /Users/Zoe/indigo_desktop_full_ws/build
	Devel Space:        [exists] /Users/Zoe/indigo_desktop_full_ws/devel
	Install Space:      [exists] /opt/ros/indigo
	DESTDIR:                     None
	--------------------------------------------------------------------------------
	Isolate Develspaces:         False
	Install Packages:            True
	Isolate Installs:            False
	--------------------------------------------------------------------------------
	Additional CMake Args:       -DCMAKE_BUILD_TYPE=Release -DPYTHON_LIBRARY=/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib -DPYTHON_INCLUDE_DIR=/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/include/python2.7
	Additional Make Args:        VERBOSE=1
	Additional catkin Make Args: None
	Internal Make Job Server:    True
	--------------------------------------------------------------------------------
	Whitelisted Packages:        None
	Blacklisted Packages:        None
	--------------------------------------------------------------------------------
	Workspace configuration appears valid.
	-------------------------------------------------------------------------------- 
	Found '224' packages in 0.6 seconds. 
	[build] Skipping package 'catkin'                                                                                                                                                   
	[build] Skipping package 'octomap'                                                                                                                                                  
	[build] Skipping package 'orocos_kdl'                                                                                                                                               
	[build] Skipping package 'stage'                                                                                                                                                    
	[build] Skipping package 'genmsg'                                                                                                                                                   
	[build] Skipping package 'angles'                                                                                                                                                   
	[build] Skipping package 'bond_core'                                                                                                                                                
	[build] Skipping package 'cmake_modules'                                                                                                                                            
	[build] Skipping package 'common_msgs'                                                                                                                                              
	[build] Skipping package 'common_tutorials'                                                                                                                                         
	[build] Skipping package 'convex_decomposition'                                                                                                                                     
	[build] Skipping package 'cpp_common'                                                                                                                                               
	[build] Skipping package 'desktop'                                                                                                                                                  
	[build] Skipping package 'desktop_full'                                                                                                                                             
	[build] Skipping package 'diagnostics'                                                                                                                                              
	[build] Skipping package 'eigen_stl_containers'                                                                                                                                     
	[build] Skipping package 'executive_smach'                                                                                                                                          
	[build] Skipping package 'gazebo_ros_pkgs'                                                                                                                                          
	[build] Skipping package 'geometry'                                                                                                                                                 
[... cut ...]
	  File "/usr/local/lib/python2.7/site-packages/catkin_tools/verbs/catkin_build/cli.py", line 296, in main
	    summarize_build=opts.summarize  # Can be True, False, or None
	  File "/usr/local/lib/python2.7/site-packages/catkin_tools/verbs/catkin_build/build.py", line 756, in build_isolated_workspace
	    set_error_state(error_state)
	  File "/usr/local/lib/python2.7/site-packages/catkin_tools/verbs/catkin_build/build.py", line 695, in set_error_state
	    executors[x].should_shutdown = True
	KeyError: 1

mikepurvis, 2015-07-31:
Bizarre; it's definitely looking for it in the standard location. On my machine this is here:

```
$ ls -la /usr/local/lib/libopencv_ocl.2.4.11.dylib /usr/local/lib/libtbb*
lrwxr-xr-x  1 mikepurvis  admin  56  5 Jun 11:00 /usr/local/lib/libopencv_ocl.2.4.11.dylib -> ../Cellar/opencv/2.4.11_1/lib/libopencv_ocl.2.4.11.dylib
lrwxr-xr-x  1 mikepurvis  admin  43  5 Jun 10:31 /usr/local/lib/libtbb.dylib -> ../Cellar/tbb/4.3-20150424/lib/libtbb.dylib
lrwxr-xr-x  1 mikepurvis  admin  49  5 Jun 10:31 /usr/local/lib/libtbbmalloc.dylib -> ../Cellar/tbb/4.3-20150424/lib/libtbbmalloc.dylib
lrwxr-xr-x  1 mikepurvis  admin  55  5 Jun 10:31 /usr/local/lib/libtbbmalloc_proxy.dylib -> ../Cellar/tbb/4.3-20150424/lib/libtbbmalloc_proxy.dylib
```

And this is the version of that homebrew package which is installed:

```
$ brew info tbb
tbb: stable 4.3-20150611 (bottled)
A rich and complete approach to parallelism in C++
https://www.threadingbuildingblocks.org/
/usr/local/Cellar/tbb/4.3-20141023 (104 files, 1.9M)
  Poured from bottle
/usr/local/Cellar/tbb/4.3-20150424 (106 files, 1.9M) *
  Poured from bottle
From: https://github.com/Homebrew/homebrew/blob/master/Library/Formula/tbb.rb
==> Options
--c++11
	Build using C++11 mode
```

Perhaps opencv was a red herring. Have you tried `brew install --upgrade tbb`?

zoercai, 2015-07-31:
Yep my output for both location & homebrew version are similar to yours. I tried `brew unlink tbb`, `brew install tbb`, as well as `brew link tbb`, but it made no difference :disappointed:

Can you think of anything else that might be causing the problem?

mikepurvis, 2015-08-06:
I'm afraid I'm out of ideas here. If you're really committed to this, you could try the nuclear option of removing your homebrew installation and starting over from scratch.

However, I understand that @wjwwood and @j-rivero are in the process of setting up a CI job to check the process for regressions on a daily basis— it may be worth waiting until that is complete so you have the confidence of knowing that at least it's working on a clean-room environment (though I have received feedback from others that this procedure worked for them, and it did also work for me a few weeks ago on a clean Parallels OS X VM).

zoercai, 2015-08-06:
I've tried removing homebrew and doing a clean install of ROS by following both your instructions in README.md as well as the ones [here](http://www.tatsch-it.de/how-to-install-ros-indigo-on-os-x-yosemite/), unfortunately neither worked. Are the instructions missing anything that should be removed for a clean install?
If not, I will wait for the CI. Where can I check its progress?

mikepurvis, 2015-08-13:
@zoercai Not sure what the timeline is on the CI being set up, but I ran the install again last night on a clean Yosemite VM via Parallels, and got a successful build of desktop_full for indigo.

zoercai, 2015-08-13:
Ok I will try a clean install again.

mockbot, 2015-08-18:
Sorry, I have the same problem at my MacBookPro OSX 10.10.5.
A complete fresh install without ros-history-tries works perfectly at my old iMac OSX 10.10.5.

There must be any drawback with already installed homebrew from the past.

mikepurvis, 2015-08-18:
It sounds like @zoercai has tried with a completely clean Homebrew/pip installation, so that suggests that the problem is with something installed _outside_ of Homebrew. The main area where I've found problems there is with Frameworks that overlap Homebrew packages, but at least in Zoe's case, there [didn't seem to be any installed](https://github.com/mikepurvis/ros-install-osx/issues/4#issuecomment-125781785).

So there's something pre-existing on your machines that's messing up the process at this particular point, but it's not at all clear to me what it is.

mikepurvis, 2015-08-18:
This is a long shot, but one thing I noticed is that @zoercai is using the `c++` that's inside `Xcode.app`, whereas I am using the one from CLT. I have:

    /Library/Developer/CommandLineTools/usr/bin/c++  -O3 -DNDEBUG -dynamiclib -Wl,-headerpad_max_install_names 

Versus:

    /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++  -O3 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk -dynamiclib -Wl,-headerpad_max_install_names

@mockbot, any chance you're using the Xcode compiler as well?

kitizz, 2015-09-01:
I'm getting the same error - I'm also using the XCode C++:

```/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++  -O3 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk -dynamiclib -Wl,-headerpad_max_install_names```

jbartels2006, 2015-09-01:
I tried using the script on a fresh install of Yosemite with only XCode installed and received the same error described by this issue. After reading this I then installed CLT and the error went away and the script finished successfully.

mikepurvis, 2015-09-01:
Interesting, thanks very much for trying this so thoroughly. I have little experience with the full XCode bundle, unfortunately, but is it possible to install the command-line tools at the same time?

In any case, the real fix is correcting the FindTBB module. I've raised a ticket to that effect here:

https://github.com/ros/cmake_modules/issues/37

closed 2015-09-01 by mikepurvis

kitizz, 2015-09-02:
Sweet, did as @jbartels2006 said. Works as a quick fix for now. For completeness, I had to do a little more after installing CLT with:

```
$ xcode-select --install
<follow prompts>
```

Afterwards, when I entered this I got the following (it should be /Library/Developer/CommandLineTools):
```
$ xcode-select -p
/Applications/Xcode.app/Contents/Developer

$ gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
...
```

I needed to run:
```
sudo xcode-select -s /Library/Developer/CommandLineTools
```

I reran the install script and everything seems dandy.

spmaniato, 2016-01-29:
For future reference, after switching to the `CommandLineTools` as described above, you don't have to rerun the _entire_ `install` script. I believe that the following is enough:
```bash
rm -rf /opt/ros/indigo/*
rm -rf build/ devel/ # Assuming your working dir is the catkin workspace
catkin build \ 
  ... # See actual script for the 4-line-long command
```

It just saves a couple of minutes since the `install` script does a bunch of other things before it gets to build / install.

PS. Also, you can probably "`make clean`" only the two problematic packages. Haven't tested that.