mikepurvis/ros-install-osx #9: rosdep keys cannot be resolved to system dependencies

tingelst, 2015-08-14:
I get the following error messages during installation. The system is a late 2013 MacBook Pro 15 running Yosemite. I removed all traces of homebrew and pip before starting. Do you have any solution for this?

```
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
collada_urdf: No definition of [pcre-dev] for OS [osx]
urdf: No definition of [pcre-dev] for OS [osx]
stage_ros: No definition of [fluid] for OS [osx]
stage: No definition of [libfltk-dev] for OS [osx]
```

mikepurvis, 2015-08-14:
Those packages are definitely part of rosdep/osx/homebrew:

https://github.com/ros/rosdistro/blob/b8242094ec43dfa7c146e22496e60ed18f97be71/rosdep/osx-homebrew.yaml#L461
https://github.com/ros/rosdistro/blob/b8242094ec43dfa7c146e22496e60ed18f97be71/rosdep/osx-homebrew.yaml#L94
https://github.com/ros/rosdistro/blob/b8242094ec43dfa7c146e22496e60ed18f97be71/rosdep/osx-homebrew.yaml#L178

Maybe you got a bad rosdep clone or something? Consider:

```
rm -rf /etc/ros/rosdep
sudo rosdep init
rosdep update
```

And then try the rosdep line again:

    rosdep install --from-paths src --ignore-src --rosdistro ${ROS_DISTRO} -y --as-root pip:no --skip-keys="gazebo"

If it works correctly, create the install dir and run the build manually to avoid re-downloading everything:

https://github.com/mikepurvis/ros-install-osx/blob/6472e866699ced3522e9876cd2e0749fea658ba6/install#L150

tingelst, 2015-08-15:
That was it. I had a non-empty `/etc/ros/rosdep` directory from a failed installation attempt some months ago. The rest of the installation process was successful after this fix.

Thanks for the feedback! I guess this one can be closed now.

closed 2015-08-16 by mikepurvis

ihamieh, 2019-07-09:
I'm having a similar problem but i'm not sure how to solve it! 
I'm trying to install VLP16 to my ubuntu 16 with ROS kinetic but I also have Lego Loam installed and when i try the following, I get the error message! 

~/catkin_ws$ rosdep install --from-paths src --ignore-src --rosdistro=${ROS_DISTRO} -y
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
lego_loam: Cannot locate rosdep definition for [gtsam]

arm3nyn, 2020-05-25:
> I'm having a similar problem but i'm not sure how to solve it!
> I'm trying to install VLP16 to my ubuntu 16 with ROS kinetic but I also have Lego Loam installed and when i try the following, I get [...]

same here