mikepurvis/ros-install-osx #10: osx10.10.4 `pcl_ros` + brewed python trouble and solution

kpykc, 2015-08-23:
I had to add this to compile `pcl_ros`, though I'm not sure that it's not quirk of my system. 

I'm building ros packages  with: 

`-DPYTHON_LIBRARY=$(python -c "import sys; print sys.prefix")/lib/libpython2.7.dylib   -DPYTHON_INCLUDE_DIR=$(python -c "import sys; print sys.prefix")/include/python2.7`

And it produces:

 `-DPYTHON_LIBRARY=/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib -DPYTHON_INCLUDE_DIR=/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/include/python2.7`

So, I'm unaware for now why `pcl_ros` tries to link against `/usr/local/Cellar/python/2.7.9/Frameworks/Python.framework/Versions/2.7/include/python2.7`

To overcome this I used the same kind of code as already used for `Qt`:

~~~{.bash}
  if [ -d /usr/local/Cellar/python/2.7.10_2]; then
    pushd /usr/local/Cellar/python/
    if [ ! -d "2.7.9" ]; then
      ln -s 2.7.10_2 2.7.9
    fi
    popd
  fi
~~~

~~~{.bash}
0/~> sw_vers 
ProductName:	Mac OS X
ProductVersion:	10.10.4
BuildVersion:	14E46
0/~> brew config
HOMEBREW_VERSION: 0.9.5
ORIGIN: https://github.com/Homebrew/homebrew
HEAD: 1ed13e5711c1b4c1174bf5d7cdea35505efb397d
Last commit: 79 minutes ago
HOMEBREW_PREFIX: /usr/local
HOMEBREW_CELLAR: /usr/local/Cellar
HOMEBREW_BOTTLE_DOMAIN: https://homebrew.bintray.com
CPU: quad-core 64-bit ivybridge
OS X: 10.10.4-x86_64
Xcode: N/A
CLT: 6.4.0.0.1.1435007323
Clang: 6.1 build 602
X11: 2.7.7 => /opt/X11
System Ruby: 2.0.0-p481
Perl: /usr/bin/perl
Python: /usr/local/bin/python => /usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/bin/python2.7
Ruby: /usr/bin/ruby
Java: 1.6.0_65
~~~

mikepurvis, 2015-08-31:
Yuck. Looks like PCL ships its own find code for Python:

https://github.com/PointCloudLibrary/pcl/blob/master/cmake/pcl_find_python.cmake

I'll add your suggestion as a temporary fix when I've had a chance to test it on a clean VM, but please consider filing a bug with PCL about this issue.

kpykc, 2015-09-06:
I can try to fill a bug for PCL, but it's not completely clear for me how should they act on this issue. Do they need to change logic how they find python version?

mikepurvis, 2015-10-08:
Hmm. I did a full build just now, with Python 2.7.10_2, and didn't have this issue. OTOH:

```
$ ls /usr/local/Cellar/python
2.7.10		2.7.10_2	2.7.9
```

So it's not clear that this wouldn't manifest on a system with *only* the newest Python installed. Doing a build on my Parallels cleanroom is an all-day affair, unfortunately, as it's extremely slow. But I'll try it at some point. :)

grafoteka, 2015-11-05:
Hi @mikepurvis I've made a clear installation of Yosemite and try your script, but it fails compiling ROS with this error

CMake Error at /usr/local/share/pcl-1.8/PCLConfig.cmake:46 (message):
  simulation is required but glew was not found
Call Stack (most recent call first):
  /usr/local/share/pcl-1.8/PCLConfig.cmake:645 (pcl_report_not_found)
  /usr/local/share/pcl-1.8/PCLConfig.cmake:808 (find_external_library)
  CMakeLists.txt:8 (find_package)

I have tried this fix http://stackoverflow.com/questions/26311715/solving-a-difficult-compilation-issue

But how can I only launch the part of compiling ROS and not all the script?

Thanks!

kpykc, 2015-11-30:
@mikepurvis sorry for a (late)^2 reply, I just decided to do a `brew upgrade`, so all the bugs get out from the darkness again :/ (as usual, btw that is the reason I'm doing this rarely).

The issue is still persist for me.

For comparison:

```bash
0/indigo_desktop_ws> ls /usr/local/Cellar/python
2.7.10_2/
```
Unfortunately, I don't have such a _clean-room_ :), as my disk space is really precious (only 256Gb)

Though, this issue may be related to the fact that my indigo workspace is screwed-up already. I have to check that versions of packages are up to date.

mikepurvis, 2015-11-30:
Others are reporting problems with `pcl_ros`, too, so it's definitely not just you.

mikepurvis, 2015-12-09:
Okay, so I just got a clean build of Yosemite+Indigo on my main machine, starting from a fresh Homebrew install.

```
$ ls /usr/local/Cellar/python
2.7.11
$ ls /usr/local/Cellar/pcl
HEAD
```

Looks like PCL is now set to install from head via the rosdep entry: https://github.com/ros/rosdistro/blob/22e12c568e3796e8b93a257fdaf28cc8dbc24341/rosdep/osx-homebrew.yaml#L250

Once the next PCL is released, this should be switched to that version, obviously.

mikepurvis, 2016-06-17:
Should be fixed now by PCL 1.8.

closed 2016-06-17 by mikepurvis