mikepurvis/ros-install-osx #11: pip uninstall on El Capitan

hawesie, 2015-10-02:
The command
```pip freeze | xargs sudo pip uninstall -y```
no longer works on El Capitan as the new security layer prevents some files being deleted which pip owns (or thinks it owns). It fails on the first system dependency, which is altgraph on my machine:

(output, for future googlers)
```
Uninstalling altgraph-0.10.2:
Exception:
Traceback (most recent call last):
  File "/Library/Python/2.7/site-packages/pip/basecommand.py", line 211, in main
    status = self.run(options, args)
  File "/Library/Python/2.7/site-packages/pip/commands/uninstall.py", line 76, in run
    requirement_set.uninstall(auto_confirm=options.yes)
  File "/Library/Python/2.7/site-packages/pip/req/req_set.py", line 305, in uninstall
    req.uninstall(auto_confirm=auto_confirm)
  File "/Library/Python/2.7/site-packages/pip/req/req_install.py", line 716, in uninstall
    paths_to_remove.remove(auto_confirm)
  File "/Library/Python/2.7/site-packages/pip/req/req_uninstall.py", line 125, in remove
    renames(path, new_path)
  File "/Library/Python/2.7/site-packages/pip/utils/__init__.py", line 315, in renames
    shutil.move(old, new)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 299, in move
    copytree(src, real_dst, symlinks=True)
  File "/System/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 208, in copytree
    raise Error, errors
```

hawesie, 2015-10-02:
The following removes those installs which can be done by the user

```
#!/bin/bash
for i in $( pip freeze ); do
    sudo pip uninstall -y $i
done
```

closed 2015-10-07 by mikepurvis

spmaniato, 2016-01-28:
I found this issue after asking on SO. In case it helps anyone in the future, here's my (related but not the same) question / issue: http://stackoverflow.com/questions/35049907/cannot-sudo-pip-uninstall-operation-not-permitted-tmp-in-os-x-el-capitan