mikepurvis/ros-install-osx #12: El Capitan support

mbeentjes, 2015-10-06:
Are there any plans on supporting El Capitan?

mikepurvis, 2015-10-06:
Absolutely! Per #11, it sounds like it should work, especially on a clean El Cap install. Please give it a try and report back.

hawesie, 2015-10-13:
FLTK currently doesn't build on El Capitan: https://github.com/Homebrew/homebrew/issues/41581

mikepurvis, 2015-10-13:
Looks like this PR adds head/devel support for FLTK: https://github.com/Homebrew/homebrew/pull/44806

If you're really impatient, this should get you a working FLTK:

    brew reinstall --devel https://raw.githubusercontent.com/dpo/homebrew/ec46018128dde5bf466b013a6c7086d0880930a3/Library/Formula/fltk.rb

Feel free to test it out and PR in an update here to install FLTK upfront, similar to how some other things are taken care of.

jmtatsch, 2015-10-14:
Here are some more issues with El Capitan:

* pcl --HEAD is broken https://github.com/Homebrew/homebrew-science/issues/2593
* ogre1.9 is badly broken https://github.com/ros-visualization/rviz/issues/782#issuecomment-147838712

mikepurvis, 2015-10-14:
We shouldn't need pcl --HEAD— I'm running PCL 1.7.2 from the bottle without a problem (per https://github.com/Homebrew/homebrew-science/issues/2852).

I also don't recommend Ogre 1.9 and never have— these instructions have only ever recommended Ogre 1.7 in order to have working rviz and gazebo simultaneously.

hawesie, 2015-10-14:
I have an El Capitan install running with this script as at  #15

anlif, 2015-10-15:
I'm having some issues with linking when building the ROS source. Specifically with compressed_image_transport and pcl_ros. For compressed_image_transport I get:
```
ld: library not found for -ltbb
```

But with `brew install tbb` I get `Warning: tbb-4.3-20150611 already installed`. Similar issue with pcl. Any idea why the linker can't find the installed libraries? I just upgraded from Yosemite to El Capitan.

mikepurvis, 2015-10-15:
@anlif Which toolchain are you using. Per #4, there is a known issue with tbb in XCode (vs. the command-line tools).

anlif, 2015-10-16:
@mikepurvis Thanks for the tip Mike. For some reason I wasn't able to locate that issue myself. I had the full XCode toolkit, but currently testing with the command-line tools. I'm new to OSX so I didn't even know that it existed.

jriesen, 2015-10-19:
One issue I've run across is that DYLD_LIBRARY_PATH is apparently no longer exported to subshells in El Capitan unless you disable SIP.  (Apparently, this has been a hassle for an assortment of development projects out there; just Google for "DYLD_LIBRARY_PATH El Capitan".)  This means I run into frustrating problems like:

    $ rosrun turtlesim turtlesim_node 
    dyld: Library not loaded: librospack.dylib
      Referenced from: /opt/ros/indigo/bin/rospack
      Reason: image not found
    find: ftsopen: No such file or directory
    [rosrun] Couldn't find executable named turtlesim_node below 
    find: ftsopen: No such file or directory
    $ echo $DYLD_LIBRARY_PATH
    /Users/jriesen/Source/catkin_ws/devel/lib:/opt/ros/indigo/lib
    $ ls /opt/ros/indigo/lib/librospack.dylib
    /opt/ros/indigo/lib/librospack.dylib

I know this is not an issue with your script (which runs brilliantly, by the way, thank you so much!), just wanted to mention it's an issue with ROS on El Capitan.

As far as I can tell, this is *only* an issue with `rosrun` -- it's the only ROS utility in /opt/ros/indigo/bin/ that's a bash script, everything else is Python or C++.

My quick-and-dirty fix for this (for anyone else having trouble) was to add `export DYLD_LIBRARY_PATH=$DYLD_LIBRARY_PATH:/opt/ros/indigo/lib` to the start of `/opt/ros/indigo/bin/rosrun`, which resolves the issue.

mikepurvis, 2015-10-19:
@jriesen Sounds like something which should be [addressed upstream](https://github.com/ros/ros/blob/jade-devel/tools/rosbash/scripts/rosrun)— maybe create an issue on that repo?

For now perhaps there's a way to detect that SIP is active, and supply a warning to the user. Would you be willing to investigate that and supply a pull request here for such a thing?

moriarty, 2015-10-28:
Hi, thanks for the repository. I just got everything to build on El Capitan.

I had some issues, I wanted to document them here:

1. -ltbb errors: needed to accept the terms and conditions for the command line tools  https://github.com/mikepurvis/ros-install-osx/issues/4

2. A few like this one: 

        [ 96%] Linking CXX shared library
        /Users/alex/ros_ws/indigo_desktop_full_ws/devel/lib/libpcl_ros_segmentation.dylib
        ld: library not found for -lpcl_segmentation

    and also: 

        ld: library not found for -lpcl_common
        ld: library not found for -lpcl_io

    The fix was to find and replace all of the following in the file: 
    src/perception_pcl/pcl_ros/CMakeLists.txt 

    ```pcl_segmentation``` ->  ```${PCL_SEGMENTATION_LIBRARIES}```
    ```pcl_io``` -> ```${PCL_SEGMENTATION_LIBRARIES}```
    ```pcl_common``` -> ```${PCL_COMMON_LIBRARIES}```

3. I had problems with ros_pcl and GLEW.
    ```ld: framework not found GLEW```
    This fixed it: https://github.com/PointCloudLibrary/pcl/issues/1289#issuecomment-149364820

moriarty, 2015-10-28:
I also needed the DYLD_LIBRARY_PATH fix suggested by @jriesen to get ```rosrun``` working

jihoonl, 2015-10-29:
I got turtlesim working in el capitan with the following fixes. Thanks to @moriarty 

1. pcl_ros glew not found error are fixed by 
* replaciing [pcl_ros 1.2.7](https://github.com/ros-perception/perception_pcl/tree/1.2.7) version with [pcl_ros indigo-devel](https://github.com/ros-perception/perception_pcl/tree/indigo-devel) which includes https://github.com/ros-perception/perception_pcl/commit/b21adf3b0bf290752e82d502392c2f81382c8508
* and https://github.com/PointCloudLibrary/pcl/issues/1289#issuecomment-149364820 fixes `ld: framework not found GLEW`

2. DYLD_LIBRARY_PATH fixed `rosrun`. Thanks to @jriesen  and @moriarty

jihoonl, 2015-10-30:
Quite funny. I was testing the installation with another mac machine. It turned out that it required two pass compilation to compile `pcl_ros`. I have tested with *macbook pro mid 2010* and *iMac Late 2012* 

**First pass**
It gets 
```
-- Could NOT find openni (missing:  OPENNI_LIBRARY OPENNI_INCLUDE_DIRS) 
** WARNING ** apps features related to openni will be disabled
CMake Error at /usr/local/share/pcl-1.8/PCLConfig.cmake:46 (message):
  simulation is required but glew was not found
Call Stack (most recent call first):
  /usr/local/share/pcl-1.8/PCLConfig.cmake:650 (pcl_report_not_found)
  /usr/local/share/pcl-1.8/PCLConfig.cmake:813 (find_external_library)
  CMakeLists.txt:8 (find_package)

-- Configuring incomplete, errors occurred!
See also "/Users/jihoonl/research/ros/opt/indigo/indigo_desktop_full_ws/build/pcl_ros/CMakeFiles/CMakeOutput.log".
[pcl_ros] <== '/Users/jihoonl/research/ros/opt/indigo/indigo_desktop_full_ws/build/pcl_ros/build_env.sh /usr/local/bin/cmake /Users/jihoonl/research/ros/opt/indigo/indigo_desktop_full_ws/src/perception_pcl/pcl_ros -DCATKIN_DEVEL_PREFIX=/Users/jihoonl/research/ros/opt/indigo/indigo_desktop_full_ws/devel -DCMAKE_INSTALL_PREFIX=/opt/ros/indigo -DCMAKE_BUILD_TYPE=Release -DPYTHON_LIBRARY=/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib -DPYTHON_INCLUDE_DIR=/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/include/python2.7' failed with return code '1'
                                                                                                                                                       
Failed   <== pcl_ros                          [ 5.4 seconds ]                                                                                         
```

This is fixed by changing GLEW_INCLUDE_DIR in PCLConfig.cmake as suggested in http://stackoverflow.com/questions/26311715/solving-a-difficult-compilation-issue#comment52777960_28347643

**Then** it gets 
```
ld: framework not found GLEW
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [/Users/jihoonl/research/ros/opt/indigo/indigo_desktop_full_ws/devel/lib/pcl_ros/convert_pcd_to_image] Error 1
make[1]: *** [CMakeFiles/convert_pcd_to_image.dir/all] Error 2
make: *** [all] Error 2
[pcl_ros] <== '/Users/jihoonl/research/ros/opt/indigo/indigo_desktop_full_ws/build/pcl_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' failed with return code '2'
                                                                                                                                                       
Failed   <== pcl_ros                          [ 38.5 seconds ]                                                                                        
```
This gets fixed with https://github.com/PointCloudLibrary/pcl/issues/1289#issuecomment-149364820.

grafoteka, 2015-11-04:
Hi @mikepurvis which version of ROS have you installed? Jade or Indigo? Works right?

Thanks

mikepurvis, 2015-11-04:
I'm still running Yosemite + Indigo. Would like to do a El Cap install in Parallels at some point, but it's an extremely slow process— virtualization of OS X itself is awful.

gareth-cross, 2015-11-10:
The DYLD_LIBRARY_PATH issue also affected `catkin_make run_tests` for me. This was because the build.make file generated for run_tests executes `workspace/build/catkin_generated/env_cached.sh`, which switches to the sh shell, where DYLD_LIBRARY_PATH was not defined to include `/opt/ros/jade/lib`. 

A quick workaround is simply to run the gtest binary directly, the path to which is spat out by ROS' run_tests.py script anyways. Example output of the run_tests script:
```
run_tests.py: execute commands
  /Users/gareth/Documents/workspace/devel/lib/foo/gtest_foo --gtest_output=xml:/Users/gareth/Documents/workspace/build/test_results/foo/gtest-gtest_foo.xml
dyld: Library not loaded: libroscpp.dylib
  Referenced from: /Users/gareth/Documents/vworkspace/devel/lib/foo/gtest_foo
  Reason: image not found
```
Copy-paste the command and the tests run for package `foo` w/ no issues.

l0g1x, 2015-11-16:
`brew install homebrew/science/pcl -HEAD`

didnt work for me, but `brew install homebrew/science/pcl` did (without -HEAD, just as @mikepurvis suggested he used pcl 1.7)

    Krystians-MacBook-Pro:~ krystian$ brew install homebrew/science/pcl
    ==> Installing pcl from homebrew/homebrew-science
    ==> Downloading https://homebrew.bintray.com/bottles-science/pcl-1.7.2.el_capitan.bottle.2.tar.gz
    ######################################################################## 100.0%
    ==> Pouring pcl-1.7.2.el_capitan.bottle.2.tar.gz
    ==> Caveats
    .app bundles were installed.
    Run `brew linkapps pcl` to symlink these to /Applications.
    ==> Summary
    🍺  /usr/local/Cellar/pcl/1.7.2: 1070 files, 122M
    Krystians-MacBook-Pro:~ krystian$

weichnn, 2015-11-20:
I install ros indigo in EL Cap. but I get this problem. Suggestion of @jriesen cannot deal with it...Am I do anything wrong?..
```
$ rosrun turtlesim turtlesim_node
dyld: Library not loaded: librospack.dylib
  Referenced from: /Users/weichend/ros_catkin_ws/install_isolated/bin/rospack
  Reason: image not found
find: ftsopen: No such file or directory
[rosrun] Couldn't find executable named turtlesim_node below
find: ftsopen: No such file or directory
$ echo $DYLD_LIBRARY_PATH
/Users/weichend/ros_catkin_ws/install_isolated/lib
$ ls /opt/ros/indigo/lib/librospack.dylib
ls: /opt/ros/indigo/lib/librospack.dylib: No such file or directory
```

l0g1x, 2015-11-20:
`sudo vim /opt/ros/indigo/bin/rosrun` and place `DYLD_LIBRARY_PATH=$DYLD_LIBRARY_PATH:/opt/ros/indigo/lib` in there

weichnn, 2015-11-20:
thanks to @l0g1x .
One thing confuse me...I found there are no /opt/ros folder.
so I create it and follow your suggestion, but it does not work...

spaghetti-, 2015-12-10:
Linking is horribly broken on my install on el cap for some reason.

Lots of library not found errors (for example, for rospack)

I fixed a few with `install_name_tool` but what could be the cause of this issue?

spaghetti-, 2015-12-10:
Right, SIP disabling DYLD_LIBRARY_PATH. Above comment and another one above fixes it.

mikepurvis, 2015-12-10:
I'd really appreciate a PR to detect the SIP issue and prompt the user for how to resolve it.

spmaniato, 2016-01-29:
> Absolutely! Per #11, it sounds like it should work, especially on a clean El Cap install. Please give it a try and report back.

FYI, I'm up and running with ROS on (a very unclean) OS X El Capitan :tada: 
Thanks Mike and everyone else for the very helpful hints and fixes :smiley: 

For future reference, I ran into most of the problems described here and in related issues (`fltk`, `-ltbb` not found, `librospack.dylib`, etc.) The hints/hacks above solved all problems I encountered. Cheers.

mikepurvis, 2016-01-29:
@spmaniato Glad you're in action now. I'd be delighted to have anything that's relevant contributed back to the main repo.

spmaniato, 2016-01-29:
> I'd be delighted to have anything that's relevant contributed back to the main repo.

@mikepurvis Good point. I'll compile all of my lessons learned from the installation (plus the tips above) in the README's Troubleshooting section and open a pull request.

ahundt, 2016-02-03:
I've been able to install on el capitan using the instructions as a guide, though I modified some of the steps because my installation wasn't clean. 

One particular thing is that it wouldn't pass the part that checks installed python packages even after I uninstalled all of them. Could be something I did wrong, however because I don't use python that often.

spaghetti-, 2016-02-03:
Just slightly confused but what do you mean by it wouldn't part the part that checks installed python packages after you installed them?

spmaniato, 2016-02-03:
I think @ahundt means he had to remove the `exit 1` (see below). I had the same problem because that directory had some Python packages that were not installed via pip.

```bash
# Check for pip packages in the system Python directory.
  if [ $(ls /Library/Python/2.7/site-packages/ | wc -l) -gt 1 ]; then
    echo "These instructions are about to install Python from Homebrew. However, there are already"
    echo "pip packages installed against the system python, in the following path:"
    echo
    echo "    /Library/Python/2.7/site-packages/"
    echo
    echo "To proceed, please uninstall these packages:"
    echo
    echo "    for i in $( pip freeze ); do sudo pip uninstall -y $i; done"
    exit 1
  fi
```

ahundt, 2016-02-03:
@spmaniato is correct

mikepurvis, 2016-02-03:
Were those installed via `sudo python setup.py install`? How else would they get in there?

The fundamental issue is that if there's something there that's compiled something against system python, pip will avoid "reinstalling" the same package, and you'll end up with weird ABI problems at runtime. We can simply remove the `exit 1`, but I'd like to understand what's in there and how it got there.

spmaniato, 2016-02-03:
> but I'd like to understand what's in there and how it got there.

For me, the non-pip stuff were things like Aldebaran Robotics's `NAOqi` Python package and a Python package called `Polygon` (probably installed from source using sudo, or maybe MacPorts). There's also a `gurobipy` package, which I'm guessing was put there by the Gurobi Optimizer installer.

Basically these are rather exotic things that I don't think will cause any problems (to me).

mikepurvis, 2016-02-04:
That's fair. For my understanding, though, for them to have ended up in `/Library` means that they were indeed installed prior to brewed python? libqi is a perfect example of the kind of package where that's going to be a problem, as it's compiled bindings.

angerhang, 2016-02-16:
I would also vote for the removal of `exit 1` since the error message simply doesn't make sense 
```
These instructions are about to install Python from Homebrew. However, there are already
pip packages installed against the system python, in the following path:

    /Library/Python/2.7/site-packages/

To proceed, please uninstall these packages:

    for i in six==1.10.0; do sudo pip uninstall -y ; done
```
Or change the error message to something better :) Otherwise it is more difficult for those who are installing via `curl`.

mikepurvis, 2016-02-16:
You have been heard: c344dd77b79b64648f8efe5e149102927f06ff9c

angerhang, 2016-02-17:
Ok almost there. I don't know why the executable is not found when I tried to run turtlesim:
```
Hangs-MBP:turtlesim Hang$ rosrun turtlesim turtlesim_node
[rosrun] Couldn't find executable named turtlesim_node below /Users/Hang/ros_catkin_ws/src/ros_tutorials/turtlesim
```
I also tried `rosmake turtlesim`which didn't seem to work.

spmaniato, 2016-02-17:
@angerhang Have you sourced the ROS installation? The script echoes that, but doesn't do it for you:
```bash
echo "Installation successful, please source the ROS workspace:"
echo
echo "  source ${ROS_INSTALL_DIR}/setup.bash" # /opt/ros/indigo/setup.bash
echo
```

angerhang, 2016-02-17:
Actually I did `source` but it still didn't work.

I just solved it by uninstalling everything and `./install` again. I was mixing the `install` script and the tutorial from the ros website which apparently should be the direct cause.

But now everything works like magic. 
Thank you @spaghetti- @mikepurvis  : )

abi93k, 2016-02-25:
DYLD bug
I get the following error when I use **roslaunch**

```
ImportError: dlopen(/Users/Abhishek/ros_catkin_ws/install_isolated/lib/python2.7/site-packages/tf/_tf.so, 2): Library not loaded: libtf2_ros.dylib
  Referenced from: /Users/Abhishek/ros_catkin_ws/install_isolated/lib/python2.7/site-packages/tf/_tf.so
  Reason: image not found
```

Any fixes ?

ramamoorthyluxman, 2016-02-29:
I think the fix given by jriesen should solve your bug.. 

add, export DYLD_LIBRARY_PATH=$DYLD_LIBRARY_PATH:/opt/ros/indigo/lib, in the beginning of this file - /opt/ros/indigo/bin/rosrun

moriarty, 2016-03-17:
I just got it working on El Capitan + Indigo.

First, I had a protobuf + gazebo issues, but I think it is rare #33 

All of the issues were then pcl_ros issues:

1. Replace pcl_ros 1.2.7 with 1.3.0, which is for jade. I just modified the .rosinstall file. Its for jade but only a few commits ahead of the one needed (https://github.com/ros-perception/perception_pcl/commit/b21adf3b0bf290752e82d502392c2f81382c8508).

    ```
    - tar:
        local-name: perception_pcl/pcl_ros
        uri: https://github.com/ros-gbp/perception_pcl-release/archive/release/jade/pcl_ros/1.3.0-0.tar.gz
        version: perception_pcl-release-release-jade-pcl_ros-1.3.0-0
    ```
2. There was a pcl_ros + python issue: https://github.com/mikepurvis/ros-install-osx/issues/10. pcl is looking for python in 2.7.10_2 and I had 2.7.11 replace, 2.7.11 with in the following command with the version ```ls -la``` shows. Thanks @kpykc

    ```
    pushd /usr/local/Cellar/python
    ls -la 
    ln -s 2.7.11 2.7.10_2
    popd
    ```
3. There was an Eigen3 issue: https://github.com/ros-perception/perception_pcl/issues/106#issuecomment-134948296
4. There was the usual GLEW issue: https://github.com/PointCloudLibrary/pcl/issues/1289#issuecomment-149364820
5. And there is the DYLD_LIBARY_PATH issue https://github.com/mikepurvis/ros-install-osx/issues/12#issuecomment-149336634

Thanks again @mikepurvis

hawesie, 2016-03-21:
On a recent El Capitan reinstall I ran into a problem where the map_server only produced all black maps. To fix this I had to install SDL_image from source with the compilation flag SDL_IMAGE_USE_COMMON_BACKEND set. I don't know if this is due to some quirk on my machine, or a more general fault.

luqiang21, 2016-03-28:
@jriesen Thank you so much. I can't use rosrun after updating OS. Follow your instruction, I solved the problem.

fdsa1860, 2016-04-15:
@weichnn I guess you installed separately in the folder ~/ros_catkin_ws/
 So you should add ~/ros_catkin_ws/install_isolated/lib/ to your path instead.

kerapansapi, 2016-04-29:
I try to install ROS on El Capitan by following intruction on ros.org, but error happen on resolving dependencies as follows:

$ rosdep install --from-paths src --ignore-src --rosdistro jade -y
executing command [sudo -H pip install -U nose]
Password:
Downloading/unpacking nose
  Running setup.py egg_info for package nose
    no previously-included directories found matching 'doc/.build'
Installing collected packages: nose
  Running setup.py install for nose
    error: could not create '/System/Library/Frameworks/Python.framework/Versions/2.7/man': Operation not permitted
    Complete output from command /usr/bin/python -c "import setuptools;__file__='/tmp/pip-build/nose/setup.py';exec(compile(open(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-FXoKP5-record/install-record.txt --single-version-externally-managed:
    running install
running build
running build_py
running install_lib
running install_data
creating /System/Library/Frameworks/Python.framework/Versions/2.7/man
error: could not create '/System/Library/Frameworks/Python.framework/Versions/2.7/man': Operation not permitted
----------------------------------------
Command /usr/bin/python -c "import setuptools;__file__='/tmp/pip-build/nose/setup.py';exec(compile(open(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /tmp/pip-FXoKP5-record/install-record.txt --single-version-externally-managed failed with error code 1 in /tmp/pip-build/nose
Storing complete log in /var/root/Library/Logs/pip.log
ERROR: the following rosdeps failed to install
  pip: command [sudo -H pip install -U nose] failed

Any fix ?

mikepurvis, 2016-04-29:
@kerapansapi The instructions in this repo are very different from the ones on ros.org; the problem that you are running to is exactly the kind of thing we try to mitigate, by staying away from the system installations of things, and sticking to Homebrew's python, non-sudo pip use, etc.

gunshi, 2016-05-02:
Hi, I'm facing the librospack.dylib not loaded issue, that many have mentioned here before.
The output of   

>  echo $DYLD_LIBRARY_PATH

     is

> /Users/gunshigupta/ros_catkin_ws/install_isolated/lib:/Users/gunshigupta/ros_catkin_ws/install_isolated/lib

So i have placed 

> DYLD_LIBRARY_PATH=$DYLD_LIBRARY_PATH:/Users/gunshigupta/ros_catkin_ws/install_isolated/lib 

in the bash file for rosrun placed in /Users/gunshigupta/ros_catkin_ws/install_isolated/bin.
However the same error persists, and the command that produces it is:

> rosrun libuvc_camera camera_node

(I removed the camera location information from the command because I wasn't sure I was providing the correct vendor and location ID)
Any help would be appreciated. Thanks.

gunshi, 2016-05-02:
Edit: I understand the libuvc_camera error, since the package needs to be installed by me and did not come with indigo, but I still wanted to ask about the IDs.

justinthomas, 2016-05-11:
Sorry. Please disregard that previous comment. Somehow a reference for a pull request in my repo somehow snuck in here.

OmenderSharma28, 2016-06-03:
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [lib/libopencv_videoio.3.1.0.dylib] Error 1
make[1]: *** [modules/videoio/CMakeFiles/opencv_videoio.dir/all] Error 2
make: *** [all] Error 2

error coming up during make . What should i do to fix it?

mikepurvis, 2016-06-17:
El Cap has a pretty good story now.

closed 2016-06-17 by mikepurvis

spaghetti-, 2016-06-18:
Just a heads up: the badge on the readme still says yosemite so if the install is fully supported by the script on El Cap we should change that.

vietvudanh, 2016-06-24:
@mikepurvis: Thank you so much for this, I have just struggled with install ROS by myself, this repo really saved me. 

However, there is one thing that I still wondering, and may sound silly but: How do you install packages on OSX? Is there any equivalent to `apt-get install` for Ubuntu?

spaghetti-, 2016-06-24:
@vietvudanh what do you mean? there are lots of packages for brew, check out brew.sh

vietvudanh, 2016-06-24:
@spaghetti- I mean ROS package for brew. I.e I'm trying to install this: [urg-node](http://wiki.ros.org/urg_node) for Indigo.

mikepurvis, 2016-06-24:
@vietvudanh You can build whatever you want from source, or if there are specific packages you'd like included in the initial workspace, you can add them to the `ROS_EXTRA_PACKAGES` env var.

vietvudanh, 2016-06-26:
@mikepurvis: Thank you, I am aware of that. I am just wondering if there are ways to install by brew.