mikepurvis/ros-install-osx #13: Build errors in indigo installation.

japzi, 2015-10-06:
OS X Yosemite, ver 10.10.5
brew --version : 0.9.5 (git revision d430d; last commit 2015-10-05)
pip --version : pip 7.1.2 from /usr/local/lib/python2.7/site-packages (python 2.7)

    [ 25%] Built target pcl_ros_gencfg
    Scanning dependencies of target pointcloud_to_pcd
    [ 26%] Building CXX object CMakeFiles/pointcloud_to_pcd.dir/tools/pointcloud_to_pcd.cpp.o
    [ 28%] Linking CXX executable /Users/anon/indigo_desktop_full_ws/devel/lib/pcl_ros/convert_pcd_to_image
    ld: library not found for -lvtkRenderingFreeTypeOpenGL
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    make[2]: *** [/Users/anon/indigo_desktop_full_ws/devel/lib/pcl_ros/convert_pcd_to_image] Error 1
    make[1]: *** [CMakeFiles/convert_pcd_to_image.dir/all] Error 2
    make[1]: *** Waiting for unfinished jobs....
    [ 30%] Linking CXX executable /Users/anon/indigo_desktop_full_ws/devel/lib/pcl_ros/pcd_to_pointcloud
    ld: library not found for -lvtkRenderingFreeTypeOpenGL
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    make[2]: *** [/Users/anon/indigo_desktop_full_ws/devel/lib/pcl_ros/pcd_to_pointcloud] Error 1
    make[1]: *** [CMakeFiles/pcd_to_pointcloud.dir/all] Error 2
    [ 31%] Linking CXX executable /Users/anon/indigo_desktop_full_ws/devel/lib/pcl_ros/convert_pointcloud_to_image
    ld: library not found for -lvtkRenderingFreeTypeOpenGL
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    make[2]: *** [/Users/anon/indigo_desktop_full_ws/devel/lib/pcl_ros/convert_pointcloud_to_image] Error 1
    make[1]: *** [CMakeFiles/convert_pointcloud_to_image.dir/all] Error 2
    [ 33%] Linking CXX executable /Users/anon/indigo_desktop_full_ws/devel/lib/pcl_ros/pointcloud_to_pcd
    ld: library not found for -lvtkRenderingFreeTypeOpenGL
    clang: error: linker command failed with exit code 1 (use -v to see invocation)
    make[2]: *** [/Users/anon/indigo_desktop_full_ws/devel/lib/pcl_ros/pointcloud_to_pcd] Error 1
    make[1]: *** [CMakeFiles/pointcloud_to_pcd.dir/all] Error 2
    make: *** [all] Error 2
    [pcl_ros] <== '/Users/anon/indigo_desktop_full_ws/build/pcl_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' failed with return code '2'

http://answers.ros.org/question/218682/mac-os-ros-build-errors-indigo/

mikepurvis, 2015-10-06:
Please search your lib dir for the missing library and compare the result with mine below:

```
$ ls /usr/local/lib/libvtkRenderingFreeTypeOpenGL*
/usr/local/lib/libvtkRenderingFreeTypeOpenGL-6.2.1.dylib
/usr/local/lib/libvtkRenderingFreeTypeOpenGL-6.2.dylib
/usr/local/lib/libvtkRenderingFreeTypeOpenGLPython27D-6.2.1.dylib
/usr/local/lib/libvtkRenderingFreeTypeOpenGLPython27D-6.2.dylib
```

Note that VTK was [updated to v6.3 just two weeks ago](https://github.com/Homebrew/homebrew-science/commits/master/vtk.rb), so this may be a regression. I've notified upstream here:

https://github.com/Homebrew/homebrew-science/pull/2765#issuecomment-145914735

japzi, 2015-10-06:
ls /usr/local/lib/libvtkRenderingFreeTypeOpenGL*
ls: /usr/local/lib/libvtkRenderingFreeTypeOpenGL*: No such file or directory

Yeah could be the reason. I don't know the impact, but I've removed -vtkRenderingFreeTypeOpenGL from all files and built went without a problem.

mikepurvis, 2015-10-06:
Huh. It's tricky to get a good handle on this, as it seems like all the cmake files which reference `vtkRenderingFreeTypeOpenGL` are all generated either during configuration or build (for example, VTKTargets.cmake)

Can you confirm that you're building the ROS workspace from scratch (eg, no leftover build or devel stuff from a previous attempt)?

gogobd, 2015-10-06:
Hi! Thanks for catching up on this! In the mean time I was able to get through. The solution that worked for me was using a source installation of vtk 6.2.0. In that source installation there were numerous places where I had to enable building libraries explicitly (ON/OFF in cmake .txt files). Since I found no documentation I had to walk the whole way and enabled the libraries the linker was missing one by one.
Then the whole procedure <https://github.com/mikepurvis/ros-install-osx/blob/master/install> went through - i just had to avoid steps that would overwrite my changes in pcl that pointed cmake to my source installation.

mikepurvis, 2015-10-07:
Hmm, sounds awful.

So just to confirm, both of you are independently seeing this occur on relatively clean Yosemite setups?

burrimi, 2015-10-07:
I had the same issue on a clean brew and solved it also by installing vtk 6.2. (OS X 10.10.5, Xcode7)

mikepurvis, 2015-10-07:
The issue is that PCL 1.7.2 needs to be rebottled for the new VTK. The short-term solution is to install pcl from source.

I've reported here: https://github.com/Homebrew/homebrew-science/issues/2852

closed 2015-10-07 by mikepurvis

soswow, 2015-11-12:
@mikepurvis I see they fixed the problem in Homebrew/homebrew-science#2852, does it mean you can revert this change?

commit 9546c256 (2015-10-07): Build PCL from source, fixes #13