mikepurvis/ros-install-osx #14: numpy brew vs pip

hawesie, 2015-10-14:
I'm installing on El Capitan with a fresh pip and brew installs. There appears to be a problem where `numpy` is installed both by brew and pip. This results in an error when rosdep runs:

```
~/indigo_desktop_full_ws ~
executing command [pip install -U matplotlib]
Collecting matplotlib
  Using cached matplotlib-1.4.3-cp27-none-macosx_10_6_intel.macosx_10_9_intel.macosx_10_9_x86_64.macosx_10_10_intel.macosx_10_10_x86_64.whl
Collecting numpy>=1.6 (from matplotlib)
  Using cached numpy-1.10.1-cp27-none-macosx_10_6_intel.macosx_10_9_intel.macosx_10_9_x86_64.macosx_10_10_intel.macosx_10_10_x86_64.whl
Requirement already up-to-date: pytz in /usr/local/lib/python2.7/site-packages (from matplotlib)
Collecting pyparsing!=2.0.0,>=1.5.6 (from matplotlib)
  Using cached pyparsing-2.0.3-py2.py3-none-any.whl
Requirement already up-to-date: python-dateutil in /usr/local/lib/python2.7/site-packages (from matplotlib)
Requirement already up-to-date: nose>=0.11.1 in /usr/local/Cellar/numpy/1.10.0/libexec/nose/lib/python2.7/site-packages (from matplotlib)
Requirement already up-to-date: six>=1.4 in /usr/local/lib/python2.7/site-packages (from matplotlib)
Collecting mock (from matplotlib)
  Using cached mock-1.3.0-py2.py3-none-any.whl
Collecting funcsigs (from mock->matplotlib)
  Using cached funcsigs-0.4-py2.py3-none-any.whl
Collecting pbr>=0.11 (from mock->matplotlib)
  Using cached pbr-1.8.1-py2.py3-none-any.whl
Installing collected packages: numpy, pyparsing, funcsigs, pbr, mock, matplotlib
  Found existing installation: numpy 1.10.0
    DEPRECATION: Uninstalling a distutils installed project (numpy) has been deprecated and will be removed in a future version. This is due to the fact that uninstalling a distutils project will only partially uninstall the project.
    Uninstalling numpy-1.10.0:
      Successfully uninstalled numpy-1.10.0
  Rolling back uninstall of numpy
Exception:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/pip/basecommand.py", line 211, in main
    status = self.run(options, args)
  File "/usr/local/lib/python2.7/site-packages/pip/commands/install.py", line 311, in run
    root=options.root_path,
  File "/usr/local/lib/python2.7/site-packages/pip/req/req_set.py", line 646, in install
    **kwargs
  File "/usr/local/lib/python2.7/site-packages/pip/req/req_install.py", line 803, in install
    self.move_wheel_files(self.source_dir, root=root)
  File "/usr/local/lib/python2.7/site-packages/pip/req/req_install.py", line 998, in move_wheel_files
    isolated=self.isolated,
  File "/usr/local/lib/python2.7/site-packages/pip/wheel.py", line 339, in move_wheel_files
    clobber(source, lib_dir, True)
  File "/usr/local/lib/python2.7/site-packages/pip/wheel.py", line 317, in clobber
    shutil.copyfile(srcfile, destfile)
  File "/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 83, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: '/usr/local/lib/python2.7/site-packages/numpy/__config__.py'
ERROR: the following rosdeps failed to install
  pip: command [pip install -U matplotlib] failed
```

Homebrew numpy is an opencv dependency for one of the selected options.

```
banks:~ nah$ brew info opencv
homebrew/science/opencv: stable 2.4.12, HEAD
Open source computer vision library
http://opencv.org/
/usr/local/Cellar/opencv/2.4.12 (225 files, 36M) *
  Built from source with: --with-qt --with-tbb
From: https://github.com/Homebrew/homebrew-science/blob/master/opencv.rb
==> Dependencies
Build: cmake ✔, pkg-config ✔
Required: jpeg ✔, libpng ✔, libtiff ✔
Recommended: eigen ✔, openexr ✔, homebrew/python/numpy ✔
```

The solution (assuming we need to opencv optional parts) appears to be changing

```
brew install opencv --with-qt --with-eigen --with-tbb
```

to
```
  pip install numpy
  brew install opencv --with-qt --with-eigen --with-tbb --without-numpy
```

I'll open a PR if I get through the install without related errors.

mikepurvis, 2015-10-14:
Hmm. Are we certain that getting numpy from pip is the right solution here, given that it's available bottled from homebrew? At present, [rosdep's osx key for numpy uses pip](https://github.com/ros/rosdistro/blob/ad462862cd11029f1477a81fc14204e584ce901a/rosdep/osx-homebrew.yaml#L534), but I wonder if that's where the change should be made?

@wjwwood, do you have any comment on whether OS X numpy should come from pip or homebrew?

hawesie, 2015-10-14:
I'm definitely not certain.

wjwwood, 2015-10-14:
The rosdep key is for pip because that's what you use if you use the system Python. That was the recommendation in the past and that is what I'm using currently. It might be better to get it from homebrew if your python is also from homebrew. But rosdep doesn't know where you get your Python from, and basically it causes the need for a new os identifier, something like `osx_with_brewed_python`, which can have different keys. Or we have to decided that having the system Python is no longer supported and therefore the rosdep key would be for homebrew not pip.

mikepurvis, 2015-10-19:
Hmm. Based on what's in the [homebrew-python tap](https://github.com/Homebrew/homebrew-python), if we wanted to go all-in on brewed python stuff, that would mean installing python, numpy, matplotlib, pillow, and scipy upfront, from homebrew, and then skipping those keys at rosdep time. Or would rosdep detect that they were present? Does brewed pip report numpy installed when it's brewed numpy? Can someone please try this out and report any pitfalls or whatever?

It seems like it could be a reasonable path, but @hawesie's solution also looks workable and an overall similar level of hackiness. Ultimately, the objective should be reducing the amount of manually-installed crap in favour of workable rosdep keys.

@wjwwood How strong is your commitment to system python?

wjwwood, 2015-10-19:
> @wjwwood How strong is your commitment to system python?

I don't really care as long as it works. In the past the brewed Python had lots of issues, but that may not be the case now.

PR #15 (2015-10-23): Use numpy from pip rather than brew. 
This fixes #14

This also includes the short-term fltk fix for #12
commit 9fdc746d (2015-10-23): Use numpy from pip rather than brew. This fixes #14
closed 2015-10-23 by mikepurvis