mikepurvis/ros-install-osx #17: Installed without pcl_ros

grafoteka, 2015-11-05:
I've installed ros indigo with your script, but I had to remove the package pcl_ros because it was always crashing the installation.

I will try to launch something with gazebo and tell how it works.

Thanks!

mikepurvis, 2015-11-05:
Can you clarify— was this crashing as in wouldn't build, or crashing as in crashes on running. Either way, please let us know a) which version of OS X this is, and b) what specific error messages you encountered during build or run (and if at runtime, what it was you ran to produce the error).

grafoteka, 2015-11-05:
Hi @mikepurvis thank for your answer.

This is the error that I get:
[[36mpcl_ros[0m] [1m[34m==>[0m '[1m/Users/jorge/indigo_desktop_full_ws/build/pcl_ros/build_env.sh /usr/local/bin/cmake /Users/jorge/indigo_desktop_full_ws/src/perception_pcl/pcl_ros -DCATKIN_DEVEL_PREFIX=/Users/jorge/indigo_desktop_full_ws/devel -DCMAKE_INSTALL_PREFIX=/opt/ros/indigo -DCMAKE_BUILD_TYPE=Release -DPYTHON_LIBRARY=/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib -DPYTHON_INCLUDE_DIR=/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/include/python2.7[0m' [30m[1min[0m '[1m/Users/jorge/indigo_desktop_full_ws/build/pcl_ros[0m'[0m
[36m-- [0mThe C compiler identification is AppleClang 7.0.0.7000176
[0m[36m-- [0mThe CXX compiler identification is AppleClang 7.0.0.7000176
[0m[36m-- [0mCheck for working C compiler:[33m /Library/Developer/CommandLineTools/usr/bin/cc
[0m[0m[36m-- [0mCheck for working C compiler:[33m /Library/Developer/CommandLineTools/usr/bin/cc -- works
[0m[0m[36m-- [0mDetecting C compiler ABI info
[0m[36m-- [0mDetecting C compiler ABI info - done
[0m[36m-- [0mDetecting C compile features
[0m[36m-- [0mDetecting C compile features - done
[0m[36m-- [0mCheck for working CXX compiler:[33m /Library/Developer/CommandLineTools/usr/bin/c++
[0m[0m[36m-- [0mCheck for working CXX compiler:[33m /Library/Developer/CommandLineTools/usr/bin/c++ -- works
[0m[0m[36m-- [0mDetecting CXX compiler ABI info
[0m[36m-- [0mDetecting CXX compiler ABI info - done
[0m[36m-- [0mDetecting CXX compile features
[0m[36m-- [0mDetecting CXX compile features - done
[0m[36m-- [0mBoost version:[33m 1.59.0
[0m[0m[36m-- [0mFound the following Boost libraries:[33m
[0m[0m[36m-- [0m  system
[0m[36m-- [0m  filesystem
[0m[36m-- [0m  thread
[0m[36m-- [0mFound PkgConfig:[33m /usr/local/bin/pkg-config (found version "0.29") 
[0m[0m[36m-- [0mchecking for module 'eigen3'
[0m[36m-- [0m  found eigen3, version 3.2.6
[0m[36m-- [0mFound Eigen:[33m /usr/local/Cellar/eigen/3.2.6/include/eigen3  
[0m[0m[36m-- [0mEigen found (include:[33m /usr/local/Cellar/eigen/3.2.6/include/eigen3)
[0m[0m[36m-- [0mFound eigen:[33m /usr/local/Cellar/eigen/3.2.6/include/eigen3  
[0m[0m[36m-- [0mBoost version:[33m 1.59.0
[0m[0m[36m-- [0mFound the following Boost libraries:[33m
[0m[0m[36m-- [0m  system
[0m[36m-- [0m  filesystem
[0m[36m-- [0m  thread
[0m[36m-- [0m  date_time
[0m[36m-- [0m  iostreams
[0m[36m-- [0m  serialization
[0m[36m-- [0mchecking for module 'libopenni'
[0m[36m-- [0m  package 'libopenni' not found
[0m[36m-- [0mCould NOT find openni (missing:[33m  OPENNI_LIBRARY OPENNI_INCLUDE_DIRS) 
[0m[0m** WARNING ** io features related to openni will be disabled
[0m[36m-- [0mchecking for module 'libopenni2'
[0m[36m-- [0m  package 'libopenni2' not found
[0m[36m-- [0mCould NOT find OpenNI2 (missing:[33m  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
[0m[0m** WARNING ** io features related to openni2 will be disabled
[0m[36m-- [0mCould NOT find ensenso (missing:[33m  ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR) 
[0m[0m** WARNING ** io features related to ensenso will be disabled
[0m[36m-- [0mCould NOT find DAVIDSDK (missing:[33m  DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
[0m[0m** WARNING ** io features related to davidSDK will be disabled
[0m[36m-- [0mCould NOT find DSSDK (missing:[33m  _DSSDK_LIBRARIES) 
[0m[0m** WARNING ** io features related to dssdk will be disabled
[0m** WARNING ** io features related to pcap will be disabled
[0m** WARNING ** io features related to png will be disabled
[0m[36m-- [0mFound libusb-1.0:[33m /usr/local/include  
[0m[0m[36m-- [0mchecking for module 'flann'
[0m[36m-- [0m  found flann, version 1.8.4
[0m[36m-- [0mFound Flann:[33m /usr/local/Cellar/flann/1.8.4_1/lib/libflann_cpp_s.a  
[0m[0m[36m-- [0mchecking for module 'libopenni'
[0m[36m-- [0m  package 'libopenni' not found
[0m[36m-- [0mCould NOT find openni (missing:[33m  OPENNI_LIBRARY OPENNI_INCLUDE_DIRS) 
[0m[0m** WARNING ** visualization features related to openni will be disabled
[0m[36m-- [0mchecking for module 'libopenni2'
[0m[36m-- [0m  package 'libopenni2' not found
[0m[36m-- [0mCould NOT find OpenNI2 (missing:[33m  OPENNI2_LIBRARY OPENNI2_INCLUDE_DIRS) 
[0m[0m** WARNING ** visualization features related to openni2 will be disabled
[0m[36m-- [0mCould NOT find ensenso (missing:[33m  ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR) 
[0m[0m** WARNING ** visualization features related to ensenso will be disabled
[0m[36m-- [0mCould NOT find DAVIDSDK (missing:[33m  DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
[0m[0m** WARNING ** visualization features related to davidSDK will be disabled
[0m[36m-- [0mCould NOT find DSSDK (missing:[33m  _DSSDK_LIBRARIES) 
[... 246 lines cut ...]
[35m[1mScanning dependencies of target pointcloud_to_pcd[0m
[ 26%] [32mBuilding CXX object CMakeFiles/pointcloud_to_pcd.dir/tools/pointcloud_to_pcd.cpp.o[0m
[ 28%] [32m[1mLinking CXX executable /Users/jorge/indigo_desktop_full_ws/devel/lib/pcl_ros/convert_pcd_to_image[0m
ld: framework not found GLEW
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [/Users/jorge/indigo_desktop_full_ws/devel/lib/pcl_ros/convert_pcd_to_image] Error 1
make[1]: *** [CMakeFiles/convert_pcd_to_image.dir/all] Error 2
[... cut ...]
[ 33%] [32m[1mLinking CXX executable /Users/jorge/indigo_desktop_full_ws/devel/lib/pcl_ros/pointcloud_to_pcd[0m
ld: framework not found GLEW
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [/Users/jorge/indigo_desktop_full_ws/devel/lib/pcl_ros/pointcloud_to_pcd] Error 1
make[1]: *** [CMakeFiles/pointcloud_to_pcd.dir/all] Error 2
make: *** [all] Error 2
[[36mpcl_ros[0m] [1m[31m<==[0m '[1m/Users/jorge/indigo_desktop_full_ws/build/pcl_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j[0m' [31mfailed with return code[0m '[1m2[0m'[0m

grafoteka, 2015-11-06:
Hi again!, I've tried to build again and has the same problem, I've a MBP2011 with OS X Yosemite

[pcl_ros] ==> '/Users/jorge/indigo_desktop_full_ws/build/pcl_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' in '/Users/jorge/indigo_desktop_full_ws/build/pcl_ros'
Scanning dependencies of target nodelet_generate_messages_lisp
Scanning dependencies of target std_msgs_generate_messages_lisp
Scanning dependencies of target tf2_msgs_generate_messages_cpp
Scanning dependencies of target std_msgs_generate_messages_py
[  0%] Built target nodelet_generate_messages_lisp
[  0%] Built target tf2_msgs_generate_messages_cpp
Scanning dependencies of target nodelet_generate_messages_py
[  0%] Built target std_msgs_generate_messages_py
[  0%] Built target std_msgs_generate_messages_lisp
[  0%] Built target nodelet_generate_messages_py
Scanning dependencies of target nodelet_topic_tools_gencfg
Scanning dependencies of target std_msgs_generate_messages_cpp
Scanning dependencies of target nodelet_generate_messages_cpp
Scanning dependencies of target tf_generate_messages_py
[  0%] Built target nodelet_topic_tools_gencfg
[  0%] Built target nodelet_generate_messages_cpp
[  0%] Built target std_msgs_generate_messages_cpp
Scanning dependencies of target tf_generate_messages_lisp
[  0%] Built target tf_generate_messages_py
Scanning dependencies of target geometry_msgs_generate_messages_cpp
Scanning dependencies of target tf2_msgs_generate_messages_lisp
[  0%] Built target geometry_msgs_generate_messages_cpp
Scanning dependencies of target geometry_msgs_generate_messages_lisp
Scanning dependencies of target tf2_msgs_generate_messages_py
[  0%] Built target tf_generate_messages_lisp
[  0%] Built target tf2_msgs_generate_messages_lisp
[  0%] Built target tf2_msgs_generate_messages_py
Scanning dependencies of target tf_generate_messages_cpp
Scanning dependencies of target dynamic_reconfigure_generate_messages_py
[  0%] Built target geometry_msgs_generate_messages_lisp
[  0%] Built target dynamic_reconfigure_generate_messages_py
[  0%] Built target tf_generate_messages_cpp
Scanning dependencies of target dynamic_reconfigure_gencfg
Scanning dependencies of target rosgraph_msgs_generate_messages_cpp
Scanning dependencies of target actionlib_generate_messages_cpp
Scanning dependencies of target actionlib_generate_messages_py
[  0%] Built target dynamic_reconfigure_gencfg
[  0%] Built target actionlib_generate_messages_py
[  0%] Built target actionlib_generate_messages_cpp
Scanning dependencies of target dynamic_reconfigure_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_cpp
Scanning dependencies of target actionlib_generate_messages_lisp
Scanning dependencies of target dynamic_reconfigure_generate_messages_lisp
[  0%] Built target dynamic_reconfigure_generate_messages_lisp
[  0%] Built target dynamic_reconfigure_generate_messages_cpp
[  0%] Built target actionlib_generate_messages_lisp
Scanning dependencies of target rosgraph_msgs_generate_messages_lisp
Scanning dependencies of target rosgraph_msgs_generate_messages_py
Scanning dependencies of target sensor_msgs_generate_messages_py
Scanning dependencies of target roscpp_generate_messages_cpp
[  0%] Built target rosgraph_msgs_generate_messages_lisp
[  0%] Built target rosgraph_msgs_generate_messages_py
Scanning dependencies of target actionlib_msgs_generate_messages_lisp
[  0%] Built target sensor_msgs_generate_messages_py
Scanning dependencies of target actionlib_msgs_generate_messages_cpp
[  0%] Built target roscpp_generate_messages_cpp
Scanning dependencies of target roscpp_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_lisp
[  0%] Built target actionlib_msgs_generate_messages_cpp
Scanning dependencies of target actionlib_msgs_generate_messages_py
Scanning dependencies of target roscpp_generate_messages_py
Scanning dependencies of target bond_generate_messages_cpp
[  0%] Built target actionlib_msgs_generate_messages_py
[  0%] Built target roscpp_generate_messages_lisp
[  0%] Built target roscpp_generate_messages_py
Scanning dependencies of target bond_generate_messages_lisp
Scanning dependencies of target geometry_msgs_generate_messages_py
[... 82 lines cut ...]
ld: framework not found GLEW
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [/Users/jorge/indigo_desktop_full_ws/devel/lib/pcl_ros/convert_pointcloud_to_image] Error 1
make[1]: *** [CMakeFiles/convert_pointcloud_to_image.dir/all] Error 2
[ 33%] Linking CXX executable /Users/jorge/indigo_desktop_full_ws/devel/lib/pcl_ros/pointcloud_to_pcd
ld: framework not found GLEW
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [/Users/jorge/indigo_desktop_full_ws/devel/lib/pcl_ros/pointcloud_to_pcd] Error 1
make[1]: *** [CMakeFiles/pointcloud_to_pcd.dir/all] Error 2
make: *** [all] Error 2
[pcl_ros] <== '/Users/jorge/indigo_desktop_full_ws/build/pcl_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' failed with return code '2'
                                                                                                                                                                                                                             
Failed   <== pcl_ros                          [ 55.4 seconds ]                                                                                                                                                              
[... cut ...]
[build] There were '1' errors:                                                                                                                                                                                              

Failed to build package 'pcl_ros' because the following command:

# Command to reproduce:
cd /Users/jorge/indigo_desktop_full_ws/build/pcl_ros && /Users/jorge/indigo_desktop_full_ws/build/pcl_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j; cd -

# Path to log:
cat /Users/jorge/indigo_desktop_full_ws/build/build_logs/pcl_ros.log

Exited with return code: 2

anlif, 2015-11-06:
See #16

grafoteka, 2015-11-09:
Hi @anlif #16 didn't work for me

mahaarbo, 2015-11-16:
The make file for the PCL seems to try to link GLEW from the wrong place. Have you looked at 
https://github.com/PointCloudLibrary/pcl/issues/1289#issuecomment-154898988 ?

mikepurvis, 2016-07-25:
Should be resolved now that PCL 1.8 is released.

closed 2016-07-25 by mikepurvis

lucbettaieb, 2016-07-25:
I just tried fresh and it gave me the same error as in https://github.com/mikepurvis/ros-install-osx/issues/18.

```
Errors     << pcl_ros:make /Users/luc/code/ros-install-osx/indigo_desktop_full_ws/logs/pcl_ros/build.make.000.log
make[2]: *** No rule to make target `/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks/AGL.framework', needed by `/Users/luc/code/ros-install-osx/indigo_desktop_full_ws/devel/.private/pcl_ros/lib/pcl_ros/convert_pointcloud_to_image'.  Stop.
make[2]: *** Waiting for unfinished jobs....
make[2]: *** No rule to make target `/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.10.sdk/System/Library/Frameworks/AGL.framework', needed by `/Users/luc/code/ros-install-osx/indigo_desktop_full_ws/devel/.private/pcl_ros/lib/pcl_ros/pointcloud_to_pcd'.  Stop.
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/convert_pointcloud_to_image.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/pointcloud_to_pcd.dir/all] Error 2
make: *** [all] Error 2
```

mikepurvis, 2016-07-25:
Why guess when you can know?

```
$ brew info pcl
homebrew/science/pcl: stable 1.8.0 (bottled), HEAD
Library for 2D/3D image and point cloud processing
http://www.pointclouds.org/
/usr/local/Cellar/pcl/1.8.0 (1,142 files, 140.0M) *
  Poured from bottle on 2016-06-15 at 22:13:45
From: https://github.com/Homebrew/homebrew-science/blob/master/pcl.rb
==> Dependencies
Build: cmake ✘, pkg-config ✘
Required: boost ✔, eigen ✔, flann ✔, cminpack ✔, qhull ✔, libusb ✔, glew ✔
Recommended: vtk ✔
Optional: qt ✔, qt5 ✘, openni ✘, openni2 ✘
```

lucbettaieb, 2016-07-25:
Speakin' tha truth.

```
brew info pcl                                                                                                 1 ↵
homebrew/science/pcl: stable 1.8.0 (bottled), HEAD
Library for 2D/3D image and point cloud processing
http://www.pointclouds.org/
/usr/local/Cellar/pcl/1.7.2 (1,124 files, 126.9M) *
  Poured from bottle on 2015-04-08 at 06:40:47
From: https://github.com/Homebrew/homebrew-science/blob/master/pcl.rb
==> Dependencies
Build: cmake ✔, pkg-config ✔
Required: boost ✔, eigen ✘, flann ✘, cminpack ✔, qhull ✘, libusb ✘, glew ✘, jpeg ✔, libpng ✔, libtiff ✔, szip ✔, hdf5 ✘
Recommended: vtk ✘
Optional: qt ✔, qt5 ✘, openni ✘, openni2 ✘
==> Options
--with-cuda
	Build with cuda support
--with-examples
	Build pcl examples.
--with-openni
	Build with openni support
--with-openni2
	Build with openni2 support
--with-qt
	Build with qt support
--with-qt5
	Build with qt5 support
--without-apps
	Build without apps.
--without-tools
	Build without tools.
--without-vtk
	Build without vtk support
--HEAD
	Install HEAD version
==> Caveats
.app bundles were installed.
Run `brew linkapps pcl` to symlink these to /Applications.
```

lucbettaieb, 2016-07-25:
Running `brew upgrade pcl` and will try again.  Thanks for the common sense.

mikepurvis, 2016-07-25:
No sweat. Probably a leftover from previously.

sakurazhu, 2016-12-14:
hello ,
i met this problems.  but i don't know how to solve ? could you please give me a hand?

Could NOT find ensenso (missing:  ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR) 
** WARNING ** io features related to ensenso will be disabled
-- Could NOT find DAVIDSDK (missing:  DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
** WARNING ** io features related to davidSDK will be disabled
-- Could NOT find DSSDK (missing:  _DSSDK_LIBRARIES) 
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
-- Could NOT find ensenso (missing:  ENSENSO_LIBRARY ENSENSO_INCLUDE_DIR) 
** WARNING ** visualization features related to ensenso will be disabled
-- Could NOT find DAVIDSDK (missing:  DAVIDSDK_LIBRARY DAVIDSDK_INCLUDE_DIR) 
** WARNING ** visualization features related to davidSDK will be disabled
-- Could NOT find DSSDK (missing:  _DSSDK_LIBRARIES) 
** WARNING ** visualization features related to dssdk will be disabled
-- Could NOT find RSSDK (missing:  _RSSDK_LIBRARIES) 
** WARNING ** visualization features related to rssdk will be disabled
-- looking for PCL_COMMON
-- looking for PCL_OCTREE
-- looking for PCL_IO
-- looking for PCL_STEREO
-- looking for PCL_KDTREE
-- looking for PCL_SEARCH
-- looking for PCL_SAMPLE_CONSENSUS
-- looking for PCL_FILTERS
-- looking for PCL_2D
-- looking for PCL_FEATURES
-- looking for PCL_REGISTRATION
-- looking for PCL_ML
-- looking for PCL_RECOGNITION
-- looking for PCL_TRACKING
-- looking for PCL_GEOMETRY
-- looking for PCL_KEYPOINTS
-- looking for PCL_SURFACE
-- looking for PCL_SEGMENTATION
-- looking for PCL_VISUALIZATION
-- looking for PCL_PEOPLE
-- looking for PCL_OUTOFCORE
-- Configuring done