mikepurvis/ros-install-osx #18: pcl_ros compile issue OSX 10.11

mjm522, 2015-11-05:
I am using the statement to [curl https://raw.githubusercontent.com/mikepurvis/ros-install-osx/master/install | bash] to install ROS in OSX. The installation gets stuck with following error:

Scanning dependencies of target pointcloud_to_pcd
make[2]: *** No rule to make target `/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/System/Library/Frameworks/AGL.framework', needed by `/Users/ad/indigo_desktop_full_ws/devel/lib/pcl_ros/pointcloud_to_pcd'.  Stop.
make[2]: *** Waiting for unfinished jobs....
[ 26%] Building CXX object CMakeFiles/pointcloud_to_pcd.dir/tools/pointcloud_to_pcd.cpp.o
make[1]: *** [CMakeFiles/convert_pcd_to_image.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/pcd_to_pointcloud.dir/all] Error 2
make[1]: *** [CMakeFiles/convert_pointcloud_to_image.dir/all] Error 2
make[1]: *** [CMakeFiles/pointcloud_to_pcd.dir/all] Error 2
make: *** [all] Error 2
[pcl_ros] <== '/Users/ad/indigo_desktop_full_ws/build/pcl_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' failed with return code '2'
                                                                                                     
Failed   <== pcl_ros                          [ 35.8 seconds ]                                      
Finished <== gazebo_ros                       [ 38.5 seconds ]                                      
[build] There were '1' errors:                                                                      

Failed to build package 'pcl_ros' because the following command:

# Command to reproduce:
cd /Users/ad/indigo_desktop_full_ws/build/pcl_ros && /Users/ad/indigo_desktop_full_ws/build/pcl_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j; cd -

# Path to log:
cat /Users/ad/indigo_desktop_full_ws/build/build_logs/pcl_ros.log

Exited with return code: 2 

Could somebody help?

mikepurvis, 2015-11-05:
Suspect this is the same issue as reported in #17. Is this El Cap or Yosemite, and are you working from a full XCode install, or just the CLI tools?

mjm522, 2015-11-06:
Yeah it is XCode.. how can I install ROS using your script without taking pcl_ros? Want to figure whether any other package has issues.

mikepurvis, 2015-11-06:
Mark that package as to-skip by putting a `CATKIN_IGNORE` file in it, then re-run the build.

See: https://github.com/mikepurvis/ros-install-osx#step-by-step

Note that there's at least one other known issue with the XCode toolchain, see: https://github.com/mikepurvis/ros-install-osx/issues/4

mikepurvis, 2016-06-17:
Please open a new issue if you encounter this again.

closed 2016-06-17 by mikepurvis