mikepurvis/ros-install-osx #19: rqt_plot error

grafoteka, 2015-11-06:
When I try to launch rosrun rqt_plot rqt_plot I've got the next error:

$ rosrun rqt_plot rqt_plot 
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/rqt_plot/rqt_plot", line 6, in <module>
    from rqt_plot.plot import Plot
  File "/opt/ros/indigo/lib/python2.7/site-packages/rqt_plot/plot.py", line 43, in <module>
    from .data_plot import DataPlot
  File "/opt/ros/indigo/lib/python2.7/site-packages/rqt_plot/data_plot/__init__.py", line 47, in <module>
    from mat_data_plot import MatDataPlot
  File "/opt/ros/indigo/lib/python2.7/site-packages/rqt_plot/data_plot/mat_data_plot.py", line 49, in <module>
    import matplotlib
  File "/usr/local/lib/python2.7/site-packages/matplotlib/__init__.py", line 1131, in <module>
    rcParams = rc_params()
  File "/usr/local/lib/python2.7/site-packages/matplotlib/__init__.py", line 975, in rc_params
    return rc_params_from_file(fname, fail_on_error)
  File "/usr/local/lib/python2.7/site-packages/matplotlib/__init__.py", line 1100, in rc_params_from_file
    config_from_file = _rc_params_in_file(fname, fail_on_error)
  File "/usr/local/lib/python2.7/site-packages/matplotlib/__init__.py", line 1018, in _rc_params_in_file
    with _open_file_or_url(fname) as fd:
  File "/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/contextlib.py", line 17, in __enter__
    return self.gen.next()
  File "/usr/local/lib/python2.7/site-packages/matplotlib/__init__.py", line 1000, in _open_file_or_url
    encoding = locale.getdefaultlocale()[1]
  File "/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/locale.py", line 543, in getdefaultlocale
    return _parse_localename(localename)
  File "/usr/local/Cellar/python/2.7.10_2/Frameworks/Python.framework/Versions/2.7/lib/python2.7/locale.py", line 475, in _parse_localename
    raise ValueError, 'unknown locale: %s' % localename
ValueError: unknown locale: UTF-8

Does anyone know how to solve it?

mikepurvis, 2015-11-17:
A quick googling [reveals the following suggestion](http://stackoverflow.com/a/19961403):

```
export LC_ALL=en_US.UTF-8
export LANG=en_US.UTF-8
```

Would like to understand how your machine differs from other ones which don't seem to have this issue, and if there's something we can do in the installer to make it not happen. What version of OS X is this, and what's your system's default language? Thanks.

mikepurvis, 2016-06-17:
Please reopen if you're interested to pursue this further.

closed 2016-06-17 by mikepurvis