mikepurvis/ros-install-osx #22: `pip install wstool` needs sudo

goretkin, 2015-12-23:
```
$ pip install wstool
Collecting wstool
Requirement already satisfied (use --upgrade to upgrade): vcstools>=0.1.38 in /usr/local/lib/python2.7/site-packages (from wstool)
Requirement already satisfied (use --upgrade to upgrade): pyyaml in /usr/local/lib/python2.7/site-packages (from wstool)
Requirement already satisfied (use --upgrade to upgrade): python-dateutil in /usr/local/lib/python2.7/site-packages (from vcstools>=0.1.38->wstool)
Requirement already satisfied (use --upgrade to upgrade): six>=1.5 in /usr/local/lib/python2.7/site-packages (from python-dateutil->vcstools>=0.1.38->wstool)
Installing collected packages: wstool
Exception:
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/pip/basecommand.py", line 211, in main
    status = self.run(options, args)
  File "/usr/local/lib/python2.7/site-packages/pip/commands/install.py", line 311, in run
    root=options.root_path,
  File "/usr/local/lib/python2.7/site-packages/pip/req/req_set.py", line 646, in install
    **kwargs
  File "/usr/local/lib/python2.7/site-packages/pip/req/req_install.py", line 803, in install
    self.move_wheel_files(self.source_dir, root=root)
  File "/usr/local/lib/python2.7/site-packages/pip/req/req_install.py", line 998, in move_wheel_files
    isolated=self.isolated,
  File "/usr/local/lib/python2.7/site-packages/pip/wheel.py", line 371, in move_wheel_files
    clobber(source, dest, False, fixer=fixer, filter=filter)
  File "/usr/local/lib/python2.7/site-packages/pip/wheel.py", line 317, in clobber
    shutil.copyfile(srcfile, destfile)
  File "/usr/local/Cellar/python/2.7.11/Frameworks/Python.framework/Versions/2.7/lib/python2.7/shutil.py", line 83, in copyfile
    with open(dst, 'wb') as fdst:
IOError: [Errno 13] Permission denied: '/usr/local/etc/bash_completion.d/wstool-completion.bash'
```

at least on OS X 10.9

mikepurvis, 2015-12-23:
Looks like you still have some root-owned stuff in `/usr/local`. It's a blunt instrument, but consider:

```
sudo chown -R $USER /usr/local/*
```

Alternatively, and even more bluntly, just remove (or move/backup) your homebrew install altogether and start over.

goretkin, 2015-12-23:
I am sorry, my eyes had told me it was trying to write into `/etc`. Thank you.

closed 2015-12-23 by goretkin