mikepurvis/ros-install-osx #23: who installs `/usr/local/opt/python/bin/python2.7`?

goretkin, 2015-12-23:
I notice that the ROS Python scripts (rosnode, rosmsg, etc.) have a shebang `#!/usr/local/opt/python/bin/python2.7`, which does exist and is symlinked to `../Frameworks/Python.framework/Versions/2.7/bin/python`

However, I also have `/usr/local/bin/python` which is symlinked to `../Cellar/python/2.7.11/bin/python`.

Did the script install the python distribution under `/usr/local/opt`? It was my understanding that this ROS installation script was supposed to use a brew python, otherwise what difference would it make if there were pip-installed modules in the brew python?

goretkin, 2015-12-23:
Furthermore, I have a directory `~/Library/Python/2.7/lib/python/site-packages/` that has an IPython module in it (I tried installing IPython) and also a `homebrew.pth` with the following contents
``` 
$ cat ~/Library/Python/2.7/lib/python/site-packages/homebrew.pth 
/usr/local/lib/python2.7/site-packages
/usr/local/lib/python2.7/site-packages
/usr/local/lib/python2.7/site-packages
/usr/local/lib/python2.7/site-packages
/usr/local/lib/python2.7/site-packages
/usr/local/lib/python2.7/site-packages
```

Did the script do this? The reason I noticed it is that an older version of IPython was masking the new version I had installed with pip install

mikepurvis, 2015-12-23:
If you have `/usr/local/opt/python` on your system, that was installed either manually, or via macports. Neither Homebrew nor system Python have anything to do with `/opt`.

What is the contents of your `PATH` var?

```
$ echo $PATH
/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin
```

goretkin, 2015-12-23:
thanks for that clarification.
```
echo $PATH
/usr/local/bin::/usr/bin:/bin:/usr/sbin:/sbin:/usr/local/bin:/opt/X11/bin:/usr/local/git/bin:/usr/texbin
```

And indeed 
```
$ which python
/usr/local/bin/python
```

So then I don't know why the interpreter are set as they are for the ROS Python scripts. Except for maybe `pip` had the wrong interpreter.
```
$ head -n 1 $(which pip)
#!/usr/local/opt/python/bin/python2.7
```

Do you know if I can start over so that ROS picks up the correct python, without recompiling and redownloading every package?

mikepurvis, 2015-12-23:
Okay, yes, those match for me, have updated my comment above.

You don't need to redownload, but rebuilding is going to be your only surefire way of getting it right. See: https://github.com/mikepurvis/ros-install-osx#step-by-step

closed 2016-07-25 by mikepurvis