mikepurvis/ros-install-osx #28: Error: The `brew link` step did not complete successfully

mcevzar, 2016-02-03:
I get the error below and can't seem to find ROS anywhere in dashboard.

Error: The `brew link` step did not complete successfully
The formula built, but is not symlinked into /usr/local
Could not symlink share/man/man5/png.5
/usr/local/share/man/man5 is not writable.

You can try again using:
  brew link libpng

mikepurvis, 2016-02-03:
Sounds like a homebrew permissions issue. Please try:

```
sudo chown -R "$USER":admin /usr/local
sudo chown -R "$USER":admin /Library/Caches/Homebrew
brew link libpng
```

See also: http://stackoverflow.com/a/16450503/109517

mcevzar, 2016-02-04:
Thank you. Now I get the next error message.

Command "/usr/local/opt/python/bin/python2.7 -c "import setuptools, tokenize;__file__='/private/var/folders/42/7jm2xqzj6ns2h46fl9n3lv3m0000gn/T/pip-build-EnJbcM/Pillow/setup.py';exec(compile(getattr(tokenize, 'open', open)(__file__).read().replace('\r\n', '\n'), __file__, 'exec'))" install --record /var/folders/42/7jm2xqzj6ns2h46fl9n3lv3m0000gn/T/pip-030CUb-record/install-record.txt --single-version-externally-managed --compile" failed with error code 1 in /private/var/folders/42/7jm2xqzj6ns2h46fl9n3lv3m0000gn/T/pip-build-EnJbcM/Pillow

ERROR: the following rosdeps failed to install
  pip: command [pip install -U Pillow] failed

mikepurvis, 2016-06-17:
The only thing I could suggest at this point would be to nuke your homebrew install and python site packages dir.

closed 2016-06-17 by mikepurvis

secdevco, 2017-10-06:
Thanks so much this original answer worked for me on inkscape installation through homebrew