mikepurvis/ros-install-osx #29: warehouse_ros doesn't build on OSX10.11

fjonath1, 2016-02-17:
The dependency of warehouse_ros (mongodb-dev) does not build on OSX 10.11.
I updated a brew formula that uses mongodb-dev patched to build on OSX 10.11. 
[This is the formula](https://github.com/fjonath1/homebrew-deps/blob/master/mongodb-dev.rb)
[And this is the patched mongodb-dev repository](https://github.com/fjonath1/mongodb-ros-osx.git)

I think this changes will be useful for other OSX users, but I'm not sure if and how I can integrate it with your ros build scripts.

mikepurvis, 2016-02-17:
Hi! There isn't really a place for your work in this immediate repo. You'd either want to either a) get the formula merged in to ros/homebrew-deps, where it would be available to all OS X ROS users, or b) work with upstream Homebrew to make [their formula](https://github.com/Homebrew/homebrew/blob/master/Library/Formula/mongodb.rb) do what you need— perhaps with a flag? Note that rosdep's Homebrew plugin does support specifying flags for dependencies, for example PCL uses `--HEAD`: https://github.com/ros/rosdistro/blob/master/rosdep/osx-homebrew.yaml#L252

fjonath1, 2016-02-17:
Hi.
I'm trying to make the formula changes on [ros/homebrew-deps](https://github.com/ros/homebrew-deps/pull/12#issuecomment-184996774) now.
Thank you for your input.

closed 2016-02-17 by fjonath1