mikepurvis/ros-install-osx #30: pcl_ros fails to build on OSX 10.11, and workaround

fferri, 2016-03-04:
I'm (manually) following the install script, up to the point:

    catkin build     -DCMAKE_BUILD_TYPE=Release     -DPYTHON_LIBRARY=$(python -c "import sys; print sys.prefix")/lib/libpython2.7.dylib     -DPYTHON_INCLUDE_DIR=$(python -c "import sys; print sys.prefix")/include/python2.7

where build fails when starting to build pcl_ros, with the error:

[  0%] Built target std_msgs_generate_messages_py
[ 20%] Built target pcl_ros_gencfg
make[2]: *** No rule to make target `/usr/local/Cellar/qt/4.8.7_1/lib/QtGui.framework', needed by `/Users/federico/indigo_desktop_full_ws/devel/lib/pcl_ros/convert_pcd_to_image'.  Stop.
make[1]: *** [CMakeFiles/convert_pcd_to_image.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
make[2]: *** No rule to make target `/usr/local/Cellar/qt/4.8.7_1/lib/QtGui.framework', needed by `/Users/federico/indigo_desktop_full_ws/devel/lib/pcl_ros/convert_pointcloud_to_image'.  Stop.
make[1]: *** [CMakeFiles/convert_pointcloud_to_image.dir/all] Error 2
make[2]: *** No rule to make target `/usr/local/Cellar/qt/4.8.7_1/lib/QtGui.framework', needed by `/Users/federico/indigo_desktop_full_ws/devel/lib/pcl_ros/pcd_to_pointcloud'.  Stop.
make[1]: *** [CMakeFiles/pcd_to_pointcloud.dir/all] Error 2
make[2]: *** No rule to make target `/usr/local/Cellar/qt/4.8.7_1/lib/QtGui.framework', needed by `/Users/federico/indigo_desktop_full_ws/devel/lib/pcl_ros/pointcloud_to_pcd'.  Stop.
make[1]: *** [CMakeFiles/pointcloud_to_pcd.dir/all] Error 2
make: *** [all] Error 2
[pcl_ros] <== '/Users/federico/indigo_desktop_full_ws/build/pcl_ros/build_env.sh /usr/bin/make --jobserver-fds=3,4 -j' failed with return code '2'

so I went to /usr/local/Cellar/qt to see what's there, and in fact I did not have 4.8.7_1, so I create a symlink 4.8.7_1 pointing to my 4.8.7_2 installed qt, and restarted build.

Note that the workaround did now work the first time, and the second time I did another try without ANY symlink in /usr/local/Cellar/qt (so also removing the one created in the install script), then the third time I tried with the workaround I just described, so maybe this needs to be double checked.

fferri, 2016-03-04:
update: it's still failing to build, also after the workaround

mikepurvis, 2016-03-04:
After putting in the workaround symlink, did you remove the build and devel folders for a fully clean build?

Looks like errno=22 is often tied to weird toolchain/environment configs. Can you confirm that you're using the command line tools and not XCode's GCC to compile? Please post the results of `which gcc` and also `env`.

fferri, 2016-03-04:
$ ls -l $(which gcc)
-rwxr-xr-x  1 root  wheel  18176 Jan 14 03:12 /usr/bin/gcc

$ env
TERM_PROGRAM=iTerm.app
SHELL=/bin/bash
TERM=xterm-256color
CLICOLOR=1
TMPDIR=/var/folders/hq/r_l8wk257_71w4rkrsj5b7d80000gn/T/
Apple_PubSub_Socket_Render=/private/tmp/com.apple.launchd.n9CZMbzxl3/Render
USER=federico
RLWRAP_EDITOR=vi +%L
SSH_AUTH_SOCK=/private/tmp/com.apple.launchd.qNPRSZNKnk/Listeners
__CF_USER_TEXT_ENCODING=0x1F5:0x0:0x0
PAGER=vimpager
PATH=/usr/local/sbin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/usr/local/MacGPG2/bin:/Library/TeX/texbin
PWD=/Users/federico/indigo_desktop_full_ws
JAVA_HOME=/Library/Java/JavaVirtualMachines/jdk1.8.0_60.jdk/Contents/Home
EDITOR=vim
ITERM_PROFILE=Default
XPC_FLAGS=0x0
PS1=\[\e[0;36m\]\h:\[\e[1m\]\W \[\e[0;36m\]\$\[\e[0m\]
HISTCONTROL=erasedups
XPC_SERVICE_NAME=0
RLWRAP_HOME=/Users/federico/.rlwrap/
SHLVL=1
HOME=/Users/federico
ECLIPSEDIR=/opt/eclipse-clp-6.1_201
LOGNAME=federico
LC_CTYPE=UTF-8
PROMPT_COMMAND=echo -ne "\033]0;${USER}@${HOSTNAME%%.*}:${PWD/#$HOME/~}\007"
DISPLAY=/private/tmp/com.apple.launchd.rot917xnYj/org.macosforge.xquartz:0
HISTFILE=/dev/null
_=/usr/bin/env
OLDPWD=/usr/local/Cellar/qt

fferri, 2016-03-04:
Now I'm rebuilding, but I can already report it from the CMake output I see:

-- Check for working C compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/cc -- works
...
-- Check for working CXX compiler: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++ -- works

fferri, 2016-03-04:
After removing build and devel, it built successfully, thanks @mikepurvis !
This was the content of /usr/local/Cellar/qt:
total 16
lrwxr-xr-x   1 federico  admin    7 Mar  4 19:57 4.8.6 -> 4.8.7_2
lrwxr-xr-x   1 federico  admin    7 Mar  4 17:39 4.8.7_1 -> 4.8.7_2
drwxr-xr-x  19 federico  admin  646 Mar  4 17:33 4.8.7_2

mikepurvis, 2016-03-04:
Sweet, good to know.

closed 2016-03-04 by mikepurvis