mikepurvis/ros-install-osx #32: pcl_ros compilation error with vtk 7.0.0

hawesie, 2016-03-15:
It looks like Homebrew recently upgrade vtk to 7.0.0. When running the install script this caused the compilation of `pcl_ros` to fail with the error:

```
ld: library not found for -lvtkftgl
```

If I use a previous version of vtk from Homebrew:

```
cd /usr/local/Library/Taps/homebrew/homebrew-science/
git checkout 092849afd643314eda1699ab6e6f1a793df91521 vtk.rb
```

and then comment out `brew update` from the script to prevent this change being discarded during script execution, everything compiles fine.

mikepurvis, 2016-03-15:
Looks like another issue with PCL caching cmake find module results inside the bottle. You can see this in action by grepping `/usr/local/Cellar/pcl/*/share/pcl-*/PCLConfig.cmake` for `vtkftgl`.

Please try:

```
brew install pcl --build-from-source
```

If after this you can compile pcl_ros against VTK 7.0, then we should request that Homebrew re-build the PCL bottle.

mikepurvis, 2016-03-15:
Oh, yuck. Looks like there's more to it: https://github.com/Homebrew/homebrew-science/issues/3370

mikepurvis, 2016-03-15:
So a potentially better workaround (for now) then, is:

```
brew install pcl --without-vtk
```

mikepurvis, 2016-03-15:
@hawesie Did that work for you? If so, we should either ament the install script here, or possibly just add the install flag in rosdep:

https://github.com/ros/rosdistro/blob/bd3d91ffe36c0ec55459b322514797757482d290/rosdep/osx-homebrew.yaml#L250

@wjwwood @paulbovbel How critical is VTK to PCL? We seem to have this synchronization issue a lot; it would be great to make `--without-vtk` the default.

wjwwood, 2016-03-15:
It is used for the visualization tools in PCL and some of the examples use that. I don't know how important it is (I haven't tried using PCL with ROS without VTK), but I think the biggest risk is that more people have to build PCL themselves after installing because the one we deliver is incomplete. I'm ok with that, but I'll abstain to voting one way or the other on the issue. Also it probably prevents us from using a bottle for pcl.

mikepurvis, 2016-03-15:
As far as I know, we can't use the bottle anyway, since PCL's release policy is to not have any for like 18 months: https://github.com/PointCloudLibrary/pcl/releases

spmaniato, 2016-03-17:
Had the same problem after a `brew upgrade`. For future reference, I fixed it with:
```
brew remove pcl
brew install pcl --without-vtk
``` 
because `brew reinstall pcl --without-vtk` failed. I was then able to compile `pcl_ros` :+1:

ahundt, 2016-04-07:
The best option may be for us to create a versioned installer for vtk 6.3 in:
https://github.com/Homebrew/homebrew-versions
or
https://github.com/Homebrew/homebrew-science

and have pcl depend on that until then

ahundt, 2016-04-07:
as you may see I've created a pull request I'm hoping will fix this.

quick fix (use at your risk):

```
brew uninstall pcl
brew uninstall vtk
brew untap homebrew/science
brew tap ahundt
brew install pcl
```

mikepurvis, 2016-04-07:
Easier:

```
brew install https://raw.githubusercontent.com/ahundt/homebrew-science/vtk6/vtk6.rb
brew install https://raw.githubusercontent.com/ahundt/homebrew-science/vtk6/pcl.rb
```

I'd accept a PR to change the install script to this, in advance of a homebrew-science merge.

ahundt, 2016-04-09:
@mikepurvis your solution is definitely better, the problem I ran into with rosdep automatically retapping homebrew/science packages is definitely  avoided with those lines. I'd make a pull request but I've got deadlines in the near future so I can't test a full reinstall.

ahundt, 2016-04-09:
minor modification to the above:
```
brew install https://raw.githubusercontent.com/ahundt/homebrew-science/vtk6/vtk6.rb --build-from-source
brew install https://raw.githubusercontent.com/ahundt/homebrew-science/vtk6/pcl.rb --build-from-source
```

mikepurvis, 2016-06-15:
PCL 1.8.0 is released, which should finally clean up a bunch of this stuff: https://github.com/Homebrew/homebrew-science/pull/3752

mikepurvis, 2016-06-17:
Resolved in #41.

closed 2016-06-17 by mikepurvis