mikepurvis/ros-install-osx #33: protoc version for gazebo5

moriarty, 2016-03-16:
While running the script, on El Capitan, I ran into this issue:

```
...
[ 80%] Built target gazebo_ros_paths_plugin
In file included from /Users/alex/workspace/ros-install-osx/indigo_desktop_full_ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_api_plugin.cpp:25:
In file included from /Users/alex/workspace/ros-install-osx/indigo_desktop_full_ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_api_plugin.h:34:
In file included from /usr/local/Cellar/gazebo5/5.2.1/include/gazebo-5.2/gazebo/physics/physics.hh:2:
In file included from /usr/local/Cellar/gazebo5/5.2.1/include/gazebo-5.2/gazebo/physics/Actor.hh:24:
In file included from /usr/local/Cellar/gazebo5/5.2.1/include/gazebo-5.2/gazebo/physics/Model.hh:34:
In file included from /usr/local/Cellar/gazebo5/5.2.1/include/gazebo-5.2/gazebo/physics/Entity.hh:29:
In file included from /usr/local/Cellar/gazebo5/5.2.1/include/gazebo-5.2/gazebo/msgs/msgs.hh:24:
In file included from /usr/local/Cellar/gazebo5/5.2.1/include/gazebo-5.2/gazebo/msgs/MessageTypes.hh:7:
/usr/local/Cellar/gazebo5/5.2.1/include/gazebo-5.2/gazebo/msgs/axis.pb.h:17:2: error: This file was generated by an older version of protoc which is
#error This file was generated by an older version of protoc which is
 ^
/usr/local/Cellar/gazebo5/5.2.1/include/gazebo-5.2/gazebo/msgs/axis.pb.h:18:2: error: incompatible with your Protocol Buffer headers.  Please
#error incompatible with your Protocol Buffer headers.  Please
 ^
/usr/local/Cellar/gazebo5/5.2.1/include/gazebo-5.2/gazebo/msgs/axis.pb.h:19:2: error: regenerate this file with a newer version of protoc.
#error regenerate
...
```

This is because the gazebo message header files generated an included in the tar include:

```
#if GOOGLE_PROTOBUF_VERSION < 2006000
#error This file was generated by a newer version of protoc which is
#error incompatible with your Protocol Buffer headers.  Please update
#error your headers.
#endif
#if 2006001 < GOOGLE_PROTOBUF_MIN_PROTOC_VERSION
#error This file was generated by an older version of protoc which is
#error incompatible with your Protocol Buffer headers.  Please
#error regenerate this file with a newer version of protoc.
#endif
```

I was able to get around it by doing what the error message told me to do, regenerating the protobuf headers and dylib:

1. Clone gazebo and get version 5.2.1:
     ```hg clone https://bitbucket.org/osrf/gazebo && cd gazebo && hg up gazebo5_5.2.1```
2. Built it:  ```mkdir build && cd build && cmake .. && make -j8```
3. Copy the dylib: 
    ```cp ./gazebo/msgs/libgazebo_msgs.5.2.1.dylib /usr/local/Cellar/gazebo5/5.2.1/lib```
4. Copy the headers: 
    ```cp ./gazebo/msgs/*.pb.h /usr/local/Cellar/gazebo5/5.2.1/include/gazebo-5.2/gazebo/msgs/```
5. Copy the hh file: 
    ```cp ./gazebo/msgs/*.hh /usr/local/Cellar/gazebo5/5.2.1/include/gazebo-5.2/gazebo/msgs/```

Now it builds until pcl_ros - I'll read over the latest comments in #12, because my notes from last time didn't help.

I just wanted to know if anyone else has experienced this.

I think this may actually be an issue for https://github.com/osrf/homebrew-simulation/ because the brew formula for ```gazebo5``` depends on a specific version of ```protobuf```

scpeters, 2016-03-17:
Maybe the issue is that we depend on `protobuf-c`, which depends on `protobuf`. If you installed the `protobuf-c` bottle, that may be build against the old version of `protobuf`. What if you build `protobuf-c` from source against your `--devel` version of `protobuf`?

moriarty, 2016-03-18:
@scpeters, thanks - I have a new ssd arriving from Amazon this weekend and will try that on my desktop. I'll post back here with the results.

At the moment, everything is working on my laptop by copying over the files, it's a hack but for now I won't touch it.

moriarty, 2016-03-24:
I'll close this for now until I have time to test. I seem to be the only one affected.

closed 2016-03-24 by moriarty

q4z1, 2016-06-17:
A general solution is to manually re-generate the files with protoc:

protoc <file.proto> --cpp_out=<output_dir>

Random-Word, 2016-11-02:
Tensorflow requires a very specific version of protobuf to behave - so specific it's a commit and not a release tag. I'm glad the apt package for Gazebo comes with the protobuf source so I could follow your suggestion and just recompile everything. Thanks @q4z1.