mikepurvis/ros-install-osx #38: Unclear on how to install packages for ros

korymath, 2016-04-25:
Most likely a simple question here, I am confused as to how to best install packages on ROS on OSX when I don't have access to apt-get? 

Does this mean that I need to download and build the sources from scratch?

Thank you. Again, apologies if this is an easy issue.

mikepurvis, 2016-04-25:
Yes, you have to build everything from scratch, which is what this repo is about, hopefully streamlining that process to make it as painless as possible. Note that you can overlay workspaces multiple times with catkin, so you can build additional stuff on top of desktop_full, and then extend that for the stuff you're actively developing, or you can use the `ROS_EXTRA_PACKAGES` envvar to specify extra things you'd like to be part of the base `/opt/ros/indigo` workspace.

korymath, 2016-04-25:
Hi Mike, 
Thank you for the quick reply. That is fantastic. Much appreciated.

closed 2016-04-25 by mikepurvis