mikepurvis/ros-install-osx #45: Rosdep error during installation

Numero7, 2016-07-08:
The install script halts at some point, indicating a rosdep error.
OSX El Capitan 10.11.5
Output below

update complete.
~/Documents/Code/ros-install-osx/indigo_desktop_full_ws/src/rviz/ogre_media/models ~/Documents/Code/ros-install-osx/indigo_desktop_full_ws ~/Documents/Code/ros-install-osx
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  5863  100  5863    0     0   8485      0 --:--:-- --:--:-- --:--:-- 21958
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2488  100  2488    0     0  10435      0 --:--:-- --:--:-- --:--:-- 10587
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 11488  100 11488    0     0  46758      0 --:--:-- --:--:-- --:--:-- 48472
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 42238  100 42238    0     0   137k      0 --:--:-- --:--:-- --:--:--  138k
~/Documents/Code/ros-install-osx/indigo_desktop_full_ws ~/Documents/Code/ros-install-osx

ERROR: Rosdep experienced an internal error.
Please go to the rosdep page [1] and file a bug report with the message below.
[1] : http://www.ros.org/wiki/rosdep

rosdep version: 0.11.5

Bad installer [homebrew]: Error while parsing brew info for 'tango-icon-theme'
 * Output of `brew info tango-icon-theme --json=v1`:
 
 * Error while parsing:
 Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/rosdep2/platforms/osx.py", line 184, in is_installed
    pkg_info = json.loads(std_out)
  File "/usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/__init__.py", line 339, in loads
    return _default_decoder.decode(s)
  File "/usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 364, in decode
    obj, end = self.raw_decode(s, idx=_w(s, 0).end())
  File "/usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/lib/python2.7/json/decoder.py", line 382, in raw_decode
    raise ValueError("No JSON object could be decoded")
ValueError: No JSON object could be decoded

Numero7, 2016-07-08:
$ brew info tango-icon-theme --json=v1
Error: Formulae found in multiple taps: 
 * ros/deps/tango-icon-theme
 * ros/groovy/tango-icon-theme

Please use the fully-qualified name e.g. ros/deps/tango-icon-theme to refer the formula.

mikepurvis, 2016-07-08:
`brew untap ros/groovy` should resolve this for you.

Numero7, 2016-07-08:
thanks

Actually in the meantime I did nuke brew

Best regards

mikepurvis, 2016-07-08:
FWIW, we used to detect and auto-untap `ros/hydro`:

https://github.com/mikepurvis/ros-install-osx/commit/5facd972e1956c73e63a70f6b897d172cc26fc79#diff-19ad89bc3e3c9d7ef68b89523eff1987L92

But I had no idea there might be homebrew installations out there which `ros/groovy` tapped. 😃

closed 2016-07-08 by mikepurvis