mikepurvis/ros-install-osx #52: homebrew: command [brew install nose] failed

lucbettaieb, 2016-07-25:
```
...
[xacro] Fetching https://github.com/ros-gbp/xacro-release/archive/release/indigo/xacro/1.9.5-0.tar.gz (version xacro-release-release-indigo-xacro-1.9.5-0) to /Users/luc/code/ros-install-osx/indigo_desktop_full_ws/src/xacro
[xacro] still active
[xacro] Done.

update complete.
~/code/ros-install-osx/indigo_desktop_full_ws/src/rviz/ogre_media/models ~/code/ros-install-osx/indigo_desktop_full_ws ~/code/ros-install-osx
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  5863  100  5863    0     0   7280      0 --:--:-- --:--:-- --:--:--  7274
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100  2488  100  2488    0     0   6012      0 --:--:-- --:--:-- --:--:--  6024
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 11488  100 11488    0     0  38312      0 --:--:-- --:--:-- --:--:-- 38293
  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current
                                 Dload  Upload   Total   Spent    Left  Speed
100 42238  100 42238    0     0  97711      0 --:--:-- --:--:-- --:--:-- 97773
~/code/ros-install-osx/indigo_desktop_full_ws ~/code/ros-install-osx
Error: No available formula with the name "nose"
executing command [brew install nose]
Error: No available formula with the name "nose"
==> Searching for similarly named formulae...
Error: No similarly named formulae found.
==> Searching taps...
Error: No formulae found in taps.
ERROR: the following rosdeps failed to install
  homebrew: command [brew install nose] failed
```

Any idea?  When i try installing `nose` with `pip install --upgrade nose`, it works.  Not sure why homebrew is trying to install it and failing each time.

mikepurvis, 2016-07-25:
This bug was introduced by a hasty merge of #49. Fixed in 44e577bc5a77ff980077965ccf55eee87d59b372.

closed 2016-07-25 by mikepurvis

lucbettaieb, 2016-07-25:
@mikepurvis Thanks