mikepurvis/ros-install-osx #53: Compilation failed - ImportError: No module named catkin.environment_cache

lilouch, 2016-07-25:
Hi,

I'm not able to compile successfully ROS on OS X El Capitan.
I get this error :

```
errors     << geometry_msgs:cmake /Users/libraries/ros-install-osx/indigo_desktop_full_ws/logs/geometry_msgs/build.cmake.000.log                             
Traceback (most recent call last):
  File "/Users/libraries/ros-install-osx/indigo_desktop_full_ws/build/geometry_msgs/catkin_generated/generate_cached_setup.py", line 20, in <module>
    from catkin.environment_cache import generate_environment_script
ImportError: No module named catkin.environment_cache
CMake Error at /opt/ros/indigo/share/catkin/cmake/safe_execute_process.cmake:11 (message):
  execute_process(/usr/local/bin/python
  "/Users/libraries/ros-install-osx/indigo_desktop_full_ws/build/geometry_msgs/catkin_generated/generate_cached_setup.py")
  returned error code 1
Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/all.cmake:186 (safe_execute_process)
  /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:51 (include)
  CMakeLists.txt:4 (find_package)

cd /Users/libraries/ros-install-osx/indigo_desktop_full_ws/build/geometry_msgs; catkin build --get-env geometry_msgs | catkin env -si  /usr/local/bin/cmake /Users/libraries/ros-install-osx/indigo_desktop_full_ws/src/common_msgs/geometry_msgs --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/Users/libraries/ros-install-osx/indigo_desktop_full_ws/devel/.private/geometry_msgs -DCMAKE_INSTALL_PREFIX=/opt/ros/indigo -DCATKIN_ENABLE_TESTING=0 -DCMAKE_BUILD_TYPE=Release -DPYTHON_LIBRARY=/usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib -DPYTHON_INCLUDE_DIR=/usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/include/python2.7; cd -
..........................................................................................................................................................................
Failed     << geometry_msgs:cmake                             [ Exited with code 1 ]                                                                                      
Failed    <<< geometry_msgs                                   [ 2.8 seconds ]                                                                                             
Abandoned <<< pr2_description                                 [ Unrelated job failed ]                                                                                    
Abandoned <<< roslisp                                         [ Unrelated job failed ]                                                                                    
Abandoned <<< rosmsg                                          [ Unrelated job failed ]                                                                                    
Abandoned <<< rospy                                           [ Unrelated job failed ]                                                                                    
Abandoned <<< rosservice                                      [ Unrelated job failed ]                                                                                    
Abandoned <<< roslaunch                                       [ Unrelated job failed ]                                                                                    
Abandoned <<< pluginlib                                       [ Unrelated job failed ]                                                                                    
Abandoned <<< qt_gui_cpp                                      [ Unrelated job failed ]                                                                                    
Abandoned <<< resource_retriever                              [ Unrelated job failed ]                                                                                    
Abandoned <<< rosconsole_bridge                               [ Unrelated job failed ]                                                                                    
Abandoned <<< rostest                                         [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_action                                      [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_bag                                         [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_bag_plugins                                 [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_console                                     [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_dep                                         [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_graph                                       [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_logger_level                                [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_moveit                                      [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_msg                                         [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_nav_view                                    [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_plot                                        [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_pose_view                                   [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_publisher                                   [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_py_console                                  [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_reconfigure                                 [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_robot_dashboard                             [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_robot_monitor                               [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_robot_steering                              [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_runtime_monitor                             [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_service_caller                              [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_shell                                       [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_srv                                         [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_tf_tree                                     [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_top                                         [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_topic                                       [ Unrelated job failed ]                                                                                    
[... cut ...]
[build]   Failed:    1 packages failed.                                                                                                                                   
[build] Runtime: 1 minute and 47.7 seconds total.                                                                                                                         
[build] Note: Workspace packages have changed, please re-source setup files to use them.

```

I tried to look on the internet but did not find any suitable solution.
Could you please take a look on that ? Thank

mikepurvis, 2016-07-25:
This is a long-standing race condition issue with catkin_tools 0.4.0, with a solution proposed here: https://github.com/catkin/catkin_tools/pull/391

If you `rm -rf build/geometry_msgs` and then `catkin build --start-with geometry_msgs` in your workspace, it should be able to pick up where it left off.

lilouch, 2016-07-25:
I did the modifications you have mentioned but I get an error now when I do 
`catkin build --start-with geometry_msgs`

Here the results : 
```
Traceback (most recent call last):
  File "/usr/local/bin/catkin", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python2.7/site-packages/catkin_tools/commands/catkin.py", line 258, in main
    catkin_main(sysargs)
  File "/usr/local/lib/python2.7/site-packages/catkin_tools/commands/catkin.py", line 163, in catkin_main
    argument_preprocessors = create_subparsers(parser, verbs)
  File "/usr/local/lib/python2.7/site-packages/catkin_tools/commands/catkin.py", line 58, in create_subparsers
    desc = load_verb_description(verb)
  File "/usr/local/lib/python2.7/site-packages/catkin_tools/commands/catkin.py", line 45, in load_verb_description
    return entry_point.load()
  File "/usr/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2229, in load
    return self.resolve()
  File "/usr/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2235, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/usr/local/lib/python2.7/site-packages/catkin_tools/verbs/catkin_build/__init__.py", line 17, in <module>
    from .cli import main
  File "/usr/local/lib/python2.7/site-packages/catkin_tools/verbs/catkin_build/cli.py", line 63, in <module>
    from .build import build_isolated_workspace
  File "/usr/local/lib/python2.7/site-packages/catkin_tools/verbs/catkin_build/build.py", line 57, in <module>
    from catkin_tools.jobs.catkin import create_catkin_build_job
  File "/usr/local/lib/python2.7/site-packages/catkin_tools/jobs/catkin.py", line 38, in <module>
    from .utils import get_env_loader
ImportError: cannot import name get_env_loader

```

mikepurvis, 2016-07-25:
Can you clarify exactly what you did? You should not have needed to modify catkin_tools at all— my suggestion was only to retry the failed package.

lilouch, 2016-07-25:
I first did as you mentioned 
`rm -rf build/geometry_msgs `
Then 
`catkin build --start-with geometry_msgs`

I got exactly the same problem (of my initial post).
So I checked your link and I replaced some files in Catkin_tool 
Maybe It was a mistake to do that ?

mikepurvis, 2016-07-25:
In that case, please just delete the whole workspace, reinstall catkin_tools, and try again. This is a transient problem, and that's the easiest fix.

closed 2016-07-25 by mikepurvis

lilouch, 2016-07-25:
Okey I had a backup and remove the modification. So I retried the failed package using your command line and here the results : 

```
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Profile:                     default
Extending:                   None
Workspace:                   /Users/libraries/ros-install-osx/indigo_desktop_full_ws
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Source Space:       [exists] /Userslibraries/ros-install-osx/indigo_desktop_full_ws/src
Log Space:          [exists] /Users/libraries/ros-install-osx/indigo_desktop_full_ws/logs
Build Space:        [exists] /Users/libraries/ros-install-osx/indigo_desktop_full_ws/build
Devel Space:        [exists] /Users/libraries/ros-install-osx/indigo_desktop_full_ws/devel
Install Space:      [exists] /opt/ros/indigo
DESTDIR:            [unused] None
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Devel Space Layout:          linked
Install Space Layout:        merged
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Additional CMake Args:       -DCATKIN_ENABLE_TESTING=0 -DCMAKE_BUILD_TYPE=Release -DPYTHON_LIBRARY=/usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib -DPYTHON_INCLUDE_DIR=/usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/include/python2.7
Additional Make Args:        None
Additional catkin Make Args: None
Internal Make Job Server:    True
Cache Job Environments:      False
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------
Whitelisted Packages:        None
Blacklisted Packages:        None
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------

---------------------------------------------------------------------------------------------------------------------------------------------------------------------------
WARNING: Your workspace is not extending any other result space, but it is set to use a `linked` devel space layout. This requires the `catkin` CMake package in your
source space in order to be built.
---------------------------------------------------------------------------------------------------------------------------------------------------------------------------

[build] Found '225' packages in 0.0 seconds.                                                                                                                              
Skipping  --- catkin                                                                                                                                                      
Skipping  --- genmsg                                                                                                                                                      
Skipping  --- gencpp                                                                                                                                                      
Skipping  --- genlisp                                                                                                                                                     
Skipping  --- genpy                                                                                                                                                       
Skipping  --- angles                                                                                                                                                      
Skipping  --- bond_core                                                                                                                                                   
Skipping  --- cmake_modules                                                                                                                                               
Skipping  --- class_loader                                                                                                                                                
Skipping  --- common_msgs                                                                                                                                                 
Skipping  --- common_tutorials                                                                                                                                            
Skipping  --- convex_decomposition                                                                                                                                        
Skipping  --- cpp_common                                                                                                                                                  
Skipping  --- desktop                                                                                                                                                     
[... cut ...]
[build]   Warnings:  1 packages succeeded with warnings.                                                                                                                  
[build]   Abandoned: 94 packages were abandoned.                                                                                                                          
[build]   Failed:    1 packages failed.                                                                                                                                   
[build] Runtime: 9.7 seconds total.  
```

mikepurvis, 2016-07-25:
Wow, bad luck. I don't think most people using ros-install-osx run into this issue as they're on 2 or 4-core laptops, where build parallelism is not as heavy. Looks like you're on an 8-core machine, though, so you're hitting it much more heavily.

I think your options are:

- run the build with lower parallelism (eg `-j4` or `-j2`).
- install catkin_tools from my branch (`pip install git+https://github.com/mikepurvis/catkin_tools@installspace-mutexing --upgrade`) and try again.

lilouch, 2016-07-26:
Okey I first tried with lower parallelism (-j4 and -j2) by using 
  `catkin build --limit-status-rate 1 -j2`

```
I get this error : 
Starting  >>> filters                                                                                                                                                     
__________________________________________________________________________________________________________________________________________________________________________
Errors     << filters:cmake /Users/libraries/ros-install-osx/indigo_desktop_full_ws/logs/filters/build.cmake.000.log                                         
CMake Error at /opt/ros/indigo/share/catkin/cmake/test/tests.cmake:17 (message):
  add_rostest() is not available when tests are not enabled.  The CMake code
  should only use it inside a conditional block which checks that testing is
  enabled:

  if(CATKIN_ENABLE_TESTING)

    add_rostest(...)

  endif()

Call Stack (most recent call first):
  CMakeLists.txt:46 (add_rostest)

cd /Users/libraries/ros-install-osx/indigo_desktop_full_ws/build/filters; catkin build --get-env filters | catkin env -si  /usr/local/bin/cmake /Users/libraries/ros-install-osx/indigo_desktop_full_ws/src/filters --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/Users/libraries/ros-install-osx/indigo_desktop_full_ws/devel/.private/filters -DCMAKE_INSTALL_PREFIX=/opt/ros/indigo -DCATKIN_ENABLE_TESTING=0 -DCMAKE_BUILD_TYPE=Release -DPYTHON_LIBRARY=/usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib -DPYTHON_INCLUDE_DIR=/usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/include/python2.7; cd -
..........................................................................................................................................................................
Failed     << filters:cmake                                   [ Exited with code 1 ]                                                                                      
Failed    <<< filters                                         [ 2.6 seconds ]                                                                                             
Abandoned <<< message_filters                                 [ Unrelated job failed ]                                                                                    
Abandoned <<< rospy_tutorials                                 [ Unrelated job failed ]                                                                                    
Abandoned <<< topic_tools                                     [ Unrelated job failed ]                                                                                    
Abandoned <<< urdf                                            [ Unrelated job failed ]                                                                                    
Abandoned <<< xacro                                           [ Unrelated job failed ]                                                                                    
Abandoned <<< self_test                                       [ Unrelated job failed ]                                                                                    
Abandoned <<< pr2_description                                 [ Unrelated job failed ]                                                                                    
Abandoned <<< rosmsg                                          [ Unrelated job failed ]                                                                                    
Abandoned <<< rosservice                                      [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_action                                      [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_bag                                         [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_bag_plugins                                 [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_console                                     [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_dep                                         [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_graph                                       [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_logger_level                                [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_moveit                                      [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_msg                                         [ Unrelated job failed ]                                                                                    
Abandoned <<< rqt_nav_view                                    [ Unrelated job failed ]                                                                                    
[... cut ...]
[build]   Abandoned: 84 packages were abandoned.                                                                                                                          
[build]   Failed:    1 packages failed.                                                                                                                                   
[build] Runtime: 4 minutes and 44.9 seconds total.                                                                                                                        
[build] Note: Workspace packages have changed, please re-source setup files to use them.
```

spmaniato, 2016-07-26:
I'm getting the `CATKIN_ENABLE_TESTING` related error as well:

```
CMake Error at /opt/ros/indigo/share/catkin/cmake/test/tests.cmake:17 (message):
  add_rostest() is not available when tests are not enabled.  The CMake code
  should only use it inside a conditional block which checks that testing is
  enabled:

  if(CATKIN_ENABLE_TESTING)

    add_rostest(...)

  endif()

Call Stack (most recent call first):
  CMakeLists.txt:46 (add_rostest)
```

I'll try without the recently added flags and report back.

spmaniato, 2016-07-26:
@lilouch For me, the `CATKIN_ENABLE_TESTING` issue above was resolved after `catkin clean` and then `catkin config / build` _without_ the `-DCATKIN_ENABLE_TESTING=0` flag.

mikepurvis, 2016-07-26:
Yes, that was preemptive. Dropped here: 2f8008be0b30f0c65c8937088147137208214043

lilouch, 2016-07-28:
Okey I'm now able to compile 179 over 194 packages. I have this error now : 

```
Starting  >>> stage_ros                                                                                                                                   
__________________________________________________________________________________________________________________________________________________________
Errors     << pcl_ros:make /Users/libraries/ros-install-osx/indigo_desktop_full_ws/logs/pcl_ros/build.make.004.log                           
ld: library not found for -lvtkFiltersPython
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [/Users/libraries/ros-install-osx/indigo_desktop_full_ws/devel/.private/pcl_ros/lib/pcl_ros/convert_pointcloud_to_image] Error 1
make[1]: *** [CMakeFiles/convert_pointcloud_to_image.dir/all] Error 2
make: *** [all] Error 2
cd /Users/libraries/ros-install-osx/indigo_desktop_full_ws/build/pcl_ros; catkin build --get-env pcl_ros | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..........................................................................................................................................................
Failed     << pcl_ros:make                                    [ Exited with code 2 ]                                                                      
Failed    <<< pcl_ros                                         [ 0.4 seconds ]                                                                             
Abandoned <<< tf_conversions                                  [ Unrelated job failed ]                                                                    
Abandoned <<< turtle_tf                                       [ Unrelated job failed ]                                                                    
Abandoned <<< collada_urdf                                    [ Unrelated job failed ]                                                                    
Abandoned <<< pointcloud_to_laserscan                         [ Unrelated job failed ]                                                                    
Abandoned <<< rqt_robot_dashboard                             [ Unrelated job failed ]                                                                    
Abandoned <<< interactive_marker_tutorials                    [ Unrelated job failed ]                                                                    
Abandoned <<< laser_assembler                                 [ Unrelated job failed ]                                                                    
Abandoned <<< laser_filters                                   [ Unrelated job failed ]                                                                    
Abandoned <<< gazebo_plugins                                  [ Unrelated job failed ]                                                                    
Abandoned <<< robot_state_publisher                           [ Unrelated job failed ]                                                                    
Abandoned <<< rviz                                            [ Unrelated job failed ]                                                                    
Abandoned <<< librviz_tutorial                                [ Unrelated job failed ]                                                                    
Abandoned <<< rqt_rviz                                        [ Unrelated job failed ]                                                                    
Abandoned <<< rviz_plugin_tutorials                           [ Unrelated job failed ]                                                                    
Abandoned <<< rviz_python_tutorial                            [ Unrelated job failed ]                                                                    
Abandoned <<< urdf_tutorial                                   [ Unrelated job failed ]                                                                    
Finished  <<< gazebo_ros                                      [ 2.3 seconds ]                                                                             
Finished  <<< laser_geometry                                  [ 0.7 seconds ]                                                                             
Finished  <<< stage_ros                                       [ 1.3 seconds ]                                                                             
[build] Summary: 177 of 194 packages succeeded.                                                                                                           
[build]   Ignored:   31 packages were skipped or are blacklisted.                                                                                         
[build]   Warnings:  10 packages succeeded with warnings.                                                                                                 
[build]   Abandoned: 16 packages were abandoned.                                                                                                          
[build]   Failed:    1 packages failed.                                                                                                                   
[build] Runtime: 1 minute and 47.9 seconds total.      
```              

It says that vtkfilters is not found but I checked and I have well installed VTK-7.0 from the website. 
If I write `"locate vtkfiltersPython" ` on the command line, I get : 
```
[... cut ...]
/usr/local/lib/python2.7/site-packages/vtk/vtkFiltersPythonPython.so
```

I suspect that actually It's not **vtkFiltersPython** but **vtkFiltersPythonPython**. What do you think ?

I checked also my path variable `echo $PATH`
`/usr/local/bin:/usr/local/sbin:/opt/local/bin:/opt/local/sbin:/opt/local/bin:/opt/local/sbin:/opt/local/bin:/opt/local/sbin:/usr/local/bin:/usr/local/bin:/usr/bin:/bin:/usr/sbin:/sbin:/opt/X11/bin:/Library/TeX/texbin:/usr/local/lib`

and my LIBRARY_PATH variable : 
`/usr/local/lib`

We are almost there !
Thank in advance.

toyoxx, 2017-11-08:
Sorry to hijack this thread, but I think it might be important. 

I was trying to install ROS using the script, failing miserably several times. In this thread I found the answer to my problems, which was to run the build with lower parallelism (-j2). I think that it could be good, in order to reduce the risk of the script failing, to lower the parallelism directly in the install script. 

I'm not sure whether there would be any negative effects from doing this, other than slowing the build speed. I'd prefer a slower, more reliable script than one faster but that fails most of the time.