mikepurvis/ros-install-osx #58: Failure installing pyqwt

bhaskara, 2016-09-20:
See https://github.com/Homebrew/homebrew-core/pull/4850.  This makes the install fail.  After adding the boneyard tap I still got a build failure:

`==> Installing pyqwt from homebrew/boneyard
==> Installing dependencies for homebrew/boneyard/pyqwt: qwt
==> Installing homebrew/boneyard/pyqwt dependency: qwt
==> Downloading https://homebrew.bintray.com/bottles/qwt-6.1.3.el_capitan.bottle.1.tar.gz
########################################################################################################### 100.0%
==> Pouring qwt-6.1.3.el_capitan.bottle.1.tar.gz
🍺  /usr/local/Cellar/qwt/6.1.3: 1,668 files, 18.9M
==> Installing homebrew/boneyard/pyqwt
==> Downloading https://downloads.sourceforge.net/project/pyqwt/pyqwt5/PyQwt-5.2.0/PyQwt-5.2.0.tar.gz
==> Downloading from http://pilotfiber.dl.sourceforge.net/project/pyqwt/pyqwt5/PyQwt-5.2.0/PyQwt-5.2.0.tar.gz
########################################################################################################### 100.0%
==> Patching
patching file configure/configure.py
==> python configure.py --module-install-path=/usr/local/Cellar/pyqwt/5.2.0/lib/python2.7/site-packages/PyQt4/Qwt5
==> make install
Last 15 lines from /Users/bhaskara/Library/Logs/Homebrew/pyqwt/02.make:
2016-09-20 08:45:57 -0700

make
install

clang++ -c -pipe -stdlib=libc++ -mmacosx-version-min=10.7 -fPIC -O2 -Wall -W -DNDEBUG -DHAS_NUMPY -DQT_NO_DEBUG -DQT_CORE_LIB -DQT_GUI_LIB -I. -I/usr/local/Cellar/sip/4.18.1/include -I/usr/local/lib/python2.7/site-packages/numpy/c
ore/include -I/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.11.sdk/System/Library/Frameworks/Python.framework/Versions/2.7/include/python2.7 -I/usr/local/opt/qt/mkspecs/default -I/usr
/local/opt/qt/lib/QtCore.framework/Headers -I/usr/local/opt/qt/lib/QtGui.framework/Headers -I/usr/local/opt/qt/include -F/usr/local/opt/qt/lib -o sip_iqtcmodule.o sip_iqtcmodule.cpp                                                
In file included from sip_iqtcmodule.cpp:7:
In file included from ./sipAPI_iqt.h:11:
/usr/local/Cellar/sip/4.18.1/include/sip.h:32:10: fatal error: 'Python.h' file not found
#include <Python.h>
         ^
1 error generated.
make[1]: *** [sip_iqtcmodule.o] Error 1
make: *** [install] Error 2

READ THIS: https://git.io/brew-troubleshooting
pyqwt was moved to homebrew-boneyard because it has unfixable issues.
Please do not file any issues about this. Sorry!

`

mikepurvis, 2016-09-20:
If it truly is a dependency, we'll have to move the formula to https://github.com/ros/homebrew-deps

However, I'm not seeing what it is which actually depends on this— the rosdep key is `python-qt-bindings-qwt5`, and I see no instances of that at all in the indigo rosdistro cache. So maybe it's a transient dependency of something, Gazebo perhaps?

@wjwwood Any insight to offer here?

wjwwood, 2016-09-20:
It's optionally used by `rqt_plot` I believe. I'm guessing that the above error is because Python isn't installed? I don't know.

bhaskara, 2016-09-20:
Python is certainly installed :)  However, yes, I simply skipped that key in the rosdep update and it didn't seem to affect the build.

mikepurvis, 2016-09-20:
Would like to figure out how it's getting pulled in, but in the meantime, please send a PR for the skip.

closed 2016-09-20 by mikepurvis