mikepurvis/ros-install-osx #63: Cannot install on macOS Sierra

kubark42, 2016-09-30:
It looks like `qt` might be a blocker for ROS Indigo on macOS Sierra.

```
executing command [brew install qt]
qt: This formula either does not compile or function as expected on macOS versions newer than El Capitan due to an upstream incompatibility.
Error: An unsatisfied requirement failed this build.
ERROR: the following rosdeps failed to install
  homebrew: command [brew install qt] failed
```

Is there a workaround for not installing `qt` at all?

mikepurvis, 2016-09-30:
See: https://github.com/Homebrew/homebrew-core/blob/master/Formula/qt.rb#L33-L36

Per REP-003, Kinetic's requirement is for QT5: http://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021

So perhaps try installing desktop_full for Kinetic instead, and see if there's anything still pulling in `qt` instead of `qt5`? A rosdep override should take care of that if so.

kubark42, 2016-10-03:
This would involve changing https://github.com/mikepurvis/ros-install-osx/blob/77bf795d06930d3456b640c40b900c1dc892b70d/install#L25 from 

```
ROS_DISTRO=${ROS_DISTRO:-indigo}
```
to
```
ROS_DISTRO=${ROS_DISTRO:-kinetic}
```
right?

mikepurvis, 2016-10-03:
Just `export ROS_DISTRO=kinetic` in your terminal prior to invoking the script.

kubark42, 2016-10-03:
Hmm... It seems that brew Qt is just broken right now on macOS: https://github.com/Homebrew/homebrew-core/issues/5391

I already have Qt5.7 installed system-wide, is there a way to modify the script to use that Qt instead of installing its own? This is preferable on many fronts, since it's really best _not_ to have ROS pull in a system-wide package which will compete with an already-installed version.

mikepurvis, 2016-10-03:
We'll follow whatever Homebrew does (which has conventionally been not to dupe system stuff, but that seems not to be the policy so much these days).

kubark42, 2016-10-03:
So is this an intractable problem, then?

- brewed Qt4 is intentionally removed on >=10.12
- brewed Qt5 is broken
- ROS can't be pointed at local installations of Qt
- ROS can't be installed without Qt

Is there an alternate route I'm missing here?

kubark42, 2016-10-03:
P.S. It looks like Indigo with Qt4 might be doable yet: https://github.com/Homebrew/homebrew-core/pull/5216/files#diff-659a13040ca5049988275ef8f03dd6e9L33

mikepurvis, 2016-10-04:
You could also set `ROS_CONFIGURATION=ros_base` to avoid all the GUI/sim packages for now. That isn't a long-term solution, but may help with bootstrapping in the short term.

kubark42, 2016-10-04:
Okay, rosdep successfully completed on Indigo with a combination of `brew install cartr/qt4/qt` (from https://github.com/Homebrew/homebrew-core/pull/5216#issuecomment-251478385) and an update to `log4cxx` (see https://github.com/Homebrew/homebrew-core/issues/5599#issuecomment-251506155).

`catkin build --limit-status-rate 1` is running right now, I'll report back here once it's done.

kubark42, 2016-10-04:
@mikepurvis, I've successfully resolved this problem. I don't know if you want to close this now, or leave it open until all the upstream fixes take hold. It might also make sense to check for macOS Sierra and call `brew install cartr/qt4/qt` before doing `rosdep` so that it grabs the right one.

mikepurvis, 2016-10-04:
Let's hold it open until things are fixed by default. I'd be delighted for a PR that does what you describe.

MikeMcQuaid, 2016-10-05:
> brewed Qt4 is intentionally removed on >=10.12

To clarify on this: Qt is unsupported on 10.12 by the creators of Qt, not just Homebrew. If they continued to support it on 10.12 we'd do so too. It will likely never have another bugfix or security release and there's known security vulnerabilities in Qt 4's WebKit.

mikepurvis, 2016-10-05:
@MikeMcQuaid Makes sense. As above, ROS Kinetic is _supposed_ to [depend on QT 5.3](http://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021), so it's probably just a matter of working with @dirk-thomas, @wjwwood, and the other OSRF staff who maintain core ROS packagesto understand where the breakages are and correct them.

kubark42, 2016-10-07:
@mikepurvis Sure, but my concern with rerunning this script is that it might break things. ROS on macOS works for me, but only kinda-just. I've had to make subtle adjustments in certain things, and am currently playing around the Apple system `python`, because the brew'ed python has at the past been at the root of several problems which I am currently having.

However, here's what I imagine the patch would look like. I think it would land somewhere around https://github.com/mikepurvis/ros-install-osx/blob/77bf795d06930d3456b640c40b900c1dc892b70d/install#L97.

```
  # Check for macOS Sierra
  if [[ `sw_vers -productVersion` == *"10.12"* ]]
  then
    echo "Qt4 is no longer officially supported by brew. Installing an unofficial workaround, which disables Qt's Phonon. "
    echo "See https://github.com/Homebrew/homebrew-core/pull/5216#issuecomment-251478385 for more information."
    brew install cartr/qt4/qt
  fi
```

NunchakusLei, 2017-01-31:
I tried ```brew install cartr/qt4/qt``` and updated ```log4cxx```, but the error

Failed to detect successful installation of [qt]

keep poping up. I there anything else I can do to install indigo on Mac Serria?

burf2000, 2017-04-12:
Has anyone managed to get ROS working on OSX?

kubark42, 2017-04-12:
I have parts of it working, but it's always a brittle installation. There are things which work, but the end result was that I had to abandon ROS on macOS because it was still a little early. Keeping fingers crossed that future versions will have even wider cross-platform support.

mikepurvis, 2017-04-12:
There was a period where the Yosemite/Indigo installation worked pretty well.

Given that Homebrew has now abandoned QT4, I think the only reasonable path forward here is to dump Indigo, and focus on providing a QT5/Gazebo7/Kinetic solution.

I just haven't had any time to commit to it.

ahundt, 2017-05-12:
El Capitan has worked okay too... I'm waiting on this to update my mac, haha.

Beck-Sisyphus, 2017-09-15:
Sierra still doesn't have Qt supported yet. Don't want to try the work around yet, looking forward for the Kinetic for MacOS updates.

mikepurvis, 2017-10-20:
Kinetic/Lunar can now be built at least on El Cap. Please open new tickets for remaining issues connected specifically to Sierra+.

closed 2017-10-20 by mikepurvis

ahundt, 2017-10-21:
Yay I can update my OS version! Has the updated script version worked for anyone else?

jmtatsch, 2017-10-21:
Works for me on high sierra.