mikepurvis/ros-install-osx #64: pcl_ros acting up in MacOS 10.12

eric1221bday, 2016-10-01:
So I've looked around and saw this problem pop up with El Capitan, but it seems to have been fixed with pcl 1.8. Now it's popping up again. When I try to build pcl_ros I get the error

Starting  >>> pcl_ros
____________________________________________________________________________________________________________________________________________________________________________________
Warnings   << pcl_ros:cmake /Users/eric1221bday/indigo_desktop_full_ws/logs/pcl_ros/build.cmake.000.log
** WARNING ** io features related to openni will be disabled
** WARNING ** io features related to openni2 will be disabled
** WARNING ** io features related to ensenso will be disabled
** WARNING ** io features related to davidSDK will be disabled
** WARNING ** io features related to dssdk will be disabled
** WARNING ** io features related to pcap will be disabled
** WARNING ** io features related to png will be disabled
** WARNING ** io features related to libusb-1.0 will be disabled
** WARNING ** visualization features related to openni will be disabled
** WARNING ** visualization features related to openni2 will be disabled
** WARNING ** visualization features related to ensenso will be disabled
** WARNING ** visualization features related to davidSDK will be disabled
** WARNING ** visualization features related to dssdk will be disabled
** WARNING ** visualization features related to rssdk will be disabled
** WARNING ** apps features related to openni will be disabled
CMake Warning at /opt/ros/indigo/share/catkin/cmake/test/nosetests.cmake:98 (message):
  nosetests not found, Python tests can not be run (try installing package
  'python-nose')
Call Stack (most recent call first):
  /opt/ros/indigo/share/catkin/cmake/all.cmake:147 (include)
  /opt/ros/indigo/share/catkin/cmake/catkinConfig.cmake:51 (include)
  CMakeLists.txt:11 (find_package)

CMake Warning (dev) at CMakeLists.txt:80 (add_dependencies):
  Policy CMP0046 is not set: Error on non-existent dependency in
  add_dependencies.  Run "cmake --help-policy CMP0046" for policy details.
  Use the cmake_policy command to set the policy and suppress this warning.

  The dependency target "pcl_ros_generate_messages_cpp" of target
  "pcl_ros_tf" does not exist.
This warning is for project developers.  Use -Wno-dev to suppress it.

cd /Users/eric1221bday/indigo_desktop_full_ws/build/pcl_ros; catkin build --get-env pcl_ros | catkin env -si  /usr/local/bin/cmake /Users/eric1221bday/indigo_desktop_full_ws/src/perception_pcl/pcl_ros --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/Users/eric1221bday/indigo_desktop_full_ws/devel/.private/pcl_ros -DCMAKE_INSTALL_PREFIX=/opt/ros/indigo -DCATKIN_ENABLE_TESTING=1 -DCMAKE_BUILD_TYPE=Release -DPYTHON_LIBRARY=/usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib -DPYTHON_INCLUDE_DIR=/usr/local/Cellar/python/2.7.12/Frameworks/Python.framework/Versions/2.7/include/python2.7; cd -
....................................................................................................................................................................................
____________________________________________________________________________________________________________________________________________________________________________________
Errors     << pcl_ros:make /Users/eric1221bday/indigo_desktop_full_ws/logs/pcl_ros/build.make.000.log
ld: library not found for -lpcl_io
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [/Users/eric1221bday/indigo_desktop_full_ws/devel/.private/pcl_ros/lib/pcl_ros/pointcloud_to_pcd] Error 1
make[1]: *** [CMakeFiles/pointcloud_to_pcd.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
ld: library not found for -lpcl_io
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [/Users/eric1221bday/indigo_desktop_full_ws/devel/.private/pcl_ros/lib/pcl_ros/convert_pointcloud_to_image] Error 1
make[1]: *** [CMakeFiles/convert_pointcloud_to_image.dir/all] Error 2
ld: library not found for -lpcl_io
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [/Users/eric1221bday/indigo_desktop_full_ws/devel/.private/pcl_ros/lib/pcl_ros/pcd_to_pointcloud] Error 1
make[1]: *** [CMakeFiles/pcd_to_pointcloud.dir/all] Error 2
make: *** [all] Error 2

spmaniato, 2016-11-19:
I'm getting a new compilation error today (after `brew upgrade`) and fresh`.rosinstall` file:

```
Errors     << pcl_ros:make /Users/smaniato/Projects/Robotics/indigo_desktop_full_ws/logs/pcl_ros/build.make.000.log
In file included from /Users/smaniato/Projects/Robotics/indigo_desktop_full_ws/src/perception_pcl/pcl_ros/tools/convert_pointcloud_to_image.cpp:42:
In file included from /Users/smaniato/Projects/Robotics/indigo_desktop_full_ws/src/ros_comm/roscpp/include/ros/ros.h:38:
In file included from /Users/smaniato/Projects/Robotics/indigo_desktop_full_ws/src/roscpp_core/rostime/include/ros/time.h:53:
In file included from /Users/smaniato/Projects/Robotics/indigo_desktop_full_ws/src/roscpp_core/cpp_common/include/ros/platform.h:55:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/string:436:
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cstring:79:9: error: no member named 'strcoll' in the global namespace; did you mean 'strtoll'?
using ::strcoll;
      ~~^
```

Might be coincidental that it came up during `pcl_ros` Or maybe not.

PS. I'm on macOS `10.11` ⚠️

lloydc99, 2016-11-20:
I have the same problem in pcl_ros and similar problems in several other projects.

```
Errors     << image_proc:make /Users/clloyd/ros-install-t2/indigo_desktop_full_ws/logs/image_proc/build.make.004.log   
 In file included from /Users/clloyd/ros-install-t2/indigo_desktop_full_ws/src/image_pipeline/image_proc/src/libimage_proc/processor.cpp:34:
 In file included from /Users/clloyd/ros-install-t2/indigo_desktop_full_ws/src/image_pipeline/image_proc/include/image_proc/processor.h:37:
 In file included from /usr/local/include/opencv2/core/core.hpp:53:
 In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/algorithm:626:
 /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cstring:79:9: error: no member named 'strcoll' in the global namespace; did you mean 'strtoll'?
using ::strcoll;

```

```
Errors     << image_view:make /Users/clloyd/ros-install-t2/indigo_desktop_full_ws/logs/image_view/build.make.004.log   
In file included from /Users/clloyd/ros-install-t2/indigo_desktop_full_ws/src/image_pipeline/image_view/src/nodes/disparity_view.cpp:34:
In file included from /opt/ros/indigo/include/ros/ros.h:38:
In file included from /opt/ros/indigo/include/ros/time.h:53:
In file included from /opt/ros/indigo/include/ros/platform.h:55:
In file included from /Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/string:436:
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1/cstring:79:9: error: no member named 'strcoll' in the global namespace; did you mean 'strtoll'?
using ::strcoll;

```

spmaniato, 2016-11-20:
@lloydc99 What are your macOS and Xcode versions?

PS. I'm also getting the `no member named '<something>' in the global namespace` when compiling other modules as well. Basically I can only build the `ros_base` configuration without these issues. Maybe we should open a standalone issue about this.

lloydc99, 2016-11-20:
Sorry,

Should have included:
  macOS 10.12.1
  Xcode Version 8.1 (8B62)

I agree this is probably a new and separate issue.

toyoxx, 2016-11-21:
I had this exact problem, but I had it on common_tutorials first. I headed there and they pointed me that it was a problem in `class_loader` package. I then moved the thread there but so far no comments or replies. I gave up compiling on Sierra for the time being. 

I tried switching from libc++ to libstdc++, and also switching the compilers. None of these worked. 

I'll keep trying different approaches and will let you know if I find a way to solve this.

Thanks

lloydc99, 2016-11-21:
Thanks for the update.

I'm looking at cmake options based on this very old post.
I have not had a chance to follow it all the way through.

http://forum.openscenegraph.org/viewtopic.php?t=3007&view=previous

spmaniato, 2016-11-21:
@toyoxx I don't think the macOS version is the problem since I'm on El Capitan (`10.11.6`) and still getting the same errors. Are you on Xcode 8 too maybe? It looks like the common denominator 🤔

mikepurvis, 2016-11-21:
Has anyone seen this error while using the command line tools vs. the full XCode install? In the past, there have been cryptic errors that go away when users switch to the CLT compiler.

lloydc99, 2016-11-21:
I'm not sure what I should be using. Does this help?
$ xcode-select -p
/Library/Developer/CommandLineTools

spmaniato, 2016-11-21:
I was definitely using command line tools before the Xcode 8 upgrade. But even now, I get:

```shell
$ gcc --version
Configured with: --prefix=/Library/Developer/CommandLineTools/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 7.3.0 (clang-703.0.31)
Target: x86_64-apple-darwin15.6.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```

@mikepurvis Digging a bit deeper, for me the errors are coming from:

```
/Library/Developer/CommandLineTools/usr/bin/../include/c++/v1
```

whereas for @toyoxx ([link to his gist](https://gist.github.com/toyoxx/8a8123ecf5a0ee7828a3f5745e125697)) they are coming from:

```
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1
```

knorth55, 2016-12-04:
I got same error macOS sierra, but I updated Xcode, and build was succeeded.
I think i switched to use xcode command-line-tools (I'm not sure, but my installdir of gcc looks so)
```
$ gcc --version
Configured with: --prefix=/Applications/Xcode.app/Contents/Developer/usr --with-gxx-include-dir=/usr/include/c++/4.2.1
Apple LLVM version 8.0.0 (clang-800.0.42.1)
Target: x86_64-apple-darwin16.0.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
```

spaghetti-, 2017-01-05:
No errors on `10.12` with XCode 8 (latest), pcl 1.8. Can this be closed?

mikepurvis, 2017-10-20:
I'm getting a clean build at least on El Cap, but that's with `CMAKE_FIND_FRAMEWORK=LAST` added, as suggested in #73.

closed 2017-10-20 by mikepurvis