mikepurvis/ros-install-osx #67: rosconsole error: no template named 'vector' in namespace 'std' [El Capitan, Indigo]

spmaniato, 2016-10-08:
After a `brew upgrade`, I tried to reinstall ROS Indigo on my Macbook (OS X El Capitan `10.11.6`):

```
______________________________________________________________________________________________________________________________________________________________________________________________
Errors     << rosconsole:make /Users/smaniato/Projects/Robotics/indigo_desktop_full_ws/logs/rosconsole/build.make.000.log
In file included from /Users/smaniato/Projects/Robotics/indigo_desktop_full_ws/src_old/ros_comm/rosconsole/src/rosconsole/impl/rosconsole_log4cxx.cpp:36:
/Users/smaniato/Projects/Robotics/indigo_desktop_full_ws/src_old/ros_comm/rosconsole/include/ros/console.h:121:9: error: no template named 'vector' in namespace 'std'; did you mean 'boost::mpl::vector'?
typedef std::vector<TokenPtr> V_Token;
        ^~~~~~~~~~~
        boost::mpl::vector
/usr/local/include/boost/mpl/aux_/preprocessed/gcc/vector.hpp:313:8: note: 'boost::mpl::vector' declared here
struct vector
       ^
1 error generated.
make[2]: *** [CMakeFiles/rosconsole_log4cxx.dir/src/rosconsole/impl/rosconsole_log4cxx.cpp.o] Error 1
make[1]: *** [CMakeFiles/rosconsole_log4cxx.dir/all] Error 2
make: *** [all] Error 2
cd /Users/smaniato/Projects/Robotics/indigo_desktop_full_ws/build/rosconsole; catkin build --get-env rosconsole | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..............................................................................................................................................................................................
Failed     << rosconsole:make                                 [ Exited with code 2 ]
Failed    <<< rosconsole                                      [ 11.0 seconds ]
```

It seems related to this question from 2013: http://answers.ros.org/question/95056/building-rosconsole-osx-109 and the corresponding issue: ros/ros_comm#307

I tried appending `-stdlib=libstdc++` to `CMAKE_CXX_FLAGS` and `CMAKE_EXE_LINKER_FLAGS` but that caused other errors.

In case these are relevant:

```
local-name: ros_comm/rosconsole
uri: https://github.com/ros-gbp/ros_comm-release/archive/release/indigo/rosconsole/1.11.20-0.tar.gz
version: ros_comm-release-release-indigo-rosconsole-1.11.20-0
```

```
$ brew info log4cxx
log4cxx: stable 0.10.0 (bottled)
Library of C++ classes for flexible logging
https://logging.apache.org/log4cxx/index.html
/usr/local/Cellar/log4cxx/0.10.0 (184 files, 6.6M) *
  Built from source on 2016-01-28 at 17:43:56
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/log4cxx.rb
==> Dependencies
Build: autoconf ✔, automake ✔, libtool ✔
```

```
$ brew info boost
boost: stable 1.62.0 (bottled), HEAD
Collection of portable C++ source libraries
https://www.boost.org/
/usr/local/Cellar/boost/1.62.0 (12,405 files, 396.4M) *
  Poured from bottle on 2016-10-08 at 10:13:25
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/boost.rb
==> Dependencies
Optional: icu4c ✔
==> Requirements
Optional: mpi ✘
```

spmaniato, 2016-10-08:
Looks like downgrading Boost to version 1.59.0 (`homebrew/versions/boost159`) fixes the problem 🤔

I haven't tried 1.60.0 but I suspect it might have introduced the issue.

mikepurvis, 2016-10-13:
Could you supply a PR that installs Boost 159 from the tap in order to unbreak us for now?

lloydc99, 2016-10-23:
This error in console.h looks to be a coding problem.
The typedef that is failing uses the vector type without explicitly defining it.
Maybe someone cleaned up a bad include cycle somewhere else.
Could the workaround for this be to patch the file?

--- kinetic_desktop_full_ws/src/ros_comm/rosconsole/include/ros/console.h	2016-10-23 11:28:38.000000000 -0400
***************
*** 40,45 ****
--- 40,46 ----
  #include <cstdarg>
  #include <ros/macros.h>
  #include <map>
+ #include <vector>
  
  #ifdef ROSCONSOLE_BACKEND_LOG4CXX
  #include "log4cxx/level.h"

kubark42, 2016-11-06:
As per https://github.com/ros/ros_comm/commit/74674c49a64abbdcdd5aeea016997af2959c6f05, this is a build bug which actually needed fixing.

spmaniato, 2016-11-06:
> As per ros/ros_comm@74674c4, this is a build bug which actually needed fixing.

Awesome! And it looks like there's an open PR for the `indigo-devel` fix: ros/ros_comm#926

closed 2017-10-20 by mikepurvis