mikepurvis/ros-install-osx #68: rqt_plot issues on osx el-capitan

robert-preissl, 2016-10-11:
hello,

I have been following the installation instructions for el-cap. and I got ROS working nicely on my machine. thank you very much for putting this together.

one last piece I just cannot figure out is how to get rqt_plot to work on my machine.

running rqt_plot gives:

PluginManager._load_plugin() could not load plugin "rqt_plot/Plot":
Traceback (most recent call last):
  File "/opt/ros/indigo/lib/python2.7/site-packages/qt_gui/plugin_handler.py", line 99, in load
    self._load()
  File "/opt/ros/indigo/lib/python2.7/site-packages/qt_gui/plugin_handler_direct.py", line 54, in _load
    self._plugin = self._plugin_provider.load(self._instance_id.plugin_id, self._context)
  File "/opt/ros/indigo/lib/python2.7/site-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/indigo/lib/python2.7/site-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/indigo/lib/python2.7/site-packages/rqt_gui_py/ros_py_plugin_provider.py", line 60, in load
    return super(RosPyPluginProvider, self).load(plugin_id, plugin_context)
  File "/opt/ros/indigo/lib/python2.7/site-packages/qt_gui/composite_plugin_provider.py", line 71, in load
    instance = plugin_provider.load(plugin_id, plugin_context)
  File "/opt/ros/indigo/lib/python2.7/site-packages/rqt_gui/ros_plugin_provider.py", line 94, in load
    return class_ref(plugin_context)
  File "/opt/ros/indigo/lib/python2.7/site-packages/rqt_plot/plot.py", line 55, in __init__
    self._data_plot = DataPlot(self._widget)
  File "/opt/ros/indigo/lib/python2.7/site-packages/rqt_plot/data_plot/__init__.py", line 141, in __init__
    raise RuntimeError('No usable plot type found. Install at least one of: PyQtGraph, MatPlotLib (at least 1.1.0%s) or Python-Qwt5.' % version_info)
RuntimeError: No usable plot type found. Install at least one of: PyQtGraph, MatPlotLib (at least 1.1.0) or Python-Qwt5.

I tried installing / uninstalling several libraries and version (e.g. for MatPlotLib); but it seems not to work.

if someone has any ideas / hacks, thanks a lot!

robert-preissl, 2016-10-14:
I found the issue and posting here in case folks have similar troubles.
It seems that in the latest matplotlib things have changed around the "NavigationToolbar". as a result MatPlotLib could not be loaded. to fix I changed the code in "mat_data_plot.py" from "NavigationToolbar2QTAgg" to "NavigationToolbar2QT". and it works fine now.

closed 2016-10-14 by robert-preissl

mikepurvis, 2016-10-14:
Looks like the code should have accounted for that:

https://github.com/ros-visualization/rqt_common_plugins/blob/master/rqt_plot/src/rqt_plot/data_plot/mat_data_plot.py#L82-L85

Can you describe more about the issue you encountered, in case someone else hits it?

mikepurvis, 2016-10-14:
That change went in here: https://github.com/ros-visualization/rqt_common_plugins/blob/master/rqt_plot/src/rqt_plot/data_plot/mat_data_plot.py#L82-L85

@dirk-thomas Perhaps the check should be reversed, so we try the new name _first_? I'm suspecting that we're pulling in a compatibility API or something.

dirk-thomas, 2016-10-17:
The current code already catches the import error and should have fallen back to the suggested import. Therefore I don't see any reason to change the current code.

@robert-preissl If you could provide more information about which version of `matplotlib` you are using as well as the exact code of `rqt_plot` that might help identifying why it failed for you.

mikepurvis, 2016-10-18:
My argument is that `NavigationToolbar2QTAgg` and `NavigationToolbar2QT` are not exactly aliased to each other— one is a subclass. So it might be better behaviour to just import `NavigationToolbar2QT` without the fallback. See the changelog for 1.5.x: http://matplotlib.org/api/api_changes.html#remove-navigationtoolbar2qtagg

@robert-preissl Which version of matplotlib have you got on your system? Can you give us the first few lines of `pip show matplotlib` for you? I'm on 1.5.1 here.