mikepurvis/ros-install-osx #70: brew tap ros/deps produces warnings

ivoysey, 2016-10-17:
Following http://wiki.ros.org/kinetic/Installation/OSX/Homebrew/Source produces the following warnings, which it asserts that I should report!

```
% brew tap ros/deps
==> Tapping ros/deps
Cloning into '/usr/local/Homebrew/Library/Taps/ros/homebrew-deps'...
remote: Counting objects: 13, done.
remote: Compressing objects: 100% (12/12), done.
remote: Total 13 (delta 1), reused 4 (delta 0), pack-reused 0
Unpacking objects: 100% (13/13), done.
Warning: Calling Formula.sha1 is deprecated!
Use Formula.sha256 instead.
/usr/local/Homebrew/Library/Taps/ros/homebrew-deps/gfortran.rb:11:in `<class:Gfortran>'
Please report this to the ros/deps tap!

Warning: Calling SoftwareSpec#sha1 is deprecated!
Use SoftwareSpec#sha256 instead.
/usr/local/Homebrew/Library/Taps/ros/homebrew-deps/gfortran.rb:11:in `<class:Gfortran>'
Please report this to the ros/deps tap!

Warning: Calling Resource#sha1 is deprecated!
Use Resource#sha256 instead.
/usr/local/Homebrew/Library/Taps/ros/homebrew-deps/gfortran.rb:11:in `<class:Gfortran>'
Please report this to the ros/deps tap!

Warning: Calling BottleSpecification#sha1 is deprecated!
Use BottleSpecification#sha256 instead.
/usr/local/Homebrew/Library/Taps/ros/homebrew-deps/gfortran.rb:16:in `block in <class:Gfortran>'
Please report this to the ros/deps tap!

Warning: Calling BottleSpecification#sha1 is deprecated!
Use BottleSpecification#sha256 instead.
/usr/local/Homebrew/Library/Taps/ros/homebrew-deps/gfortran.rb:17:in `block in <class:Gfortran>'
Please report this to the ros/deps tap!

Warning: Calling BottleSpecification#sha1 is deprecated!
Use BottleSpecification#sha256 instead.
/usr/local/Homebrew/Library/Taps/ros/homebrew-deps/gfortran.rb:18:in `block in <class:Gfortran>'
Please report this to the ros/deps tap!

Tapped 9 formulae (46 files, 51.2K)
% 
```

mikepurvis, 2016-10-17:
These errors are benign, but if you'd be interested to fix them, I'm sure https://github.com/ros/homebrew-deps would accept a PR.

closed 2016-10-17 by mikepurvis

spmaniato, 2016-10-17:
FYI, they updated a bunch of formulas a while ago (ros/homebrew-deps#19) but not [gfortran.rb](https://github.com/ros/homebrew-deps/blob/master/gfortran.rb) for some reason. All of these warnings are coming from that one formula.