mikepurvis/ros-install-osx #71: Missing deps in homebrew

shomin, 2016-10-28:
Formulae for shiboken, pyside, and pyqwt have been moved to the boneyard in homebrew master.  This breaks a number of dependencies in the install.  It looks like they aren't planning on adding them back until they support QT5.  Namely, on instal (El Capitan), I get:

> Installing collected packages: pygraphviz
> Successfully installed pygraphviz-1.3.1
> Error: No available formula with the name "shiboken" 
> Error: No available formula with the name "pyside" 
> executing command [brew install shiboken]
> Error: No available formula with the name "shiboken" 
> ==> Searching for similarly named formulae...
> Error: No similarly named formulae found.
> ==> Searching taps...
> Error: No formulae found in taps.
> ERROR: the following rosdeps failed to install
> homebrew: command [brew install shiboken] failed

Shiboken in the boneyard:

> brew search shiboken
> No formula found for "shiboken".
> ==> Searching pull requests...
> Closed pull requests:
> shiboken: migrate to boneyard (https://github.com/Homebrew/homebrew-core/pull/5726)

Some info on plan to remove QT4:
https://github.com/Homebrew/homebrew-core/pull/5725

shomin, 2016-10-28:
Looks like @mikepurvis is already aware of these issues from #69.  Let me know if I should close this.

mikepurvis, 2016-10-28:
The new ticket makes sense to leave open IMO, since the problem is now a lot worse.

I think the only way forward in the short term is to create our own tap with qt4 and all the boneyarded stuff, and use that for ros-install-osx. Eventually when everything's QT5-friendly, we can see about getting back on the upstream formulae, but who knows what the timeline will be for that.

In theory, [ROS Kinetic supports QT 5](http://www.ros.org/reps/rep-0003.html#kinetic-kame-may-2016-may-2021), but I haven't had a chance to do any investigation of that claim personally.

mikepurvis, 2017-10-20:
We now have a QT 5 build, so shiboken/pyside are no longer needed.

closed 2017-10-20 by mikepurvis