mikepurvis/ros-install-osx #73: Xcode 8.1 on Sierra 10.12 has multiple "no member XXX in global namespace" errors

lloydc99, 2016-11-29:
This is forked from issue #64.
I get these errors compiling several packages:
```
Errors     << nodelet_tutorial_math:make /Users/clloyd/ros-install-t2/indigo_desktop_full_ws/logs/nodelet_tutorial_math/build.make.001.log                                                                        
In file included from /Users/clloyd/ros-install-t2/indigo_desktop_full_ws/src/common_tutorials/nodelet_tutorial_math/src/plus.cpp:30:
In file included from /opt/ros/indigo/include/pluginlib/class_list_macros.h:40:
In file included from /opt/ros/indigo/include/class_loader/class_loader.h:33:
In file included from /usr/local/include/boost/shared_ptr.hpp:17:
In file included from /usr/local/include/boost/smart_ptr/shared_ptr.hpp:23:
In file included from /usr/local/include/boost/config/no_tr1/memory.hpp:21:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/memory:610:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cstring:79:9: error: no member named 'strcoll' in the global namespace
using ::strcoll;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cstring:80:9: error: no member named 'strxfrm' in the global namespace
using ::strxfrm;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cstring:82:9: error: no member named 'memchr' in the global namespace; did you mean 'wmemchr'?
using ::memchr;
      ~~^
/usr/include/wchar.h:150:10: note: 'wmemchr' declared here
wchar_t *wmemchr(const wchar_t *, wchar_t, size_t);
         ^
```

I think I've created a minimal case to reproduce this problem.
In this case, it seems to be triggered by depending on the nodelet package.

CMakeLists.txt:

```
cmake_minimum_required(VERSION 2.8.3)
project(anyExecutable)

#This generates errors
find_package(catkin REQUIRED COMPONENTS nodelet )

#This compiles
#find_package(catkin REQUIRED )

## Setup include directories
include_directories(${catkin_INCLUDE_DIRS})
#add_definitions("-dM -E")

add_library(anyExecutable main.cpp)
target_link_libraries( anyExecutable ${catkin_LIBRARIES})

```

main.cpp:

```

#include <boost/shared_ptr.hpp>

int main(int , char** ) {

    boost::shared_ptr<int> blah;

    return (blah)?0:1;
}

```

spmaniato, 2016-11-29:
For the record, I'm on El Capitan (`10.11.6`) and still getting the same errors. I think Xcode `8.1` is the culprit but haven't tried downgrading yet (http://stackoverflow.com/a/14756127/2653356)

helenol, 2016-11-30:
I suspect boost 1.62 here. I'm still on Xcode 6.4 (Build version 6E35b) and get the same exact errors after upgrading to boost 1.62 for gazebo7.

helenol, 2016-11-30:
Spent a few more hours hitting my head against this; downgrading down to boost 1.60 (which my original install was based on) has not fixed the issues.
May have to TimeMachine my whole system back and see if I can somehow salvage my previous working ROS install.

lloydc99, 2016-12-01:
I switched the compiler options to output the preprocessed code from the minimal example.
When the nodelet component is included many of the include files switch from
/usr/include to /System/Library/Frameworks/Kernel.framework/Headers.

I've included copies of the output here.

[good_preproc.txt](https://github.com/mikepurvis/ros-install-osx/files/623496/good_preproc.txt)
[bad_preproc.txt](https://github.com/mikepurvis/ros-install-osx/files/623497/bad_preproc.txt)

helenol, 2016-12-01:
Got it working with my old .rosinstall file from back in last February, which I've attached for others. I also updated gazebo_ros to the latest version in that file, as this was necessary to get gazebo 7.4 to work.
As a disclaimer, I based this on a previous install and haven't tested in a complete clean one.

System setup:
El Cap 10.11.2
ROS indigo
XCode 6.4, but using Command Line Tools anyway.
gazebo7 + Boost 1.62.0.

Changes I had to make:
Patch this in: https://github.com/ros/ros_comm/pull/926 (from https://github.com/mikepurvis/ros-install-osx/issues/67 )
In `src/common_msgs/sensor_msgs/include/sensor_msgs/point_cloud_conversion.h`
add the `#include <boost/format.hpp>` header.
Cross fingers and hope.

[rosinstall.txt](https://github.com/mikepurvis/ros-install-osx/files/624740/rosinstall.txt)

I'm no closer to actually figuring out **what** causes these errors, but I suppose it isn't necessarily boost and not necessarily XCode. If someone has a chance to try with this super old rosinstall file and also gets it to work, then must just be something in the ROS code that changed recently..

spmaniato, 2016-12-01:
> I'm no closer to actually figuring out what causes these errors, but I suppose it isn't necessarily boost and not necessarily XCode. If someone has a chance to try with this super old rosinstall file and also gets it to work, then must just be something in the ROS code that changed recently.

If that's the case, then here's a potential culprit: ros/class_loader#44

I'm planning on downgrading `class_loader` to `0.3.4` and then attempting to build `desktop_full` again. I'll report my findings.

lloydc99, 2016-12-03:
I compiled the simple example in two separate build directories, one broken and one working and compared the files.

In anyExecutable.dir/flags.cmake the CXX_INCLUDES went from blank to a list of catkin built dirs.

`CXX_INCLUDES = -I/opt/ros/indigo/include -I/System/Library/Frameworks/Kernel.framework/Headers -I/usr/local/include -I/usr/local/Cellar/console_bridge/0.2.5/include`

I went to the bad config dir and ran the make command taking out the Kernel.framework include and
it compiled, when I added the Kernel.framework dir it had the error signature.

I'm looking to see where that directory gets added in cmake/catkin.

```
Chuck:build_bad $ /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"anyExecutable\" -DanyExecutable_EXPORTS -I/opt/ros/indigo/include  -I/usr/local/include -I/usr/local/Cellar/console_bridge/0.2.5/include  -fPIC   -o CMakeFiles/anyExecutable.dir/main.cpp.o -c /Users/clloyd/devel/boost_bug/src/main.cpp

Chuck:build_bad $ /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++   -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"anyExecutable\" -DanyExecutable_EXPORTS -I/opt/ros/indigo/include -I/System/Library/Frameworks/Kernel.framework/Headers -I/usr/local/include -I/usr/local/Cellar/console_bridge/0.2.5/include  -fPIC   -o CMakeFiles/anyExecutable.dir/main.cpp.o -c /Users/clloyd/devel/boost_bug/src/main.cpp
In file included from /Users/clloyd/devel/boost_bug/src/main.cpp:2:
In file included from /usr/local/include/boost/shared_ptr.hpp:17:
In file included from /usr/local/include/boost/smart_ptr/shared_ptr.hpp:23:
In file included from /usr/local/include/boost/config/no_tr1/memory.hpp:21:
In file included from /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/memory:610:
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cstring:79:9: error: no member named 'strcoll' in the global namespace
using ::strcoll;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cstring:80:9: error: no member named 'strxfrm' in the global namespace
using ::strxfrm;
      ~~^
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/../include/c++/v1/cstring:82:9: error: no member named 'memchr' in the global namespace; did you mean 'wmemchr'?
using ::memchr;
      ~~^
```

lloydc99, 2016-12-03:
Success!

If I change the default search method for OSX frameworks from FIRST to LAST, everything compiles. I'm testing a few things now, but I'll try to have a pull request this weekend.

```
  catkin config --install  --install-space ${ROS_INSTALL_DIR} --cmake-args \
    -DCMAKE_FIND_FRAMEWORK=LAST \
    -DCATKIN_ENABLE_TESTING=1 \
   ...
```

prclibo, 2017-01-17:
Met same problem and setting `CMAKE_FIND_FRAMEWORK ` does not work for me. Quick question @lloydc99 are you using compiler from Xcode or another e.g. gcc from brew?

lloydc99, 2017-01-17:
I'm using the Xcode compiler.
Are you able to reproduce the problem and fix with the minimal example above?

prclibo, 2017-01-17:
@lloydc99 
Ah... solved because I only delete `build_isolated` but forgot to delete the `devel_isolated` folder before recompile. I was too careless.

Thank you for the response!

westeast, 2017-05-22:
@lloydc99 
do you mean this when I compile pos on mac I should use this command 
`./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release  -DCMAKE_FIND_FRAMEWORK=LAST -DCATKIN_ENABLE_TESTING=1`

lloydc99, 2017-05-22:
Yes,

The -DCMAKE_FIND_FRAMEWORK=LAST is the part that fixes the bug.
The rest of the line is the normal invocation.

Hope it works for you!

closed 2017-10-20 by mikepurvis