mikepurvis/ros-install-osx #75: Error: No available formula with the name "pyqt"

mhhabdelwahab, 2016-12-04:
~/ros-install-osx/indigo_desktop_full_ws ~/ros-install-osx
Error: No available formula with the name "pyqt" 
executing command [brew install pyqt]
Error: No available formula with the name "pyqt" 
==> Searching for similarly named formulae...
This similarly named formula was found:
pyqt5 ✔
To install it, run:
  brew install pyqt5 ✔
==> Searching taps...
Error: No formulae found in taps.
ERROR: the following rosdeps failed to install
  homebrew: command [brew install pyqt] failed

Any Suggestions?

BrutusTT, 2017-02-02:
pyqt is deprecated and broken with brew. the formular for it you can find in the boneyard.

You can try:
brew tap homebrew/boneyard
brew install pyqt

see https://github.com/Homebrew/homebrew-boneyard/blob/master/README.md

mikepurvis, 2017-10-20:
Please see #84 for ideas and updates around QT5.

closed 2017-10-20 by mikepurvis