mikepurvis/ros-install-osx #76: is there a way to take care of uninstalling?

mark2n, 2016-12-15:
I don't know if there's a way to uninstall everything nicely? 
I'm not saying that I want to uninstall it, but just out of curiousity.

spmaniato, 2016-12-15:
@mark2n In terms of the ROS installation itself, it should be as simple as `rm -rf /opt/ros/$ROS_DISTRO` (e.g. `rm -rf /opt/ros/indigo`) And you probably wanna delete the catkin workspace as well, as it will contain `src`, `build`, `devel`, and `log` directories.

The Homebrew packages you would have to remove on your own I think.

mark2n, 2016-12-15:
@spmaniato that sounds reasonable. Thanks a lot!

mikepurvis, 2016-12-15:
For the ROS stuff itself, just delete `/opt/ros` and `/etc/ros`.

Cleaning up Homebrew is a trickier story. TBH you'd probably be best off nuking your Homebrew installation and just starting over with reinstalling only the packages you want.

We've talked at various points about mirroring all the homebrew formulae we depend on and installing them into a separate path, eg `/opt/ros/deps` as opposed to `/usr/local`. Homebrew maintainers tell me to go this route every time I complain about QT4/5 or other dependency issues— the big cost is that I think we'd have to give up upstream bottles, which would really suck.