mikepurvis/ros-install-osx #77: rosdep missing

BrutusTT, 2017-01-31:
Execution fails due to rosdep being not found.

This sudo prompt adds the the brewed python rosdep yaml to /etc/ros/rosdep/10-ros-install-osx.list.
Password:
./install: line 111: rosdep: command not found

Any ideas ?

spmaniato, 2017-01-31:
@BrutusTT Sounds like [L100](https://github.com/mikepurvis/ros-install-osx/blob/master/install#L100) did not complete successfully: 

```shell
pip install -U setuptools rosdep ...
```

What do you get if you manually type: `pip install -U rosdep` ? 🤔

BrutusTT, 2017-01-31:
myBook:ros-install-osx ingo$ pip install -U setuptools rosdep
Requirement already up-to-date: setuptools in /usr/local/lib/python2.7/site-packages
Requirement already up-to-date: rosdep in /usr/local/lib/python2.7/site-packages
Requirement already up-to-date: six>=1.6.0 in /usr/local/lib/python2.7/site-packages (from setuptools)
Requirement already up-to-date: appdirs>=1.4.0 in /usr/local/lib/python2.7/site-packages (from setuptools)
Requirement already up-to-date: packaging>=16.8 in /usr/local/lib/python2.7/site-packages (from setuptools)
Requirement already up-to-date: PyYAML>=3.1 in /usr/local/lib/python2.7/site-packages (from rosdep)
Requirement already up-to-date: rospkg>=1.0.37 in /usr/local/lib/python2.7/site-packages (from rosdep)
Requirement already up-to-date: rosdistro>=0.4.0 in /usr/local/lib/python2.7/site-packages (from rosdep)
Requirement already up-to-date: catkin-pkg in /usr/local/lib/python2.7/site-packages (from rosdep)
Requirement already up-to-date: pyparsing in /usr/local/lib/python2.7/site-packages (from packaging>=16.8->setuptools)
Requirement already up-to-date: argparse in /usr/local/lib/python2.7/site-packages (from catkin-pkg->rosdep)
Requirement already up-to-date: docutils in /usr/local/lib/python2.7/site-packages (from catkin-pkg->rosdep)
Requirement already up-to-date: python-dateutil in /usr/local/lib/python2.7/site-packages (from catkin-pkg->rosdep)

mikepurvis, 2017-10-20:
I believe this is due to the new split between brewed python (`python2`/`pip2`) and system python (`python`/`pip`). In any case, please follow #84 for ideas and updates.

closed 2017-10-20 by mikepurvis

plguhur, 2018-05-09:
I had the same issue. I had to re-install completely `python` with `homebrew` to solve this issue.