mikepurvis/ros-install-osx #78: Sierra Compatibility

burf2000, 2017-02-14:
Is there any plan to make this work on Sierra?  Most things fail.  Everytime I run it I get a different error

ERROR: Rosdep experienced an error: a bytes-like object is required, not 'str'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

burf2000, 2017-02-14:
So I changed to Kinetic, Everything installs ok until

ERROR: Rosdep experienced an error: a bytes-like object is required, not 'str'
Please go to the rosdep page [1] and file a bug report with the stack trace below.
[1] : http://www.ros.org/wiki/rosdep

Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/rosdep2/main.py", line 136, in rosdep_main
    exit_code = _rosdep_main(args)
  File "/usr/local/lib/python3.6/site-packages/rosdep2/main.py", line 353, in _rosdep_main
    return _package_args_handler(command, parser, options, args)
  File "/usr/local/lib/python3.6/site-packages/rosdep2/main.py", line 447, in _package_args_handler
    return command_handlers[command](lookup, packages, options)
  File "/usr/local/lib/python3.6/site-packages/rosdep2/main.py", line 654, in command_install
    installer.install(uninstalled, **install_options)
  File "/usr/local/lib/python3.6/site-packages/rosdep2/installers.py", line 485, in install
    verbose=verbose, quiet=quiet)
  File "/usr/local/lib/python3.6/site-packages/rosdep2/installers.py", line 515, in install_resolved
    command = installer.get_install_command(resolved, interactive=interactive, reinstall=reinstall, quiet=quiet)
  File "/usr/local/lib/python3.6/site-packages/rosdep2/platforms/osx.py", line 316, in get_install_command
    resolved = self.remove_duplicate_dependencies(resolved)
  File "/usr/local/lib/python3.6/site-packages/rosdep2/platforms/osx.py", line 347, in remove_duplicate_dependencies
    if brew_strip_pkg_name(other.package) == brew_strip_pkg_name(d):
  File "/usr/local/lib/python3.6/site-packages/rosdep2/platforms/osx.py", line 157, in brew_strip_pkg_name
    return package.split('/')[-1]
TypeError: a bytes-like object is required, not 'str'

burf2000, 2017-02-14:
It appears its using Python 3.6 instead of 2.7

serser, 2017-05-13:
I am also on Sierra, I followed instructions here http://wiki.ros.org/kinetic/Installation/OSX/Homebrew/Source with success, but still ros* commands are not there. Any idea?

anlif, 2017-05-22:
@serser did you source the setup.sh file?

``$ source ~/ros_catkin_ws/install_isolated/setup.bash``

mikepurvis, 2017-10-20:
Kinetic/Lunar can now be built at least on El Cap. Please open new tickets for remaining issues connected specifically to Sierra+.

closed 2017-10-20 by mikepurvis