mikepurvis/ros-install-osx #79: Failed to detect successful installation of [qt] - El Capitan

ryokochang, 2017-03-21:
Hi everyone, I've tried everything on the internet and it always happens so, does anyone have any ideas to solve this ??

~/ros-install-osx/indigo_desktop_full_ws/src/rviz/ogre_media/models ~/ros-install-osx/indigo_desktop_full_ws ~/ros-install-osx

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current

                                 Dload  Upload   Total   Spent    Left  Speed

100  5863  100  5863    0     0   4333      0  0:00:01  0:00:01 --:--:--  4333

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current

                                 Dload  Upload   Total   Spent    Left  Speed

100  2488  100  2488    0     0   2867      0 --:--:-- --:--:-- --:--:--  2866

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current

                                 Dload  Upload   Total   Spent    Left  Speed

100 11488  100 11488    0     0  14232      0 --:--:-- --:--:-- --:--:-- 14235

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current

                                 Dload  Upload   Total   Spent    Left  Speed

100 42238  100 42238    0     0  37617      0  0:00:01  0:00:01 --:--:-- 37611

~/ros-install-osx/indigo_desktop_full_ws ~/ros-install-osx

~/ros-install-osx/indigo_desktop_full_ws/src/catkin/cmake ~/ros-install-osx/indigo_desktop_full_ws ~/ros-install-osx

  % Total    % Received % Xferd  Average Speed   Time    Time     Time  Current

                                 Dload  Upload   Total   Spent    Left  Speed

100  6624  100  6624    0     0   7983      0 --:--:-- --:--:-- --:--:--  7990

~/ros-install-osx/indigo_desktop_full_ws ~/ros-install-osx

executing command [brew install qt]

Warning: qt5 is a keg-only and another version is linked to opt.

Use `brew install --force` if you want to install this version

ERROR: the following rosdeps failed to install

  homebrew: Failed to detect successful installation of [qt]

budekatude, 2017-05-22:
Hi.

I get a similar error:

```
~/indigo_desktop_full_ws ~
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
python_qt_binding: No definition of [python-qt-bindings] for OS [osx]
rviz: No definition of [libqt4-opengl-dev] for OS [osx]
gl_dependency: No definition of [python-qt-bindings-gl] for OS [osx]
webkit_dependency: No definition of [python-qt-bindings-webkit] for OS [osx]
qt_gui_cpp: No definition of [qt4-qmake] for OS [osx]
qt_gui: No definition of [python-qt-bindings] for OS [osx]
turtlesim: No definition of [libqt4] for OS [osx]
```
In a different thread somebody mentioned that homebrew removed all the qt4 related formulae. 
Is there any way around this or any kind of solution?

lujun723, 2017-05-24:
Same error,help

mikepurvis, 2017-10-20:
Please see #84 for an attempt to hack around these issues.

closed 2017-10-20 by mikepurvis