mikepurvis/ros-install-osx #80: rosdep keys unresolved

Dagrol, 2017-04-23:
I am getting the following error when I run the script:

ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
python_qt_binding: No definition of [python-qt-bindings] for OS [osx]
rviz: No definition of [libqt4-opengl-dev] for OS [osx]
gl_dependency: No definition of [python-qt-bindings-gl] for OS [osx]
qt_gui_cpp: No definition of [qt4-qmake] for OS [osx]
qt_gui: No definition of [python-qt-bindings] for OS [osx]
turtlesim: No definition of [libqt4] for OS [osx]

I tried to clean my rosdep and updated it but still no luck. Any ideas ?

mattcurf, 2017-04-27:
I see the same errors above.  Mac OS X 10.12.4

eaa3, 2017-05-16:
I appear to be having the exact same issue (OSX 10.11.6 El captain):

`ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
python_qt_binding: No definition of [python-qt-bindings] for OS [osx]
rviz: No definition of [libqt4-opengl-dev] for OS [osx]
gl_dependency: No definition of [python-qt-bindings-gl] for OS [osx]
webkit_dependency: No definition of [python-qt-bindings-webkit] for OS [osx]
qt_gui_cpp: No definition of [qt4-qmake] for OS [osx]
qt_gui: No definition of [python-qt-bindings] for OS [osx]
turtlesim: No definition of [libqt4] for OS [osx]`

ashwini, 2017-05-16:
Same issue here, 10.12

lujun723, 2017-05-24:
Mark, Same issue, 10.12

fitsir, 2017-05-24:
Mark, same issue, 10.12.5

ccckblaze, 2017-06-01:
Mark, Same issue, 10.12

ashishraste, 2017-06-17:
Ran across the same issue, it has to do with Homebrew not supporting qt4 formula, as discussed here: https://github.com/ros/rosdistro/issues/14416

mikepurvis, 2017-06-19:
Yes, the issue is with Homebrew no longer having qt4. The only reasonable path forward here is to migrate this project to qt5/Kinetic, which I haven't had the time to do.

Going to close this ticket for now in favour of #63.

closed 2017-06-19 by mikepurvis