mikepurvis/ros-install-osx #82: OSX Sierra install fails while building catkin workspace

jdcasale, 2017-08-15:
While following the [install guide](http://wiki.ros.org/kinetic/Installation/OSX/Homebrew/Source), at step 2.13, the call to ./src/catkin/bin/catkin_make_isolated --install -DCMAKE_BUILD_TYPE=Release fails with qt-related errors.  I have qt5 installed.

`CMake Error at /usr/local/Cellar/cmake/3.9.1/share/cmake/Modules/FindQt4.cmake:1318 (message):
  Found unsuitable Qt version "" from NOTFOUND, this code requires Qt 4.x
Call Stack (most recent call first):
  cmake/OpenCVFindLibsGUI.cmake:34 (find_package)
  CMakeLists.txt:556 (include)

-- Configuring incomplete, errors occurred!
See also "/Users/josh.casale/ros_catkin_ws/build_isolated/opencv3/install/CMakeFiles/CMakeOutput.log".
See also "/Users/josh.casale/ros_catkin_ws/build_isolated/opencv3/install/CMakeFiles/CMakeError.log".
<== Failed to process package 'opencv3':
  Command '['/Users/josh.casale/ros_catkin_ws/install_isolated/env.sh', 'cmake', '/Users/josh.casale/ros_catkin_ws/src/opencv3', '-DCMAKE_INSTALL_PREFIX=/Users/josh.casale/ros_catkin_ws/install_isolated', '-DCMAKE_BUILD_TYPE=Release', '-G', 'Unix Makefiles']' returned non-zero exit status 1

Reproduce this error by running:
==> cd /Users/josh.casale/ros_catkin_ws/build_isolated/opencv3 && /Users/josh.casale/ros_catkin_ws/install_isolated/env.sh cmake /Users/josh.casale/ros_catkin_ws/src/opencv3 -DCMAKE_INSTALL_PREFIX=/Users/josh.casale/ros_catkin_ws/install_isolated -DCMAKE_BUILD_TYPE=Release -G 'Unix Makefiles'`

teknolog2000, 2017-08-17:
Looks like you're trying to build Indigo on a newer Mac OS that doesn't have Qt4. Unless you strictly need Indigo, try building Kinetic by doing`ROS_DISTRO=kinetic ./install`.

You'll find yourself having other problems that I'm working my way through.

(It also looks like someone made [a tap](https://github.com/cartr/homebrew-qt4) of Qt4 for Sierra).

mikepurvis, 2017-10-20:
Please see #84 for an attempt to get this going again with QT5.

closed 2017-10-20 by mikepurvis