mikepurvis/ros-install-osx #85: Investigate Python 3

mikepurvis, 2017-10-20:
Right now, **ros-install-osx** uses brewed python2, but at some point we should see about switching this to use brewed python3, or at least having an option for it.

Also, because Homebrew's policy now is not to put a `python` or `pip` binary in the default path (`/usr/local/bin`), we'll have to monitor whether that's adequate or whether such an alias should be put into the bin directory of the resulting workspace.

eric1221bday, 2017-11-07:
So I'm here to report that while building and installing ROS just requires catkin to find the correct python version, there are many python scripts in ROS that simply assume a "python" executable especially in `#!/usr/bin/env pyhon` configuration. So at runtime many of the python scripts will not run correctly if the python executable is not symlinked to default path. I encountered this using the `spawn_model` script in `gazebo_ros`.

mikepurvis, 2017-11-07:
Everything that installs a script using `catkin_install_python` will be fine, but usage of that is not universal, as you've discovered. Certainly it would be great to push fixes into the relevant upstreams as they are discovered, but the other obvious thing would be to init a virtualenv inside the installspace, and then there'd be a `python` binary that can be whatever we want it to be.

mikepurvis, 2017-11-07:
PR up fro spawn_model: https://github.com/ros-simulation/gazebo_ros_pkgs/pull/621

mikepurvis, 2018-01-19:
Homebrew is moving their primary support to Python 3: https://brew.sh/2018/01/19/homebrew-1.5.0/

NikolausDemmel, 2018-04-05:
Some of the decisions have been taken back. In particular `/usr/local/bin/python` is still python2. Excerpt from current https://docs.brew.sh/Homebrew-and-Python:

> * python3 points to Homebrew’s Python 3.x (if installed)
> * python2 points to Homebrew’s Python 2.7.x (if installed)
> * python points to Homebrew’s Python 2.7.x (if installed) otherwise the macOS system Python. Check out brew info python if you wish to add Homebrew’s 3.x python to your PATH.
> * pip3 points to Homebrew’s Python 3.x’s pip (if installed)
> * pip and pip2 point to Homebrew’s Python 2.7.x’s pip (if installed)