mikepurvis/ros-install-osx #86: High Sierra support

jmtatsch, 2017-10-20:
Let's collect some roadblocks and solutions for building on OSX High Sierra:

- Issue: Dependency matplotlib fails to build
  Solution: https://github.com/Homebrew/homebrew-science/pull/6382

eric1221bday, 2017-11-07:
@jmtatsch Have you tried just using rqt and then selecting its image_view plugin? I was able to use image_view comfortably in High Sierra, though yes I did disable my SIP.

jmtatsch, 2017-11-07:
image_view started as rqt plugin works for me, even with SIP still on.
rqt_image_view not working directly is still an issue worthy tracking.
Does it work for you with SIP disabled?

eric1221bday, 2017-11-07:
`rqt_image_view` does indeed work for me with SIP disabled, so it looks like the workaround for SIP is not quite fullproof