mikepurvis/ros-install-osx #88: Build fails -- ld: library not found for -lassimp

chrisalbertson, 2017-10-21:
Building this on a MBP.  All  Apple software up to date.  I cleaned out everything in Homebrew before I started "dumped" every package.    The did ./install and after a couple hours got message quoted below.  Yes I can track down the library but I figured you might want to add this to your list of dependencies

******* QUOTE FOLLOWS *******

/usr/local/include/console_bridge/console.h:45:51: note: expanded from macro 'CONSOLE_BRIDGE_DEPRECATED'
#define CONSOLE_BRIDGE_DEPRECATED __attribute__ ((deprecated))
                                                  ^
2 warnings generated.
3 warnings generated.
ld: library not found for -lassimp
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [/Users/chris/ros-install-osx/lunar_desktop_full_ws/devel/.private/geometric_shapes/lib/libgeometric_shapes.dylib] Error 1
make[1]: *** [CMakeFiles/geometric_shapes.dir/all] Error 2
make: *** [all] Error 2
cd /Users/chris/ros-install-osx/lunar_desktop_full_ws/build/geometric_shapes; catkin build --get-env geometric_shapes | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
..............................................................................................................
Failed     << geometric_shapes:make                           [ Exited with code 2 ]                          
Failed    <<< geometric_shapes                                [ 15.8 seconds ]                                
Abandoned <<< rospy                                           [ Unrelated job failed ]                        
Abandoned <<< bondcpp                                         [ Unrelated job failed ]                        
Abandoned <<< camera_calibration_parsers                      [ Unrelated job failed ]                        
Abandoned <<< p

mikepurvis, 2017-10-23:
Looks like there's a related discussion here in https://github.com/ros-planning/geometric_shapes/issues/23.

I'm doubtful that just pre-installing `assimp` is going to be enough, though, given that we do have rosdep keys for that package:

https://github.com/ros/rosdistro/blob/39948946d46f1325806289a12914d6bdac55427c/rosdep/osx-homebrew.yaml#L5-L12

And there's the required dependency on it here:

https://github.com/ros-planning/geometric_shapes/blob/66352a742520f71e19a32139ec7463dcefe3b451/package.xml#L17

To clarify, did you find a path to success on 10.10? Was it just installing, or did you have do other stuff? (`brew link -f`, manipulating `DYLD_LIBRARY_PATH`, etc)

Anyway, thanks for trying this out and reporting your results!

chrisalbertson, 2017-10-23:
> Looks like there's a related discussion here in
> ros-planning/geometric_shapes#23
>
> I'm doubtful that just pre-installing assimp is going to be enough,
> though, given that we do have rosdep keys for that package:
>
> https://github.com/ros/rosdistro/blob/39948946d46f1325806289a12914d6
> bdac55427c/rosdep/osx-homebrew.yaml#L5-L12
>
> And there's the required dependency on it here:
>
> https://github.com/ros-planning/geometric_shapes/blob/
> 66352a742520f71e19a32139ec7463dcefe3b451/package.xml#L17
>
> To clarify, did you find a path to success on 10.10? Was it just
> installing, or did you have do other stuff? (brew link -f, manipulating
> DYLD_LIBRARY_PATH, etc)
>
> Anyway, thanks for trying this out and reporting your results!

I'm running macOS High Sierra 10.13

Before I started I completely removed everything to do with Home-brew but
did leave homebrew itself installed

Yes I see that it is not as simple as just finding a copy of libassimp

My fall back is easy, I run Linux in a VM on my mac.

I'll test any new version of this as it would be nice to have ROS native on
the Mac.

kdhansen, 2017-10-28:
I had the same problem, and I ended up editing the link.txt in the build folder, as proposed here: https://github.com/ros-planning/geometric_shapes/issues/23#issuecomment-284830221

However after changing c++ toolchain to CLT from the Xcode.app, it stopped complaining.

As a part of my efforts to make CMake find assimp automaically I added https://github.com/assimp/assimp/blob/master/cmake-modules/Findassimp.cmake to ls `/usr/local/share/cmake/Modules/` which may have something to do with it.

mikepurvis, 2017-11-20:
There's a ticket tracking this upstream, but it's likely to be a while before there's a solution from either assimp itself or the homebrew package: https://github.com/assimp/assimp/issues/1517

I'd be delighted for a PR which is able to make `ros-install-osx` do the right thing, ideally without hacking on files installed/owned by homebrew packages (though installing a temporary fork of assimp from your own tap is totally fair game— please just use the rosdep file to point at it).

BracketMaster, 2018-09-29:
So I changed the -lassimp argument to the location where brew installs assimp and change seems to find it just fine, but then I get a serious linking error concerning missing symbols.
`
[ 87%] Building CXX object CMakeFiles/geometric_shapes.dir/src/shape_extents.cpp.o
[100%] Linking CXX shared library /Users/yehowshuaimmanuel/src/ros-install-osx/lunar_desktop_full_ws/devel_isolated/geometric_shapes/lib/libgeometric_shapes.dylib
Undefined symbols for architecture x86_64:
  "Assimp::Importer::ReadFileFromMemory(void const*, unsigned long, unsigned int, char const*)", referenced from:
      shapes::createMeshFromBinary(char const*, unsigned long, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in mesh_operations.cpp.o
  "Assimp::Importer::SetPropertyInteger(char const*, int)", referenced from:
      shapes::createMeshFromBinary(char const*, unsigned long, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in mesh_operations.cpp.o
  "Assimp::Importer::ApplyPostProcessing(unsigned int)", referenced from:
      shapes::createMeshFromBinary(char const*, unsigned long, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in mesh_operations.cpp.o
  "Assimp::Importer::Importer()", referenced from:
      shapes::createMeshFromBinary(char const*, unsigned long, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in mesh_operations.cpp.o
  "Assimp::Importer::~Importer()", referenced from:
      shapes::createMeshFromBinary(char const*, unsigned long, Eigen::Matrix<double, 3, 1, 0, 3, 1> const&, std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&) in mesh_operations.cpp.o
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [/Users/yehowshuaimmanuel/src/ros-install-osx/lunar_desktop_full_ws/devel_isolated/geometric_shapes/lib/libgeometric_shapes.dylib] Error 1
make[1]: *** [CMakeFiles/geometric_shapes.dir/all] Error 2
make: *** [all] Error`