mikepurvis/ros-install-osx #90: Build fails: image_proc

rodrigoalmeida94, 2017-10-27:
Hello,

I am running the install script in macOS Sierra 10.12.6. I have done basically all the work-arounds that are mentioned in this issues, including the Xcode compilation as mentioned in [#4](https://github.com/mikepurvis/ros-install-osx/issues/4), but I can't get past building the image_proc package.

```
Errors     << image_proc:make /Users/rodrigoalmeida/ros-install-osx/lunar_desktop_full_ws/logs/image_proc/build.make.002.log
/Users/rodrigoalmeida/ros-install-osx/lunar_desktop_full_ws/src/image_pipeline/image_proc/src/nodelets/crop_decimate.cpp:121:66: error: no member named 'type' in 'cv::DataType<cv::Vec<unsigned char, 3> >'
  dst.create(src.rows / 2, src.cols / 2, cv::DataType<DstPixel>::type);
                                         ~~~~~~~~~~~~~~~~~~~~~~~~^
/Users/rodrigoalmeida/ros-install-osx/lunar_desktop_full_ws/src/image_pipeline/image_proc/src/nodelets/crop_decimate.cpp:237:7: note: in instantiation of function template specialization 'image_proc::debayer2x2toBGR<unsigned char>' requested here
      debayer2x2toBGR<uint8_t>(output.image, bgr, 0, 1, step, step + 1);
      ^
/Users/rodrigoalmeida/ros-install-osx/lunar_desktop_full_ws/src/image_pipeline/image_proc/src/nodelets/crop_decimate.cpp:121:66: error: no member named 'type' in 'cv::DataType<cv::Vec<unsigned short, 3> >'
  dst.create(src.rows / 2, src.cols / 2, cv::DataType<DstPixel>::type);
                                         ~~~~~~~~~~~~~~~~~~~~~~~~^
/Users/rodrigoalmeida/ros-install-osx/lunar_desktop_full_ws/src/image_pipeline/image_proc/src/nodelets/crop_decimate.cpp:245:7: note: in instantiation of function template specialization 'image_proc::debayer2x2toBGR<unsigned short>' requested here
      debayer2x2toBGR<uint16_t>(output.image, bgr, 0, 1, step, step + 1);
      ^
2 errors generated.
make[2]: *** [CMakeFiles/image_proc.dir/src/nodelets/crop_decimate.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/image_proc.dir/all] Error 2
make: *** [all] Error 2
cd /Users/rodrigoalmeida/ros-install-osx/lunar_desktop_full_ws/build/image_proc; catkin build --get-env image_proc | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
```

Any help would be greatly appreciated!

mikepurvis, 2017-10-27:
Looks like an OpenCV mismatch. Let's see your info on that package:

```
$ brew info opencv
opencv: stable 3.3.0 (bottled)
Open source computer vision library
https://opencv.org/
/usr/local/Cellar/opencv/3.3.0_3 (516 files, 125.0MB) *
  Poured from bottle on 2017-10-20 at 11:34:42
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/opencv.rb
==> Dependencies
Build: cmake ✔, pkg-config ✔
Required: eigen ✔, ffmpeg ✔, jpeg ✔, libpng ✔, libtiff ✔, openexr ✔, numpy ✔
==> Requirements
Required: python ✔, python3 ✔
```

rodrigoalmeida94, 2017-10-27:
Hey,

Here it is:

`opencv: stable 3.3.1 (bottled)
Open source computer vision library
https://opencv.org/
/usr/local/Cellar/opencv/3.3.0_3 (516 files, 122.9MB)
  Poured from bottle on 2017-10-24 at 14:45:58
/usr/local/Cellar/opencv/3.3.1 (519 files, 95.8MB) *
  Poured from bottle on 2017-10-26 at 15:38:22
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/opencv.rb
==> Dependencies
Build: cmake ✔, pkg-config ✔
Required: eigen ✔, ffmpeg ✔, jpeg ✔, libpng ✔, libtiff ✔, openexr ✔, numpy ✔
==> Requirements
Required: python ✔, python3 ✔`

mikepurvis, 2017-10-27:
Looks reasonable. Next step is confirming that that's the actual version that the ROS build is attempting to link against:

```
cd ~/ros-install-osx/lunar_desktop_full_ws/build/image_proc
source /opt/ros/lunar/setup.bash
VERBOSE=1 make
```

Paste the line that tries to build `crop_decimate.cpp.o`:

```
[ 60%] Building CXX object CMakeFiles/image_proc.dir/src/nodelets/crop_decimate.cpp.o
/Library/Developer/CommandLineTools/usr/bin/c++  -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"image_proc\" -Dimage_proc_EXPORTS -I/Users/mikepurvis/ros-install-osx/lunar_desktop_full_ws/devel/.private/image_proc/include -I/Users/mikepurvis/ros-install-osx/lunar_desktop_full_ws/src/image_pipeline/image_proc/include -I/opt/ros/lunar/include -I/opt/ros/lunar/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -isystem /usr/local/include -isystem /usr/local/include/opencv -I/usr/local/Cellar/console_bridge/0.3.2/include  -O3 -DNDEBUG -fPIC   -o CMakeFiles/image_proc.dir/src/nodelets/crop_decimate.cpp.o -c /Users/mikepurvis/ros-install-osx/lunar_desktop_full_ws/src/image_pipeline/image_proc/src/nodelets/crop_decimate.cpp
```

You want to confirm the `-isystem /usr/local/include/opencv` part, which demonstrates that you are in fact using the opencv from Homebrew and not some other one that's made its way onto your path.

kdhansen, 2017-10-27:
I face the same problem. My OpenCV, however, seems to be a tad newer:
```
$ brew info opencv
opencv: stable 3.3.1 (bottled)
Open source computer vision library
https://opencv.org/
/usr/local/Cellar/opencv/3.3.1 (519 files, 95.8MB) *
  Poured from bottle on 2017-10-26 at 11:08:56
From: https://github.com/Homebrew/homebrew-core/blob/master/Formula/opencv.rb
==> Dependencies
Build: cmake ✔, pkg-config ✔
Required: eigen ✔, ffmpeg ✔, jpeg ✔, libpng ✔, libtiff ✔, openexr ✔, numpy ✔
==> Requirements
Required: python ✔, python3 ✔
```

The line that builds `crop_decimate.cpp.o`:
```
[ 33%] Building CXX object CMakeFiles/image_proc.dir/src/nodelets/crop_decimate.cpp.o
/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/c++  -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"image_proc\" -Dimage_proc_EXPORTS -I/Users/kdh/ros-install-osx/lunar_desktop_full_ws/devel/.private/image_proc/include -I/Users/kdh/ros-install-osx/lunar_desktop_full_ws/src/image_pipeline/image_proc/include -I/opt/ros/lunar/include -I/opt/ros/lunar/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -isystem /usr/local/include -isystem /usr/local/include/opencv -I/usr/local/Cellar/console_bridge/0.3.2/include  -O3 -DNDEBUG -isysroot /Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.13.sdk -mmacosx-version-min=10.12 -fPIC   -o CMakeFiles/image_proc.dir/src/nodelets/crop_decimate.cpp.o -c /Users/kdh/ros-install-osx/lunar_desktop_full_ws/src/image_pipeline/image_proc/src/nodelets/crop_decimate.cpp
/Users/kdh/ros-install-osx/lunar_desktop_full_ws/src/image_pipeline/image_proc/src/nodelets/crop_decimate.cpp:121:66: error: no member named 'type' in
      'cv::DataType<cv::Vec<unsigned char, 3> >'
  dst.create(src.rows / 2, src.cols / 2, cv::DataType<DstPixel>::type);
                                         ~~~~~~~~~~~~~~~~~~~~~~~~^
/Users/kdh/ros-install-osx/lunar_desktop_full_ws/src/image_pipeline/image_proc/src/nodelets/crop_decimate.cpp:237:7: note: in instantiation of function template specialization
      'image_proc::debayer2x2toBGR<unsigned char>' requested here
      debayer2x2toBGR<uint8_t>(output.image, bgr, 0, 1, step, step + 1);
      ^
/Users/kdh/ros-install-osx/lunar_desktop_full_ws/src/image_pipeline/image_proc/src/nodelets/crop_decimate.cpp:121:66: error: no member named 'type' in
      'cv::DataType<cv::Vec<unsigned short, 3> >'
  dst.create(src.rows / 2, src.cols / 2, cv::DataType<DstPixel>::type);
                                         ~~~~~~~~~~~~~~~~~~~~~~~~^
/Users/kdh/ros-install-osx/lunar_desktop_full_ws/src/image_pipeline/image_proc/src/nodelets/crop_decimate.cpp:245:7: note: in instantiation of function template specialization
      'image_proc::debayer2x2toBGR<unsigned short>' requested here
      debayer2x2toBGR<uint16_t>(output.image, bgr, 0, 1, step, step + 1);
      ^
2 errors generated.
make[2]: *** [CMakeFiles/image_proc.dir/src/nodelets/crop_decimate.cpp.o] Error 1
make[1]: *** [CMakeFiles/image_proc.dir/all] Error 2
make: *** [all] Error 2
```

It is using `/usr/local/include/opencv`, so I just checked:
```
$ ls -l /usr/local/include/opencv
lrwxr-xr-x  1 kdh  admin  37 Oct 26 11:08 /usr/local/include/opencv -> ../Cellar/opencv/3.3.1/include/opencv
```

kdhansen, 2017-10-28:
Seeing the #4 issue, made me switch over to the CLT toolchain, I did a `catkin clean` and rebuild, but the issue is still the same.

rodrigoalmeida94, 2017-10-28:
Hello,

Here is the error message I get. The same as @kdhansen. 
```
[ 33%] Building CXX object CMakeFiles/image_proc.dir/src/nodelets/crop_decimate.cpp.o
/Library/Developer/CommandLineTools/usr/bin/c++  -DROSCONSOLE_BACKEND_LOG4CXX -DROS_PACKAGE_NAME=\"image_proc\" -Dimage_proc_EXPORTS -I/Users/rodrigoalmeida/ros-install-osx/lunar_desktop_full_ws/devel/.private/image_proc/include -I/Users/rodrigoalmeida/ros-install-osx/lunar_desktop_full_ws/src/image_pipeline/image_proc/include -I/opt/ros/lunar/include -I/opt/ros/lunar/share/xmlrpcpp/cmake/../../../include/xmlrpcpp -isystem /usr/local/include -isystem /usr/local/include/opencv -I/usr/local/Cellar/console_bridge/0.3.2/include  -O3 -DNDEBUG -fPIC   -o CMakeFiles/image_proc.dir/src/nodelets/crop_decimate.cpp.o -c /Users/rodrigoalmeida/ros-install-osx/lunar_desktop_full_ws/src/image_pipeline/image_proc/src/nodelets/crop_decimate.cpp
/Users/rodrigoalmeida/ros-install-osx/lunar_desktop_full_ws/src/image_pipeline/image_proc/src/nodelets/crop_decimate.cpp:121:66: error: no
      member named 'type' in 'cv::DataType<cv::Vec<unsigned char, 3> >'
  dst.create(src.rows / 2, src.cols / 2, cv::DataType<DstPixel>::type);
                                         ~~~~~~~~~~~~~~~~~~~~~~~~^
/Users/rodrigoalmeida/ros-install-osx/lunar_desktop_full_ws/src/image_pipeline/image_proc/src/nodelets/crop_decimate.cpp:237:7: note: in
      instantiation of function template specialization 'image_proc::debayer2x2toBGR<unsigned char>' requested here
      debayer2x2toBGR<uint8_t>(output.image, bgr, 0, 1, step, step + 1);
      ^
/Users/rodrigoalmeida/ros-install-osx/lunar_desktop_full_ws/src/image_pipeline/image_proc/src/nodelets/crop_decimate.cpp:121:66: error: no
      member named 'type' in 'cv::DataType<cv::Vec<unsigned short, 3> >'
  dst.create(src.rows / 2, src.cols / 2, cv::DataType<DstPixel>::type);
                                         ~~~~~~~~~~~~~~~~~~~~~~~~^
/Users/rodrigoalmeida/ros-install-osx/lunar_desktop_full_ws/src/image_pipeline/image_proc/src/nodelets/crop_decimate.cpp:245:7: note: in
      instantiation of function template specialization 'image_proc::debayer2x2toBGR<unsigned short>' requested here
      debayer2x2toBGR<uint16_t>(output.image, bgr, 0, 1, step, step + 1);
      ^
2 errors generated.
make[2]: *** [CMakeFiles/image_proc.dir/src/nodelets/crop_decimate.cpp.o] Error 1
make[1]: *** [CMakeFiles/image_proc.dir/all] Error 2
make: *** [all] Error 2
```

As you can see **-isystem /usr/local/include/opencv** so it should be fine.

mikepurvis, 2017-10-30:
I can confirm that this is a regression from OpenCV 3.3.0 -> 3.3.1. I upgraded on my system and the build of that package now fails in exactly the same way as reported by everyone else. It looks like this is a deprecated interface, based on this documentation page:

https://docs.opencv.org/3.3.1/d0/d3a/classcv_1_1DataType.html

They probably didn't mean to break this in a point release; it's possible it's as a result of https://github.com/opencv/opencv/pull/9457. Anyway, it doesn't seem like it's a terribly onerous fix if someone wants to make the required PR to upstream image_proc. Volunteers?

kdhansen, 2017-10-30:
I just submitted a RP https://github.com/ros-perception/image_pipeline/pull/302 fixing the image_proc and a related problem in stereo_image_proc

rodrigoalmeida94, 2017-10-30:
Since I had opencv 3.3.0_3 installed in my computer I used:

`brew switch opencv 3.3.0_3`

And then build again, and everything worked fine, but I guess this is just a workaround.

mikepurvis, 2017-11-01:
Thanks @kdhansen; I've submitted https://github.com/ros-perception/image_pipeline/pull/303 as well. We'll see what Vincent wants to do.

In the meantime, I've committed b6384c6929ae597db0337560b217a3398d99af36 which should allow Lunar to continue to build here.

closed 2017-11-01 by mikepurvis