mikepurvis/ros-install-osx #91: boost/tr1/unordered_set.hpp not found [Kinetic, ros_comm]

clydemcqueen, 2017-11-04:
I modified the script to install just ros_comm on Kinetic. It fails while building rospack.cpp:

Errors     << rospack:make /Users/clyde/ros-install-osx/kinetic_ros_comm_ws/logs/rospack/build.make.000.log                                                 
In file included from /Users/clyde/ros-install-osx/kinetic_ros_comm_ws/src/rospack/src/rospack.cpp:28:
/Users/clyde/ros-install-osx/kinetic_ros_comm_ws/src/rospack/include/rospack/rospack.h:108:10: fatal error: 'boost/tr1/unordered_set.hpp' file not found
#include <boost/tr1/unordered_set.hpp>
         ^

Brew installs boost 1.65.1, which includes unordered_set.hpp, but without the tr1 prefix:
/usr/local/Cellar/boost/1.65.1/include/boost/unordered_set.hpp

I'm on Sierra 10.12.3, if that matters.

clydemcqueen, 2017-11-04:
Looks like rospack has been fixed in Lunar. https://github.com/ros/rospack/commit/6fcee1b0d2b1c225c39a5820981166b99f0d8750

mikepurvis, 2017-11-14:
Closing since the issue only affects <Lunar.

closed 2017-11-14 by mikepurvis

DaHaiHuha, 2019-03-18:
i really wonder how would i came across the problem when i try to install kinetic in macos: mojava
`==> Processing catkin package: 'rospack'
==> Building with env: '/Users/apple/ros_catkin_ws/install_isolated/env.sh'
Makefile exists, skipping explicit cmake invocation...
==> make cmake_check_build_system in '/Users/apple/ros_catkin_ws/build_isolated/rospack'
==> make -j8 -l8 in '/Users/apple/ros_catkin_ws/build_isolated/rospack'
[ 33%] Building CXX object CMakeFiles/rospack.dir/src/rospack_backcompat.cpp.o
[ 33%] Building CXX object CMakeFiles/rospack.dir/src/rospack_cmdline.cpp.o
[ 33%] Building CXX object CMakeFiles/rospack.dir/src/rospack.cpp.o
[ 44%] Building CXX object CMakeFiles/rospack.dir/src/utils.cpp.o
In file included from In file included from /Users/apple/ros_catkin_ws/src/rospack/src/rospack.cpp:/Users/apple/ros_catkin_ws/src/rospack/src/rospack_cmdline.cpp28::
28:
/Users/apple/ros_catkin_ws/src/rospack/include/rospack/rospack.h:108/Users/apple/ros_catkin_ws/src/rospack/include/rospack/rospack.h::10108:: 10: fatal error: fatal error: 
      'boost/tr1/unordered_set.hpp' file not found

      'boost/tr1/unordered_set.hpp' file not found
#include <boost/tr1/unordered_set.hpp>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
#include <boost/tr1/unordered_set.hpp>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from /Users/apple/ros_catkin_ws/src/rospack/src/rospack_backcompat.cpp:29:
/Users/apple/ros_catkin_ws/src/rospack/include/rospack/rospack.h:108:10: fatal error: 
      'boost/tr1/unordered_set.hpp' file not found
#include <boost/tr1/unordered_set.hpp>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/rospack.dir/src/rospack_backcompat.cpp.o] Error 1
make[2]: *** Waiting for unfinished jobs....
/Users/apple/ros_catkin_ws/src/rospack/src/utils.cpp:31:10: fatal error: 
      'boost/tr1/unordered_set.hpp' file not found
#include <boost/tr1/unordered_set.hpp>
         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~
1 error generated.
make[2]: *** [CMakeFiles/rospack.dir/src/utils.cpp.o] Error 1
1 error generated.
make[2]: *** [CMakeFiles/rospack.dir/src/rospack.cpp.o] Error 1
1 error generated.
make[2]: *** [CMakeFiles/rospack.dir/src/rospack_cmdline.cpp.o] Error 1
make[1]: *** [CMakeFiles/rospack.dir/all] Error 2
make: *** [all] Error 2
<== Failed to process package 'rospack': 
  Command '['/Users/apple/ros_catkin_ws/install_isolated/env.sh', 'make', '-j8', '-l8']' returned non-zero exit status 2

Reproduce this error by running:
==> cd /Users/apple/ros_catkin_ws/build_isolated/rospack && /Users/apple/ros_catkin_ws/install_isolated/env.sh make -j8 -l8

Command failed, exiting.`
i would really appreciate if anyone could help