mikepurvis/ros-install-osx #93: Build error on gazebo_ros

PymZoR, 2017-11-10:
Fresh install on macOS Sierra 10.12.6
After fixing the linking error of OpenCV (see https://github.com/mikepurvis/ros-install-osx/issues/92), I got the following issue.

libprotoc: 3.4.0
ignition-transport: stable 1.4.0 (from brew)

gazebo_ros fails to build with the following error messages :

> In file included from lunar_desktop_full_ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_api_plugin.cpp:25:
> In file included from lunar_desktop_full_ws/src/gazebo_ros_pkgs/gazebo_ros/include/gazebo_ros/gazebo_ros_api_plugin.h:34:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/physics.hh:2:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/Actor.hh:24:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/Model.hh:30:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/Entity.hh:25:
> In file included from /usr/local/Cellar/ignition-transport3/3.0.1_7/include/ignition/transport3/ignition/transport/Node.hh:32:
> In file included from /usr/local/include/ignition/msgs0/ignition/msgs.hh:3:
> In file included from /usr/local/include/ignition/msgs0/ignition/msgs/MessageTypes.hh:2:
> /usr/local/include/ignition/msgs0/ignition/msgs/altimeter.pb.h:17:2: error: This file was generated by an older version of protoc which is
> #error This file was generated by an older version of protoc which is
>  ^
> /usr/local/include/ignition/msgs0/ignition/msgs/altimeter.pb.h:18:2: error: incompatible with your Protocol Buffer headers.  Please
> #error incompatible with your Protocol Buffer headers.  Please
>  ^
> /usr/local/include/ignition/msgs0/ignition/msgs/altimeter.pb.h:19:2: error: regenerate this file with a newer version of protoc.
> #error regenerate this file with a newer version of protoc.
>  ^
> In file included from /Users/pymzor/Projects/buckless/client/ros-install-osx/lunar_desktop_full_ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_api_plugin.cpp:25:
> In file included from /Users/pymzor/Projects/buckless/client/ros-install-osx/lunar_desktop_full_ws/src/gazebo_ros_pkgs/gazebo_ros/include/gazebo_ros/gazebo_ros_api_plugin.h:34:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/physics.hh:2:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/Actor.hh:24:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/Model.hh:30:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/Entity.hh:25:
> In file included from /usr/local/Cellar/ignition-transport3/3.0.1_7/include/ignition/transport3/ignition/transport/Node.hh:32:
> In file included from /usr/local/include/ignition/msgs0/ignition/msgs.hh:3:
> In file included from /usr/local/include/ignition/msgs0/ignition/msgs/MessageTypes.hh:2:
> In file included from /usr/local/include/ignition/msgs0/ignition/msgs/altimeter.pb.h:32:
> /usr/local/include/ignition/msgs0/ignition/msgs/time.pb.h:17:2: error: This file was generated by an older version of protoc which is
> #error This file was generated by an older version of protoc which is
>  ^
> /usr/local/include/ignition/msgs0/ignition/msgs/time.pb.h:18:2: error: incompatible with your Protocol Buffer headers.  Please
> #error incompatible with your Protocol Buffer headers.  Please
>  ^
> /usr/local/include/ignition/msgs0/ignition/msgs/time.pb.h:19:2: error: regenerate this file with a newer version of protoc.
> #error regenerate this file with a newer version of protoc.
>  ^
> In file included from /Users/pymzor/Projects/buckless/client/ros-install-osx/lunar_desktop_full_ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_api_plugin.cpp:25:
> In file included from /Users/pymzor/Projects/buckless/client/ros-install-osx/lunar_desktop_full_ws/src/gazebo_ros_pkgs/gazebo_ros/include/gazebo_ros/gazebo_ros_api_plugin.h:34:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/physics.hh:2:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/Actor.hh:24:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/Model.hh:30:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/Entity.hh:25:
> In file included from /usr/local/Cellar/ignition-transport3/3.0.1_7/include/ignition/transport3/ignition/transport/Node.hh:32:
> In file included from /usr/local/include/ignition/msgs0/ignition/msgs.hh:3:
> In file included from /usr/local/include/ignition/msgs0/ignition/msgs/MessageTypes.hh:2:
> In file included from /usr/local/include/ignition/msgs0/ignition/msgs/altimeter.pb.h:33:
> /usr/local/include/ignition/msgs0/ignition/msgs/header.pb.h:17:2: error: This file was generated by an older version of protoc which is
> #error This file was generated by an older version of protoc which is
>  ^
> /usr/local/include/ignition/msgs0/ignition/msgs/header.pb.h:18:2: error: incompatible with your Protocol Buffer headers.  Please
> #error incompatible with your Protocol Buffer headers.  Please
>  ^
> /usr/local/include/ignition/msgs0/ignition/msgs/header.pb.h:19:2: error: regenerate this file with a newer version of protoc.
> #error regenerate this file with a newer version of protoc.
>  ^
> In file included from /Users/pymzor/Projects/buckless/client/ros-install-osx/lunar_desktop_full_ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_api_plugin.cpp:25:
> In file included from /Users/pymzor/Projects/buckless/client/ros-install-osx/lunar_desktop_full_ws/src/gazebo_ros_pkgs/gazebo_ros/include/gazebo_ros/gazebo_ros_api_plugin.h:34:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/physics.hh:2:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/Actor.hh:24:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/Model.hh:30:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/Entity.hh:25:
[... 45 lines cut ...]
> #error incompatible with your Protocol Buffer headers.  Please
>  ^
> /usr/local/include/ignition/msgs0/ignition/msgs/pose.pb.h:19:2: error: regenerate this file with a newer version of protoc.
> #error regenerate this file with a newer version of protoc.
>  ^
> In file included from /Users/pymzor/Projects/buckless/client/ros-install-osx/lunar_desktop_full_ws/src/gazebo_ros_pkgs/gazebo_ros/src/gazebo_ros_api_plugin.cpp:25:
> In file included from /Users/pymzor/Projects/buckless/client/ros-install-osx/lunar_desktop_full_ws/src/gazebo_ros_pkgs/gazebo_ros/include/gazebo_ros/gazebo_ros_api_plugin.h:34:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/physics.hh:2:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/Actor.hh:24:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/Model.hh:30:
> In file included from /usr/local/Cellar/gazebo8/8.1.1_5/include/gazebo-8/gazebo/physics/Entity.hh:25:
> In file included from /usr/local/Cellar/ignition-transport3/3.0.1_7/include/ignition/transport3/ignition/transport/Node.hh:32:
> In file included from /usr/local/include/ignition/msgs0/ignition/msgs.hh:3:
> In file included from /usr/local/include/ignition/msgs0/ignition/msgs/MessageTypes.hh:3:
> In file included from /usr/local/include/ignition/msgs0/ignition/msgs/any.pb.h:35:
> In file included from /usr/local/include/ignition/msgs0/ignition/msgs/pose.pb.h:32:
> /usr/local/include/ignition/msgs0/ignition/msgs/vector3d.pb.h:17:2: error: This file was generated by an older version of protoc which is
> #error This file was generated by an older version of protoc which is
> 

Digging into some of these files, I found that the protobuf version requirement (which I assume is 3.3.0) was different from my system protobuf (3.4.0)

`#if GOOGLE_PROTOBUF_VERSION < 300300`

I haven't been able to go any further than that.

mikepurvis, 2017-11-10:
Looks like the issue is a bunch of stuff compiled into Gazebo, so the gazebo bottle probably needs a revision bump and rebuild: https://github.com/osrf/homebrew-simulation/blob/master/gazebo8.rb#L7

FYI @scpeters

mikepurvis, 2017-11-10:
@PymZoR In the short term, you can likely get around this by building Gazebo8 from source:

```
brew reinstall --build-from-source gazebo8
```

PymZoR, 2017-11-10:
@mikepurvis got the same error

> $ brew reinstall --build-from-source gazebo8
> ==> Reinstalling osrf/simulation/gazebo8
> ==> Downloading http://gazebosim.org/distributions/gazebo/releases/gazebo-8.1.1.tar.bz2
> Already downloaded: /Users/pymzor/Library/Caches/Homebrew/gazebo8-8.1.1.tar.bz2
> ==> Downloading https://gist.githubusercontent.com/scpeters/9199370/raw/afe595587e38737c537124a3652db99de026c272/brew_python_fix.patch
> Already downloaded: /Users/pymzor/Library/Caches/Homebrew/gazebo8--patch-c4774f64c490fa03236564312bd24a8630963762e25d98d072e747f0412df18e.patch
> ==> Patching
> ==> Applying brew_python_fix.patch
> patching file deps/opende/CMakeLists.txt
> ==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDEBUG -DCMAKE_INSTALL_PREFIX=/usr/local/Cellar/gazebo8/8.1.1_5 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_
> ==> make install
> Last 15 lines from /Users/pymzor/Library/Logs/Homebrew/gazebo8/02.make:
> In file included from /tmp/gazebo8-20171110-9130-1dnutp2/gazebo-8.1.1/gazebo/msgs/msgs.cc:26:
> In file included from /tmp/gazebo8-20171110-9130-1dnutp2/gazebo-8.1.1/gazebo/msgs/msgs.hh:33:
> /usr/local/include/ignition/msgs0/ignition/msgs/material.pb.h:17:2: error: This file was generated by an older version of protoc which is
> #error This file was generated by an older version of protoc which is
>  ^
> /usr/local/include/ignition/msgs0/ignition/msgs/material.pb.h:18:2: error: incompatible with your Protocol Buffer headers.  Please
> #error incompatible with your Protocol Buffer headers.  Please
>  ^
> /usr/local/include/ignition/msgs0/ignition/msgs/material.pb.h:19:2: error: regenerate this file with a newer version of protoc.
> #error regenerate this file with a newer version of protoc.
>  ^

mikepurvis, 2017-11-10:
Oh boo, it looks like it's a header from ignition, so all of its pieces would need to be built from source as well:

```
brew reinstall --build-from-source ignition-math3 ignition-msgs0 ignition-transport3 gazebo8
```

PymZoR, 2017-11-10:
After a long and tedious build, some Qt-related errors :(

> [ 68%] Building CXX object gazebo/physics/CMakeFiles/gazebo_physics.dir/Population.cc.o
> cd /tmp/gazebo8-20171110-21204-1ctx49r/gazebo-8.1.1/build/gazebo/physics && /usr/local/Homebrew/Library/Homebrew/shims/super/clang++  -DBUILDING_DLL -DBUILDING_DLL_GZ_PHYSICS -DHAVE_OPENGL -DLIBBULLET_VERSION=2.87 -DLIBBULLET_VERSION_GT_282 -DMAC_OS_X_VERSION=1080 -Dgazebo_physics_EXPORTS -I/tmp/gazebo8-20171110-21204-1ctx49r/gazebo-8.1.1/test/gtest/include -I/tmp/gazebo8-20171110-21204-1ctx49r/gazebo-8.1.1 -I/tmp/gazebo8-20171110-21204-1ctx49r/gazebo-8.1.1/build -isystem /tmp/gazebo8-20171110-21204-1ctx49r/gazebo-8.1.1/deps/opende/include -isystem /usr/local/include -isystem /usr/local/Cellar/tinyxml2/5.0.1/include -isystem /usr/local/include/ignition/msgs0 -isystem /usr/local/include/ignition/math3 -I/usr/local/Cellar/ignition-transport3/3.0.1_7/include/ignition/transport3 -I/usr/local/Cellar/zeromq/4.2.2/include -I/usr/local/Cellar/ossp-uuid/1.6.2_2/include/ossp -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_4/include/OGRE/RTShaderSystem -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_4/include -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_4/include/OGRE -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_4/include/OGRE/Terrain -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_4/include/OGRE/Paging -isystem /usr/local/Cellar/ogre1.9/1.9-20160714-108ab0bcc69603dba32c0ffd4bbbc39051f421c9_4/include/OGRE/Overlay -isystem /usr/local/Cellar/sdformat5/5.2.0_2/include/sdformat-5.2 -isystem /usr/local/opt/tinyxml/include -I/usr/local/Cellar/bullet/2.87/include/bullet -I/usr/local/Cellar/bullet/2.87/include -isystem /usr/local/Cellar/simbody/3.5.4_2/include/simbody  -std=c++11 -stdlib=libc++ -stdlib=libc++  -Wall -Wextra -Wno-long-long -Wno-unused-value -Wfloat-equal -Wshadow -Wswitch-default -Wmissing-include-dirs -pedantic -std=c++11   -O3 -DNDEBUG -mssse3 -msse3 -msse -msse2  -fPIC   -fPIC -I/usr/local/Cellar/bullet/2.87/include/bullet -I/usr/local/Cellar/bullet/2.87/include -I"/usr/local/Cellar/simbody/3.5.4_2/include/simbody" -std=c++11 -o CMakeFiles/gazebo_physics.dir/Population.cc.o -c /tmp/gazebo8-20171110-21204-1ctx49r/gazebo-8.1.1/gazebo/physics/Population.cc
> In file included from /tmp/gazebo8-20171110-21204-1ctx49r/gazebo-8.1.1/gazebo/gui/Actions.cc:18:
> In file included from /tmp/gazebo8-20171110-21204-1ctx49r/gazebo-8.1.1/gazebo/gui/Actions.hh:22:
> In file included from /tmp/gazebo8-20171110-21204-1ctx49r/gazebo-8.1.1/gazebo/gui/qt.h:36:
> In file included from /usr/local/opt/qt/include/QtGui/QWindow:1:
> In file included from /usr/local/opt/qt/lib/QtGui.framework/Headers/qwindow.h:51:
> In file included from /usr/local/opt/qt/include/QtGui/qsurface.h:45:
> /usr/local/opt/qt/include/QtGui/qsurfaceformat.h:62:5: error: unknown type name 'Q_ENUM'
>     Q_ENUM(FormatOption)
>     ^
> /usr/local/opt/qt/include/QtGui/qsurfaceformat.h:62:25: error: expected ';' at end of declaration list
>     Q_ENUM(FormatOption)
>                         ^
> /usr/local/opt/qt/include/QtGui/qsurfaceformat.h:71:5: error: unknown type name 'Q_ENUM'
>     Q_ENUM(SwapBehavior)
>     ^
> /usr/local/opt/qt/include/QtGui/qsurfaceformat.h:71:25: error: expected ';' at end of declaration list
>     Q_ENUM(SwapBehavior)
>                         ^
> /usr/local/opt/qt/include/QtGui/qsurfaceformat.h:79:5: error: unknown type name 'Q_ENUM'
>     Q_ENUM(RenderableType)
>     ^
> /usr/local/opt/qt/include/QtGui/qsurfaceformat.h:79:27: error: expected ';' at end of declaration list
>     Q_ENUM(RenderableType)
>                           ^
> /usr/local/opt/qt/include/QtGui/qsurfaceformat.h:86:5: error: unknown type name 'Q_ENUM'
>     Q_ENUM(OpenGLContextProfile)
>     ^
> /usr/local/opt/qt/include/QtGui/qsurfaceformat.h:86:33: error: expected ';' at end of declaration list
>     Q_ENUM(OpenGLContextProfile)
>                                 ^
> /usr/local/opt/qt/include/QtGui/qsurfaceformat.h:89:33: error: unknown type name 'FormatOptions'; did you mean 'FormatOption'?
>     /*implicit*/ QSurfaceFormat(FormatOptions options);
>                                 ^
> /usr/local/opt/qt/include/QtGui/qsurfaceformat.h:56:10: note: 'FormatOption' declared here
>     enum FormatOption {
>          ^
> /usr/local/opt/qt/include/QtGui/qsurfaceformat.h:117:21: error: unknown type name 'OpenGLContextProfile'
>     void setProfile(OpenGLContextProfile profile);
>                     ^
> /usr/local/opt/qt/include/QtGui/qsurfaceformat.h:118:5: error: unknown type name 'OpenGLContextProfile'
>     OpenGLContextProfile profile() const;
>     ^
> /usr/local/opt/qt/include/QtGui/qsurfaceformat.h:120:28: error: unknown type name 'RenderableType'
>     void setRenderableType(RenderableType type);
>                            ^
> /usr/local/opt/qt/include/QtGui/qsurfaceformat.h:121:5: error: unknown type name 'RenderableType'
>     RenderableType renderableType() const;
>     ^
> /usr/local/opt/qt/include/QtGui/qsurfaceformat.h:135:5: error: function-like macro 'QT_DEPRECATED_SINCE' is not defined
> #if QT_DEPRECATED_SINCE(5, 2)
>     ^
> /usr/local/opt/qt/include/QtGui/qsurfaceformat.h:140:37: error: no type named 'FormatOptions' in 'QSurfaceFormat'; did you mean 'FormatOption'?
[... cut ...]
>                               ~~~~~~~~~~~~~~~~^
> /usr/local/lib/QtCore.framework/Headers/qglobal.h:2471:60: note: expanded from macro 'Q_DECLARE_OPERATORS_FOR_FLAGS'
> Q_DECL_CONSTEXPR inline QFlags<Flags::enum_type> operator|(Flags::enum_type f1, Flags::enum_type f2) \
>                                                            ^~~~~
> /usr/local/opt/qt/include/QtGui/qsurfaceformat.h:56:10: note: 'FormatOption' declared here
>     enum FormatOption {
>          ^
> fatal error: too many errors emitted, stopping now [-ferror-limit=]

scpeters, 2017-11-10:
I don't understand how you have ignition-transport version 1.4.0 when gazebo8 depends on ignition-transport3. Also, we have a nightly job that tests the gazebo8 bottle and it [passed recently](https://build.osrfoundation.org/view/os_homebrew/job/gazebo-install-one_liner-homebrew-amd64/710/).

Maybe the issue is that the official homebrew version of protoc is still 1.3.0:

* https://github.com/Homebrew/homebrew-core/blob/master/Formula/protobuf-c.rb#L4

I think the bottles will only work if you are using the same version of dependencies.

scpeters, 2017-11-10:
Regarding the qt compilation error, I'm not sure exactly what caused it, but there's here's two issues I've had:

* the official `qt` (qt5) formula is keg-only, which means it doesn't sym-link to `/usr/local`, so I have to add `/usr/local/opt/qt` to my `CMAKE_PREFIX_PATH` if I'm building in a catkin workspace. Building with `brew` seems to be able to find it without extra work.
* If you have gazebo7 on this system, then it will install `qt@4`, and I've had problems trying to compile against qt5 when qt4 is also installed.  Uninstalling qt4 fixes it.

PymZoR, 2017-11-10:
@scpeters : Don't know for ignition-transport, it just appears that ignition-transport default brew version is 1.40, and somehow the build process here had me install it. Removing it does not cause any dependencies problems. I still have the ignition-transport3 bottle installed though, and i'm re-trying to build gazebo with `qt@4` removed.

scpeters, 2017-11-10:
Well `ignition-transport` is listed as a dependency of gazebo7, so if you had that installed before, that's probably where it came from.

PymZoR, 2017-11-11:
Removing `qt@4`and rebuilding from source with ignition-transport3 solved the issue.
I'm running now with another issue with `pcl_ros`, i'll open a new ticker for it. 
Thanks for the help

closed 2017-11-11 by PymZoR

ethanjyx, 2019-03-25:
@mikepurvis I tried to run `brew reinstall --build-from-source gazebo8` but still getting the following error

```
==> cmake .. -DCMAKE_C_FLAGS_RELEASE=-DNDEBUG -DCMAKE_CXX_FLAGS_RELEASE=-DNDE
Last 15 lines from /Users/yixingjiang/Library/Logs/Homebrew/gazebo8/01.cmake:
-- 	ronn not found, manpages won't be generated
-- 	OpenSource Virtual Reality (OSVR) support will be disabled.
-- END BUILD WARNINGS

-- BUILD ERRORS: These must be resolved before compiling.
-- 	Boost not found. Please install thread signals system filesystem program_options regex date_time boost version 1.40.0 or higher.
-- END BUILD ERRORS

CMake Error at CMakeLists.txt:303 (message):
  Errors encountered in build.  Please see the BUILD ERRORS above.

-- Configuring incomplete, errors occurred!
See also "/tmp/gazebo8-20190324-2823-nzdfqw/gazebo-8.6.0/build/CMakeFiles/CMakeOutput.log".
See also "/tmp/gazebo8-20190324-2823-nzdfqw/gazebo-8.6.0/build/CMakeFiles/CMakeError.log".

If reporting this issue please do so at (not Homebrew/brew or Homebrew/core):
https://github.com/osrf/homebrew-simulation/issues
```

scpeters, 2019-03-25:
The recent release of boost 1.69 required a patch for the gazebo7 formula in https://github.com/osrf/homebrew-simulation/pull/649 and I suspect gazebo8 would need a similar patch.

gazebo9 and gazebo10 have already been patched, but gazebo8 is end-of-life, so I haven't been prioritizing patches for it. I would merge patches for gazebo8 submitted via pull request if they are reasonable.

scpeters, 2019-03-26:
I went ahead and patched gazebo8 in https://github.com/osrf/homebrew-simulation/commit/4b2696b8e04aabedab734117d8e0f90d6454f9ac, but since it is end-of-life we won't be proactively solving build failures. I will review pull requests, but I encourage users to upgrade to gazebo9 or gazebo10.