mikepurvis/ros-install-osx #99: Possible Gazebo version problem?

karanchawla, 2017-12-04:
http://gazebosim.org/tutorials?tut=ros_wrapper_versions&cat=connect_ros

According to the link above ROS Kinetic/Lunar works with Gazebo version 7, but the script installs gazebo version 8. I tried running `rosrun gazebo_ros gazebo` and it looks like ros isn't communicating with gazebo. 

I only get the default rostopics instead of the list given above. I also tried a simulation which works perfectly on my Ubuntu setup but it fails to communicate with ros in OSX. 

To solve this, I purged my Gazebo8 installation and installed Gazebo7 but it still doesn't seem to be working. 

Have you faced this issue with your installation?

fmina, 2018-01-28:
I do have the same problem. ROS is not communicating with Gazebo.

karanchawla, 2018-01-28:
Since I didn't find a solution to this, I just got Parallels and am using ROS with Ubuntu. It's just easier.