mikepurvis/ros-install-osx #100: Unable to install on High Sierra

mdpradeep, 2017-12-05:
I am trying to install on High Sierra and am getting the following error:

```
Finished  <<< roscpp                                          [ 1 minute and 53.9 seconds ]
Starting  >>> rqt_gui
Finished  <<< rqt_gui                                         [ 9.7 seconds ]
Starting  >>> qt_gui_cpp
_______________________________________________________________________________________________________________________________________________________________________________
Warnings   << qt_gui_cpp:cmake /Users/pradeep/work/ros/ros-install-osx/lunar_desktop_full_ws/logs/qt_gui_cpp/build.cmake.000.log
CMake Warning at /opt/ros/lunar/share/python_qt_binding/cmake/shiboken_helper.cmake:18 (find_package):
  By not providing "FindShiboken2.cmake" in CMAKE_MODULE_PATH this project
  has asked CMake to find a package configuration file provided by
  "Shiboken2", but CMake did not find one.

  Could not find a package configuration file provided by "Shiboken2" with
  any of the following names:

    Shiboken2Config.cmake
    shiboken2-config.cmake

  Add the installation prefix of "Shiboken2" to CMAKE_PREFIX_PATH or set
  "Shiboken2_DIR" to a directory containing one of the above files.  If
  "Shiboken2" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  src/qt_gui_cpp_shiboken/CMakeLists.txt:41 (include)

CMake Warning at /opt/ros/lunar/share/python_qt_binding/cmake/shiboken_helper.cmake:23 (find_package):
  By not providing "FindPySide2.cmake" in CMAKE_MODULE_PATH this project has
  asked CMake to find a package configuration file provided by "PySide2", but
  CMake did not find one.

  Could not find a package configuration file provided by "PySide2" with any
  of the following names:

    PySide2Config.cmake
    pyside2-config.cmake

  Add the installation prefix of "PySide2" to CMAKE_PREFIX_PATH or set
  "PySide2_DIR" to a directory containing one of the above files.  If
  "PySide2" provides a separate development package or SDK, be sure it has
  been installed.
Call Stack (most recent call first):
  src/qt_gui_cpp_shiboken/CMakeLists.txt:41 (include)

CMake Warning at /opt/ros/lunar/share/python_qt_binding/cmake/shiboken_helper.cmake:34 (message):
  Shiboken binding generator NOT available.
Call Stack (most recent call first):
  src/qt_gui_cpp_shiboken/CMakeLists.txt:41 (include)

cd /Users/pradeep/work/ros/ros-install-osx/lunar_desktop_full_ws/build/qt_gui_cpp; catkin build --get-env qt_gui_cpp | catkin env -si  /usr/local/bin/cmake /Users/pradeep/work/ros/ros-install-osx/lunar_desktop_full_ws/src/qt_gui_core/qt_gui_cpp --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/Users/pradeep/work/ros/ros-install-osx/lunar_desktop_full_ws/devel/.private/qt_gui_cpp -DCMAKE_INSTALL_PREFIX=/opt/ros/lunar -DCATKIN_ENABLE_TESTING=1 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DPYTHON_EXECUTABLE=/usr/local/bin/python2 -DPYTHON_LIBRARY=/usr/local/Cellar/python/2.7.14/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib -DPYTHON_INCLUDE_DIR=/usr/local/Cellar/python/2.7.14/Frameworks/Python.framework/Versions/2.7/include/python2.7; cd -
...............................................................................................................................................................................
_______________________________________________________________________________________________________________________________________________________________________________
Errors     << qt_gui_cpp:make /Users/pradeep/work/ros/ros-install-osx/lunar_desktop_full_ws/logs/qt_gui_cpp/build.make.000.log
/bin/sh: @/Users/pradeep/work/ros/ros-install-osx/lunar_desktop_full_ws/build/qt_gui_cpp/src/qt_gui_cpp/__/__/include/qt_gui_cpp/moc_plugin_bridge.cpp_parameters: No such file or directory
/bin/sh: @/Users/pradeep/work/ros/ros-install-osx/lunar_desktop_full_ws/build/qt_gui_cpp/src/qt_gui_cpp/__/__/include/qt_gui_cpp/moc_plugin_context.cpp_parameters: No such file or directory
make[2]: *** [src/qt_gui_cpp/__/__/include/qt_gui_cpp/moc_plugin_bridge.cpp] Error 127
make[2]: *** Waiting for unfinished jobs....
make[2]: *** [src/qt_gui_cpp/__/__/include/qt_gui_cpp/moc_plugin_context.cpp] Error 127
/bin/sh: @/Users/pradeep/work/ros/ros-install-osx/lunar_desktop_full_ws/build/qt_gui_cpp/src/qt_gui_cpp/__/__/include/qt_gui_cpp/moc_plugin.cpp_parameters: No such file or directory
make[2]: *** [src/qt_gui_cpp/__/__/include/qt_gui_cpp/moc_plugin.cpp] Error 127
make[1]: *** [src/qt_gui_cpp/CMakeFiles/qt_gui_cpp.dir/all] Error 2
make: *** [all] Error 2
cd /Users/pradeep/work/ros/ros-install-osx/lunar_desktop_full_ws/build/qt_gui_cpp; catkin build --get-env qt_gui_cpp | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -
...............................................................................................................................................................................
Failed     << qt_gui_cpp:make                                 [ Exited with code 2 ]
Failed    <<< qt_gui_cpp                                      [ 7.9 seconds ]
Abandoned <<< image_geometry                                  [ Unrelated job failed ]
```

Any ideas how I can get past this ?

sjulier, 2017-12-21:
I saw this problem when I was trying to build ROS, but I didn't look at it in detail because I had lots of other issues I was trying to fix. While trying to fix those, the problem seemed to go away. However, I can't say which - if any - of the things I did actually fixed the problem.

Fro what I recall, these were the steps I took after which the problem disappeared:

1. I deleted the build and develop directories. 
2. Check the python executable. Homebrew named the executables python2 and python3. I had to set up a symbolic link from /user/local/bin/python to /user/local/bin/python2. This fixed a problem which claimed that numpy was not installed with cv_bridge.

I began my ROS building efforts with a fresh install of homebrew and clearing the contents of /Library/python2.7/site-packages, so I don't think there were any weird background contextual issues.

mdpradeep, 2017-12-21:
Thanks so much @sjulier. I am able to get past that stage now. But I have a new problem.

```
Errors     << geometric_shapes:cmake /Users/pradeep/work/ros/ros-install-osx/lunar_desktop_full_ws/logs/geometric_shapes/build.cmake.000.log
Traceback (most recent call last):
  File "/usr/local/lib/python2.7/site-packages/catkin_tools/execution/executor.py", line 109, in async_job
    **stage.async_execute_process_kwargs))
  File "/usr/local/lib/python2.7/site-packages/trollius/tasks.py", line 259, in _step
    result = coro.send(value)
  File "/usr/local/lib/python2.7/site-packages/osrf_pycommon/process_utils/async_execute_process_trollius.py", line 69, in _async_execute_process_pty
    stderr_master, stderr_slave = pty.openpty()
  File "/usr/local/Cellar/python/2.7.14/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pty.py", line 29, in openpty
    master_fd, slave_name = _open_terminal()
  File "/usr/local/Cellar/python/2.7.14/Frameworks/Python.framework/Versions/2.7/lib/python2.7/pty.py", line 71, in _open_terminal
    raise os.error, 'out of pty devices'
OSError: out of pty devices
```
I did some search on this but could not figure out how to fix this issue. 

Thanks again for your help.

sjulier, 2017-12-21:
I have the same problem. I tried increasing the number of ptys available via:

sudo sysctl -w kern.tty.ptmx_max=999

but that didn't make much difference.

Assuming the problem is that too many pseudo-terminals are left dangling because the resources aren't being cleaned up properly, I installed findutils and am currently building all of the packages one-by-one using this slightly cheesy code:

for i in `gfind src -maxdepth 1 -type d -not -name '.*' -printf '%f '`;do catkin build -p1 -j1 $i;done

It takes forever to run, but I'm trying to minimise the number of sessions that get started. Still hasn't finished yet, but it hasn't crashed out either.

mdpradeep, 2017-12-22:
Thanks so much @sjulier. I was able to build and also start core. Closing this issue. Thanks again.

berkaydeniz, 2017-12-25:
while it is a great workaround, this is not a real solution. I would recommend reopening this issue.

Caramelle, 2018-01-03:
@mdpradeep Could you please share how did you get past your first issue?

lloydc99, 2018-01-07:
I had the same problem and I found a zombie cmake process stuck in the background.

lsof | grep ^cmake
-> bunch of open files

ps aux | fgrep cmake
->zombie cmake build

I killed the cmake process and then everything worked.

mdpradeep, 2018-01-08:
Reopening this issue as per request above.

mdpradeep, 2018-01-08:
@Caramelle, I was able to get past the first issue based on @sjulier suggestion - 

```
Check the python executable. Homebrew named the executables python2 and python3. I had to set up a symbolic link from /user/local/bin/python to /user/local/bin/python2. This fixed a problem which claimed that numpy was not installed with cv_bridge.
```

lloydc99, 2018-01-08:
I had the same problem. I added a python script to the shim dir to do the same thing.

sjulier, 2018-02-01:
This explains homebrew's rationale for having python2 and python3 executables:

https://docs.brew.sh/Homebrew-and-Python.html

I think it's to have a clear separation between system installed and homebrew installed versions of Python (which has been a nightmare in the past). Have to confess I don't have time to see if there's an elegant way to specify the python executable name.

tomlankhorst, 2018-03-23:
Temporarily increase the number of available pseudo terminals in Mac OS:

`sudo sysctl -w kern.tty.ptmx_max=999`