mikepurvis/ros-install-osx #103: Error: No available formula with the name "homebrew/science/matplotlib"

vivekchand, 2018-01-03:
```
Error: No available formula with the name "homebrew/science/matplotlib" 
==> Searching for a previously deleted formula (in the last month)...
Warning: homebrew/science is shallow clone. To get complete history run:
  git -C "$(brew --repo homebrew/science)" fetch --unshallow

homebrew/science/matplotlib was deleted from homebrew/science in commit 224eed91:
  Deprecate Homebrew/homebrew-science (https://github.com/Homebrew/homebrew-science/issues/6616)

To show the formula before removal run:
  git -C "$(brew --repo homebrew/science)" show 224eed91^:matplotlib.rb

If you still use this formula consider creating your own tap:
  https://docs.brew.sh/How-to-Create-and-Maintain-a-Tap.html
ERROR: the following rosdeps failed to install
  homebrew: command [brew install homebrew/science/matplotlib] failed

```

Caramelle, 2018-01-03:
I got around this by installing matplotlib with pip. 

In rosdeps.yaml replace
```
python-matplotlib:
  osx:
    homebrew:
      packages: [homebrew/science/matplotlib]
```

with

```
python-matplotlib:
  osx:
    pip:
      packages: [matplotlib]
```

fmina, 2018-01-24:
`matplotlib` moved to `brewsci/science`
maybe you can `brew tap brewsci/science` instead of `homebrew/science`.

NikolausDemmel, 2018-01-28:
> maybe you can brew tap brewsci/science instead of homebrew/science.

Doesn't work for me, see https://github.com/mikepurvis/ros-install-osx/pull/104#issuecomment-361026480

PR #104 (2018-01-29): Install matplotlib from pip
Fixes #103
commit 3cb191f9 (2018-01-29): Install matplotlib from pip instead of homebrew (#104)

Fixes #103
closed 2018-01-29 by mikepurvis