mikepurvis/ros-install-osx #107: install Failing at rospack

sushil-sundaram, 2018-01-22:
**System Details
macOS High Sierra
version:10.13.2**

This is what appears when the build fails
`Errors     << rospack:make /Users/sushilsundaram/Documents/Docker/ROSDocker/ros-install-osx/lunar_desktop_full_ws/logs/rospack/build.make.000.log                                               
make[2]: *** No rule to make target `/Users/sushilsundaram/ROS/lib/libpython2.7.dylib', needed by `/Users/sushilsundaram/Documents/Docker/ROSDocker/ros-install-osx/lunar_desktop_full_ws/devel/.private/rospack/lib/librospack.dylib'.  Stop.
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [CMakeFiles/rospack.dir/all] Error 2
make: *** [all] Error 2
cd /Users/sushilsundaram/Documents/Docker/ROSDocker/ros-install-osx/lunar_desktop_full_ws/build/rospack; catkin build --get-env rospack | catkin env -si  /usr/bin/make --jobserver-fds=6,7 -j; cd -`

This appears at the end
`[build] Summary: 26 of 196 packages succeeded.                                                                                                                                                  
[build]   Ignored:   31 packages were skipped or are blacklisted.                                                                                                                               
[build]   Warnings:  6 packages succeeded with warnings.                                                                                                                                        
[build]   Abandoned: 169 packages were abandoned.                                                                                                                                               
[build]   Failed:    1 packages failed.                                                                                                                                                         
[build] Runtime: 3 minutes and 29.4 seconds total.                                                                                                                                              
[build] Note: Workspace packages have changed, please re-source setup files to use them.`

famishedrover, 2018-01-26:
Similar issue , my build stops at , roscpp_traits with similar error.

famishedrover, 2018-01-26:
`Errors     << roscpp_traits:cmake /Users/muditverma/ros-install-osx/lunar_desktop_full_ws/logs/roscpp_traits/build.cmake.000.log                   
CMake Error: Generator: execution of make failed. Make command was: "/usr/bin/make" "cmTC_8a057/fast"
CMake Error at /usr/local/Cellar/cmake/3.10.2/share/cmake/Modules/CMakeTestCCompiler.cmake:52 (message):
  The C compiler

    "/usr/bin/cc"

  is not able to compile a simple test program.

  It fails with the following output:

    Change Dir: /Users/muditverma/ros-install-osx/lunar_desktop_full_ws/build/roscpp_traits/CMakeFiles/CMakeTmp

    Run Build Command:"/usr/bin/make" "cmTC_8a057/fast"
    Too many open files
    Generator: execution of make failed. Make command was: "/usr/bin/make" "cmTC_8a057/fast"

  CMake will not be able to correctly generate this project.
Call Stack (most recent call first):
  CMakeLists.txt:2 (project)

cd /Users/muditverma/ros-install-osx/lunar_desktop_full_ws/build/roscpp_traits; catkin build --get-env roscpp_traits | catkin env -si  /usr/local/bin/cmake /Users/muditverma/ros-install-osx/lunar_desktop_full_ws/src/roscpp_core/roscpp_traits --no-warn-unused-cli -DCATKIN_DEVEL_PREFIX=/Users/muditverma/ros-install-osx/lunar_desktop_full_ws/devel/.private/roscpp_traits -DCMAKE_INSTALL_PREFIX=/opt/ros/lunar -DCATKIN_ENABLE_TESTING=1 -DCMAKE_BUILD_TYPE=Release -DCMAKE_FIND_FRAMEWORK=LAST -DPYTHON_EXECUTABLE=/usr/local/bin/python2 -DPYTHON_LIBRARY=/usr/local/Cellar/python/2.7.14_2/Frameworks/Python.framework/Versions/2.7/lib/libpython2.7.dylib -DPYTHON_INCLUDE_DIR=/usr/local/Cellar/python/2.7.14_2/Frameworks/Python.framework/Versions/2.7/include/python2.7; cd -
...................................................................................................................................................
Failed     << roscpp_traits:cmake                             [ Exited with code 1 ]                                                               
Failed    <<< roscpp_traits                                   [ 1.5 seconds ]                                                                      
Abandoned <<< rosunit                                         [ Unrelated job failed ]                                                             
Abandoned <<< message_runtime                                 [ Unrelated job failed ]                                                             
Abandoned <<< mk                                              [ Unrelated job failed ]                                                             
Abandoned <<< qt_dotgraph                                     [ Unrelated job failed ]                                                             
Abandoned <<< qt_gui                                          [ Unrelated job failed ]                                                             
Abandoned <<< qt_gui_py_common                                [ Unrelated job failed ]                                                             
Abandoned <<< rosbuild                                        [ Unrelated job failed ]                                                             
Abandoned <<< roslisp                                         [ Unrelated job failed ]                                                             
Abandoned <<< rospy                                           [ Unrelated job failed ]                                                             
Abandoned <<< rosservice                                      [ Unrelated job failed ]                                                             
Abandoned <<< roscpp_serialization                            [ Unrelated job failed ]                                                             
Abandoned <<< python_qt_binding                               [ Unrelated job failed ]                                                             
Abandoned <<< roslaunch                                       [ Unrelated job failed ]                                                             
Abandoned <<< angles                                          [ Unrelated job failed ]                                                             
Abandoned <<< rosconsole                                      [ Unrelated job failed ]                                                             
Abandoned <<< pluginlib                                       [ Unrelated job failed ]                                                             
Abandoned <<< qt_gui_cpp                                      [ Unrelated job failed ]                                                             
Abandoned <<< resource_retriever                              [ Unrelated job failed ]                                                             
Abandoned <<< rosconsole_bridge                               [ Unrelated job failed ]                                                             
Abandoned <<< roslz4                                          [ Unrelated job failed ]                                                             
Abandoned <<< rosbag_storage                                  [ Unrelated job failed ]                                                             
Abandoned <<< rostest                                         [ Unrelated job failed ]                                                             
[... cut ...]
[build]   Abandoned: 155 packages were abandoned.                                                                                                  
[build]   Failed:    1 packages failed.                                                                                                            
[build] Runtime: 4 minutes and 36.5 seconds total.                                                                                                 
[build] Note: Workspace packages have changed, please re-source setup files to use them.`

vrgimael, 2018-04-13:
I'm having the same problem.. Any updates here?