mikepurvis/ros-install-osx #110: No definition of [google-mock] for OS [osx]

ejalaa12, 2018-02-07:
Hello,

I am trying to install ros-lunar for osx with your script.
I did comment the homebrew/science tap. (because it is deprecated)
And I run into this issue while doing the `rosdep` command:

```bash
ERROR: the following packages/stacks could not have their rosdep keys resolved
to system dependencies:
roslisp: No definition of [google-mock] for OS [osx]
catkin: No definition of [google-mock] for OS [osx]
```

Should I build google-mock myself? If so, how do I say to rosdep where to find it?
If not, what should I do ?

Thank you.

quinnabrvau, 2018-02-07:
I'm having the same issue, I tried building google-mock myself but it doesn't seem to reference it still

mikepurvis, 2018-02-07:
Homebrew doesn't package gtest/gmock because the maintainers explicitly recommend against doing so— they're prefer you vendor the source into your projects rather than having a globally-installed version.

GTest is currently handled by a formula on the `ros/deps` tap, here: https://github.com/ros/homebrew-deps/blob/master/gtest.rb

So GMock should be handled the same way.

In the very immediate, if you've already built and install GMock yourself, you can modify your `rosdeps.yaml` overlay file to blank out the dependency, similar to how we do for opencv and qt:

https://github.com/mikepurvis/ros-install-osx/blob/master/rosdeps.yaml

quinnabrvau, 2018-02-08:
It installs as gmock instead of google-mock, is there a way to change the name that catkin is looking for or do I need to change all of the gmock->google-mock ? 
thanks for the help

ejalaa12, 2018-02-08:
@mikepurvis Great! I blanked the dependency as proposed. and it worked! I am now stuck during compilation for `xmlrpcpp`. I will look into it later. But I guess that's another issue. Thank you 👍 

So what could be a long-term solution for Gmock ?

ejalaa12, 2018-02-09:
So I just had a look for the compilation error for `xmlrpcpp`, it is actually because it cannot find where I installed `gmock`. I guess I have to modify the `CMakeLists.txt` of the `xmlrpcpp` package, but I'm not sure on how to do that.

neomanic, 2018-02-13:
@ejalaa12 I'm having the same issue. This *isn't* because the build can't find gmock but rather because when building the tests it tries to link them against the symbols from libxmlrpcpp.dylib (which is in `/ros-install-osx/lunar_desktop_full_ws/devel/.private/xmlrpcpp/lib`) but said newly-built library isn't passed to the link command.

I'm not smart enough to quickly figure out how to fix this, so I've just skipped building the tests for xmlrpcpp by commenting out the last 3 lines in `CMakelists.txt` starting with `if (CATKIN_ENABLE_TESTING)` and now it builds. (And I'm on to the next error, but that's a different story.)

My build log from running `catkin build -v --start-with xmlrpcpp`:
```[ 81%] Linking CXX shared library /Users/Josh/Code/ros-install-osx/lunar_desktop_full_ws/devel/.private/xmlrpcpp/lib/libtest_fixtures.dylib
cd /Users/Josh/Code/ros-install-osx/lunar_desktop_full_ws/build/xmlrpcpp/test && /usr/local/Cellar/cmake/3.10.2/bin/cmake -E cmake_link_script CMakeFiles/test_fixtures.dir/link.txt --verbose=1
/usr/bin/c++ -O3 -DNDEBUG -dynamiclib -Wl,-headerpad_max_install_names  -o /Users/Josh/Code/ros-install-osx/lunar_desktop_full_ws/devel/.private/xmlrpcpp/lib/libtest_fixtures.dylib -install_name /Users/Josh/Code/ros-install-osx/lunar_desktop_full_ws/devel/.private/xmlrpcpp/lib/libtest_fixtures.dylib CMakeFiles/test_fixtures.dir/test_fixtures.cpp.o  -L/opt/ros/lunar/lib /usr/local/lib/libboost_system-mt.dylib /usr/local/lib/libboost_thread-mt.dylib /usr/local/lib/libboost_chrono-mt.dylib /usr/local/lib/libboost_date_time-mt.dylib /usr/local/lib/libboost_atomic-mt.dylib 
Undefined symbols for architecture x86_64:
  "XmlRpc::XmlRpcValue::invalidate()", referenced from:
      XmlRpc::XmlRpcValue::operator=(char const*) in test_fixtures.cpp.o
  "XmlRpc::XmlRpcValue::operator=(XmlRpc::XmlRpcValue const&)", referenced from:
      XmlRpc::XmlRpcValue::operator=(char const*) in test_fixtures.cpp.o
  "XmlRpc::XmlRpcServer::bindAndListen(int, int)", referenced from:
      XmlRpcTest::SetUp() in test_fixtures.cpp.o

[... snip the rest, it's just more missing symbols ...]
```

ejalaa12, 2018-02-17:
@neomanic Thank you! I also commented these test so I can continue the installation as well. I managed to solve the next errors and it compiled successfully.

However now that I run the roscore I get the following error:
`libc++abi.dylib: terminating with uncaught exception of type std::length_error: vector`
I'm still to figure out where it comes from. (also another story i guess)

Karsten1987, 2018-02-18:
I've just run into the same problem (std::length_error). Don't really have a clue yet where it actually comes from.

vrgimael, 2018-04-13:
Running into the same problem... Any news here?

plguhur, 2018-05-09:
I just install it. Just to resume the workarounds:

First, make sure your python install is clean. Uninstalling everything (except the system one), and installing it with brew worked for me. A `brew postinstall python2` was necessary to recover `pip`. 

Then, `lunar` seems to be the easiest version to install.

For the google-mock issue, make sure you have installed google-mock, and then remove the dependencies by adding the following in `ros-install-osx/rosdeps.yaml`:

(see answer above from @mikepurvis)

```
google-mock:
  osx:
    homebrew:
      packages: []
```

For the `rostime` issue, you need to switch from 1.67 to 1.63:

(see answer from @Arsapol in https://github.com/mikepurvis/ros-install-osx/issues/116):

```
brew reinstall https://raw.githubusercontent.com/Homebrew/homebrew-core/2898efab0f331be9663a640c2fae8b191b9538b2/Formula/boost.rb
```

For the `xmlrpcpp` issue, you need to comment the three last lines of `lunar_desktop_full_ws/src/ros_comm/xmlrpcpp/CMakeLists.txt`. See details from @neomanic above.

asimonov, 2018-05-09:
@mikepurvis,

great work!

but why would you not simply add `--skip-keys gtest --skip-keys google-mock` to `rosdep install`? and also set `-DCATKIN_ENABLE_TESTING=0` in `catkin config`

this is what i have to manually do to get it working.

mikepurvis, 2018-05-09:
There used to be a big raft of `--skip-keys` flags; adding a rosdep file of overrides is functionally equivalent but IMO more flexible, since it lets you do things like change a key to map to something else (as with Gazebo).

I have a mixed relationship with `CATKIN_ENABLE_TESTING`— in the past there have been packages which would fail to configure when you set it to zero. If that's better these days, we can add it again.

ShuminY, 2018-05-12:
there is already no Gmock dependency in the rosdeps.yaml, but the install script still returned "No definition of [google-mock] for OS [osx] " unless the --skip-keys was added. anything else yet to be done if not using the --skip-keys option?

basavaraj1995, 2019-09-12:
`rosdep install --from-paths src --ignore-src --skip-keys python-wxtools --skip-keys google-mock --rosdistro kineti`

HochGit, 2020-02-29:
> `rosdep install --from-paths src --ignore-src --skip-keys python-wxtools --skip-keys google-mock --rosdistro kinetic`
Great answer! it works for me, I think you might have missed a letter _c_ at the end.