mikepurvis/ros-install-osx #114: std::length_error when starting roscore

Karsten1987, 2018-02-18:
I am putting the error mentioned in https://github.com/mikepurvis/ros-install-osx/issues/110#issuecomment-366438956 into a separate issue.

After digging a bit around, the `std::length_error` is thrown in the `XMLRpcServer()`, when calling 'resize` on the vector. https://github.com/ros/ros_comm/blob/lunar-devel/utilities/xmlrpcpp/src/XmlRpcServer.cpp#L37

When commenting out the following section, the roscore doesn't throw the length_error exception anymore, however pub/sub still doesn't seem to work correctly.

https://github.com/ros/ros_comm/blob/lunar-devel/utilities/xmlrpcpp/src/XmlRpcServer.cpp#L32-L36
```
if(getrlimit(RLIMIT_NOFILE, &limit) == 0) {
   max_files = limit.rlim_max;
} else {
  XmlRpcUtil::error("Could not get open file limit: %s", strerror(errno));
}
```

ejalaa12, 2018-02-20:
+1. Same here.
The rosocore throws the following error:
```
[ERROR] [1519167618.751419000]: [registerService] Failed to contact master at [ejalaa.home:11311].  Retrying...
```

qiao-tw, 2018-02-23:
Same here. I tried to replace 
`max_files = limit.rlim_max;`
with
`max_files = std::min((unsigned long long)limit.rlim_max, (unsigned long long)pollfds.max_size());`
but the issue was not solved...

SHTseng, 2018-03-05:
Hi, I currently solve this problem by replacing most of the ros_comm sub-package to lunar-edge branch,
but keep the rosbag and rosbag_storage with lunar-devel branch.
I am able to pass the compilation and start the roscore, however there are still several issues that I can't test turtlesim successfully.

tomlankhorst, 2018-03-23:
Was anyone able to fix this issue? 
I feel like compiling with GCC instead of Clang might solve some of the issues but that would also require compilation of some dependencies with GCC to maintain compatibility. 
@SHTseng , how can I configure another source of lunar packages (edge, devel)?

NikolausDemmel, 2018-03-30:
Reverting ros_comm to 1.13.5 is a workaround, see also https://github.com/ros/ros_comm/issues/1357#issuecomment-377637899

NikolausDemmel, 2018-03-30:
> how can I configure another source of lunar packages (edge, devel)?

Depends a bit how you created your workspace? With the install script from this repository?

dirk-thomas, 2018-03-30:
Please consider filling a problem like this upstream in the future. It has been over a month that this issue is known here but until now no ticket was filled in the `ros_comm` repo containing the code in question. [My comment there](https://github.com/ros/ros_comm/issues/1357#issuecomment-377636182) describes what the actual problem is.