mikepurvis/ros-install-osx #115: High Sierra: Services only connect in one direction

idlebear, 2018-02-18:
I've successfully completed several builds with both kinetic, and lunar, and though the build is successful, when I run against roscore, connections aren't completed.  For instance, if I want the turtle sim to work, I have to start turtle_teleop_key first, then turtlesim_node.  If started in the other order, the nodes appear in the graph (visible in rqt_graph), but aren't connected.

Platform:
```
System Version:	macOS 10.13.3 (17D47)
  Kernel Version:	Darwin 17.4.0
Apple LLVM version 9.0.0 (clang-900.0.39.2)
Target: x86_64-apple-darwin17.4.0
Thread model: posix
InstalledDir: /Library/Developer/CommandLineTools/usr/bin
```
Also reproducible with just roscore and rostopic.   Starting roscore:
```
gadgetzan:kinetic_ros_comm_ws bear$ roscore
... logging to /Users/bear/.ros/log/8b1105b5-14ca-11e8-8281-acde48001122/roslaunch-gadgetzan-18198.log
Checking log directory for disk usage. This may take awhile.
Press Ctrl-C to interrupt

started roslaunch server http://gadgetzan:49649/
ros_comm version 1.12.12

SUMMARY
========

PARAMETERS
 * /rosdistro: kinetic
 * /rosversion: 1.12.12

NODES

auto-starting new master
process[master]: started with pid [18205]
ROS_MASTER_URI=http://gadgetzan:11311/

setting /run_id to 8b1105b5-14ca-11e8-8281-acde48001122
process[rosout-1]: started with pid [18208]
started core service [/rosout]
```

Followed by rostopic:
```
rostopic echo rosout
```

Nothing is echoed -- which isn't unexpected as nothing else is connected.  However, running roswtf shows:
```

gadgetzan:kinetic_ros_comm_ws bear$ roswtf
No package or stack in context
================================================================================
Static checks summary:

No errors or warnings
================================================================================
Beginning tests of your ROS graph. These may take awhile...
analyzing graph...
... done analyzing graph
running graph rules...
... done running graph rules

Online checks summary:

Found 1 error(s).

ERROR The following nodes should be connected but aren't:
 * /rostopic_18298_1518972718392->/rosout (/rosout)
```

The symptoms are the same with turtles -- nodes aren't connected.  Non-connection also occurs if the master is run on an otherwise working Ubuntu kinetic install.  Ubuntu clients work fine, but clients from OSX can't make connections.

I built it on a clean system (removed all pip and uninstalled brew) and had the following issues to work around:

* catkin build ... crashes with the out of files/ptys error.  Built with catkin_make_isolated, using the same compile options, instead.  catkin_make_isolated can be found in src/catkin/bin.

* rospack complains about missing boost/tr1/unordered_map, etc.  Replaced with boost/unordered_map.

* rosbag_storage/src/bz2_stream.cpp and lz4_stream.cpp both use an undefined logError/logWarning -- redefined as console_bridge::log...

Thanks in advance for any help or insight that can be provided.

Barry

idlebear, 2018-02-18:
One last comment -- the result is the same with and without SIP disabled

horagong, 2018-08-08:
I think you need to try changing ros_comm version.
https://github.com/ros/ros_comm/issues/1469